contiki-ng/contiki-ng #1493: CC1312: calculating SHA2 results in crash if data length is larger then 512 bytes and not a multiple of 64

tdesmet, 2021-01-22:
I want to use the functions in the coresdk to calculate an SHA2 hash but when the data I am trying to hash is larger then 512 bytes and the length is not a multiple of 64 then the cc1312 crashes.

When I try the same code in code composer studio with the ti sdk everything works

Example code
```c
#include "contiki.h"

#include <stdio.h> /* For printf() */
#include <ti/drivers/SHA2.h>
/*---------------------------------------------------------------------------*/
PROCESS(hello_world_process, "Hello world process");
AUTOSTART_PROCESSES(&hello_world_process);

/*---------------------------------------------------------------------------*/
PROCESS_THREAD(hello_world_process, ev, data)
{
  static struct etimer timer;

  PROCESS_BEGIN();

  /* Setup a periodic timer that expires after 10 seconds. */
  etimer_set(&timer, CLOCK_SECOND / 10);
  while(1) {
    SHA2_Params params;
    SHA2_Handle handle;
    int_fast16_t result;
    uint8_t digest[SHA2_DIGEST_LENGTH_BYTES_256];
    SHA2_Params_init(&params);
    params.returnBehavior = SHA2_RETURN_BEHAVIOR_BLOCKING;
    handle = SHA2_open(0, &params);
    if (!handle)
    {
      printf("Failed to open SHA2\n");
      continue;
    }
    result = SHA2_setHashType(handle, SHA2_HASH_TYPE_256);
    if (result != SHA2_STATUS_SUCCESS) {
      SHA2_close(handle);
      printf("SHA2_setHashTypeFailed: %d\n", result);
      continue;
    }
    uint8_t buffer1[512];
    uint8_t buffer2[19];

    result = SHA2_addData(handle, buffer1, 512);
    result = SHA2_addData(handle, buffer2, 19);
    printf("call finalize\n");
    result = SHA2_finalize(handle, digest);
    printf("I never get here\n");
    if (result != SHA2_STATUS_SUCCESS) {
      SHA2_close(handle);
      printf("SHA2 finish Failed: %d\n", result);
      continue;
    }
    printf("SHA2: ");
    for (uint8_t i = 0;i < 32; i++) {
      printf("%02X", digest[i]);
    }
    printf("\n");
    SHA2_close(handle);
    /* Wait for the periodic timer to expire and then restart the timer. */
    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&timer));
    etimer_reset(&timer);
  }

  PROCESS_END();
}
/*---------------------------------------------------------------------------*/

```

The crash happens on the call to SHA2_finalize.
It does not crash if I remove `result = SHA2_addData(handle, buffer2, 19);` Or If I change the length of buffer2 to be a multiple of 64 or If I remove `result = SHA2_addData(handle, buffer1, 512);`

damien-v, 2021-02-04:
It has probably nothing to do with the issue but your timer will expire after 100ms, not 10s.

tdesmet, 2021-02-12:
> It has probably nothing to do with the issue but your timer will expire after 100ms, not 10s.

you are correct on both accounts, I forgot to change the comment while testing things. But the timer can be ignored anyway.

damien-v, 2021-03-15:
I had a look at it, I'm new to the cc1312 but I worked previously with the cc1310 so bear with me if there are things I don't fully grab.

Just a quick note in the snippet above

```c
SHA2_init();
```
is missing, in case someone tried to reproduce.

But yes, I see the same behaviour on the CC1312r1 launchpad. The reset is caused by the watchdog, I increased the reload value in watchdog-arch.c:

```c
const WatchdogCC26XX_HWAttrs watchdogCC26XXHWAttrs[CC1312R1_LAUNCHXL_WATCHDOGCOUNT] = {
    {
        .baseAddr    = WDT_BASE,
        .reloadValue = 10000 /* Reload value in milliseconds */
    },
};
```

And it took longer (x10) to trigger a reset. So the call to `SHA2_finalize` is blocking indefinitely.

However, when trying to debug further I got a bit stuck, since we apparently don't compile the driver code but use directly the drivers_cc13x2.am4fg library.

Could that be the difference with the CCS setup and the contiki setup?

tdesmet, 2021-08-02:
Woops, forgot about this.

So I tested the same thing as you did. and reached the same conclusions. 
In CCS I went back to an old version of the simplelink sdk  that was released around the same time the corse sdk in contiki-ng was updated.

Link to the issue I created on TI's forum when I ran into this issue: https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz-group/sub-1-ghz/f/sub-1-ghz-forum/972362/cc1312r-sha2-on-contiki-results-in-crash-if-data-length-is-larger-then-512-bytes-and-not-a-multiple-of-64

The person responding said they tested with simplelink_cc13x2_26x2_sdk_3_10_00_53 (Core SDK 4.10.01.00) and could not reproduce. I can't remember which version I tested but I think I tested all of sdk's released around that time.

I would test with a newer version of the coresdk but I lack the experience and knowledge to use a newer version.