contiki-ng/contiki-ng #1496: Minor issues with simplelink-platform watchdog

arurke, 2021-02-01:
_Disclaimer: I am not using simplelink-platform myself, so I might well be mistaken. I noticed these during work with the cc26x0-cc13x0 platform watchdog._

----

1. It seems [watchdog_init()](https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/simplelink-cc13xx-cc26xx/dev/watchdog-arch.c#L79) eventually will call [WatchdogCC26XX_initHw()](https://github.com/contiki-ng/coresdk_cc13xx_cc26xx/blob/master/source/ti/drivers/watchdog/WatchdogCC26XX.c#L205) which will actually enable the watchdog - which I believe is not the intention of init().
2. `watchdog_stop()` does not actually stop the watchdog, it only clears the current interrupt and restarts the counter. Thus the WD may still trigger at the next timeout: https://github.com/contiki-ng/contiki-ng/blob/48a3799e2d2c52e91cb7153b163bb26dfd9a4b4f/arch/cpu/simplelink-cc13xx-cc26xx/dev/watchdog-arch.c#L136-L141

tdesmet, 2021-03-02:
According to TI's documentation you can only stop the watchdog by a hardware reset on this platform.

See https://software-dl.ti.com/simplelink/esd/simplelink_cc13x0_sdk/4.10.02.04/exports/docs/tidrivers/doxygen/html/_watchdog_c_c26_x_x_8h.html
> Watchdog_close() is not supported by this driver implementation. Once started, the Watchdog timer can only be stopped by a hardware reset.

arurke, 2021-03-02:
In cc13x0-cc26x0 platform, the reset functionality is disabled via `WatchdogResetDisable()`. I have not tested it but I assume the WD will thus appear to OS and users as if it was disabled.