contiki-ng/contiki-ng #1497: Cooja from contiker in windows 10 closes after a few minutes

damien-v, 2021-02-01:
Hello,

I am using cooja from the contiker/contiki-ng docker image on a windows host with Xming.

I start the session with the following command
`docker run --privileged --mount type=bind,source=C:\repos\contiki-ng,destination=/home/user/contiki-ng -e DISPLAY=host.docker.internal:0.0 -it contiker/contiki-ng cooja
`

and it works fine. However, if I am inactive for a few minutes, the session closes as if I had exited cooja manually:

`     [java]  INFO [AWT-EventQueue-0] (ContikiMoteType.java:580) - Parsed section at 0x37500 ( 1008 == 0x3f0 bytes)
     [java]  INFO [AWT-EventQueue-0] (ContikiMoteType.java:580) - Parsed section at 0x37900 ( 56504 == 0xdcb8 bytes)
     [java]  INFO [AWT-EventQueue-0] (ContikiMoteType.java:502) - mtype897.cooja: offsetting Cooja mote address space: 0xc2169000
     [java]  INFO [AWT-EventQueue-0] (ContikiMoteType.java:580) - Parsed section at 0xc21a0500 ( 1008 == 0x3f0 bytes)
     [java]  INFO [AWT-EventQueue-0] (ContikiMoteType.java:580) - Parsed section at 0xc21a0900 ( 56504 == 0xdcb8 bytes)
     [java]  INFO [Thread-2] (Simulation.java:253) - Simulation main loop started, system time: 1612197456655
     [java]  INFO [Thread-2] (Simulation.java:312) - Simulation main loop stopped, system time: 1612197466221   Duration: 9566 ms       Simulated time 7915547 ms       Ratio 827.4667572653146
     [java] Java Result: 1

BUILD SUCCESSFUL
Total time: 29 minutes 54 seconds`

This doesn't happen if the simulation keeps running, only when it is stopped.

Any idea how to prevent this? I am using contiki-ng 4.5, if someone thinks it could be fixed in release 4.6 I'll give it a try.

sungminlee114, 2022-05-23:
Same problems for me. Any updates?

sungminlee114, 2022-05-25:
@pjonsson 

Yes.

Starting docker (bash as entrypoint) using
`docker run --privileged --sysctl net.ipv6.conf.all.disable_ipv6=0 --mount type=bind,source="/My/directory/contiki-ng",destination=/home/user/contiki-ng -e DISPLAY="host.docker.internal:0.0" -ti contiker/contiki-ng`

and running cooja with
`cd tools/cooja`
`ant`

do the same.

Any traces I can share?

sungminlee114, 2022-05-25:
In addition, here is the spec of host CPU
- CPU : Intel(R) Core(TM) i7-9700K
- RAM : 32GB
- OS : Windows 10(build: 19042.1706), 64bit 

I've also tracked the CPU and RAM utilization, but they all remain low enough.

pjonsson, 2022-05-25:
I don't use Windows, but I had a look at the Docker image (which is built from `tools/docker/Dockerfile`). I can't find any traces of ssh configuration in `/etc` and no ssh package is installed in the Dockerfile. I also had a look if `TMOUT` is set, and I can't find any traces of that in the docker image either.

I continuously run the Docker image on Linux (without Cooja). Some days I don't type anything in that window at all and I've never been logged out. I can try to reproduce this issue on Linux by just starting Cooja though, how long is "a couple of minutes", 5-10?

My best guess is that it's something on the Windows side, but unfortunately I can't help with that. If you have a different shell available on your Windows machine, it might be worth trying to start docker from that shell instead. I'm not sure if powershell is available everywhere, but I think git for Windows comes with Bash, and WSL might have yet another shell included.

If you're dealing with some kind of timeout due to inactivity, here's a potential workaround by periodically sending invisible characters:

https://unix.stackexchange.com/questions/222283/read-only-tmout-variable-workaround-to-disable-automatically-exit-shell-when-t

If that works, it would probably be worth adding to the cooja script in the Docker image.

damien-v, 2022-05-25:
Thank you for the input @pjonsson 
"a couple of minutes" is give or take 8 minutes (I haven't timed it precisely). It always happens when I go grab a cup of coffee or decide to read my emails...
I'm trying with powershell and if it doesn't help will try with other shells if possible - not really my area of expertise but really worth giving it a try. I'll get back with the results!

sungminlee114, 2022-05-25:
Alongside the hypothesis @damien-v came up with, I've managed to solve the problem by turning off the cloud auto-syncing program (Google drive auto-sync, in my case). But since this could be a temporal solution, I'll report if this problem occurs again.

Thanks much to pjonsson, damien  :)

damien-v, 2022-05-25:
> Thank you for the input @pjonsson "a couple of minutes" is give or take 8 minutes (I haven't timed it precisely). It always happens when I go grab a cup of coffee or decide to read my emails... I'm [...]

So, I've tried running from powershell and git bash (MINGW). It's the same, the session is terminated after roughly 8 minutes.

pjonsson, 2022-05-26:
To replicate your scenario, I tried starting cooja and then making some coffee and responding to some emails.
```
~$ contiker cooja
Buildfile: /home/user/contiki-ng/tools/cooja/build.xml

...

run:
     [java]  INFO [AWT-EventQueue-0] (Cooja.java:2841) - External tools default settings: /external_tools_linux.config
     [java]  WARN [AWT-EventQueue-0] (Cooja.java:2873) - Error when reading user settings from: /home/user/.cooja.user.properties

BUILD SUCCESSFUL
Total time: 18 minutes 36 seconds
```
The window focus was never on Cooja during this time, so it's probably not something inside the docker image that causes the problem.

Could you try editing `/home/user/cooja` inside docker to this:
```
#!/bin/bash
ant -Dbasedir=/home/user/contiki-ng/tools/cooja -f /home/user/contiki-ng/tools/cooja/build.xml run &
while sleep 120; do printf '\33[0n'; done
```
and then starting cooja through the script without exiting docker and see if the problem still remain?

I have made two changes to the file, there is an added `&` at the end of the second line, and the entire third line. The docker image does not seem to have an editor, but you can install an editor with `sudo apt update && sudo apt install emacs`.

damien-v, 2022-05-27:
Hi @pjonsson ,

I tried your suggestion, and it works! the cooja window had now been opened for several hours without me touching it (and emails, and coffee 😉 ).
Thank you so much for your help here. It's an easy workaround! 

Where do we go from here, is it possible to implement the workaround in the docker image, or should we simply document the workaround?

pjonsson, 2022-05-29:
I think it should be included in the docker image, but it would be nice if ctrl-c still works as intended and we don't litter processes around after us. Could you try this script and see if it still works for you?
```
#!/bin/bash

# Periodically print an invisible character to keep Windows docker from
# terminating the session for inactivity.
/bin/bash -c "while sleep 120; do printf '\33[0n'; done" &
PID=$!

trap exit_cleanup INT

function exit_cleanup() {
  kill $PID
}

ant -e -logger org.apache.tools.ant.listener.SimpleBigProjectLogger -f $HOME/contiki-ng/tools/cooja/build.xml run

exit_cleanup
```
If the script works as intended, I suggest you update tools/docker/Dockerfile and make a pull request with it. I don't know if it's possible, but it would be even better if the script is put as a file under tools/docker and the Dockerfile copies the script into the image so we avoid an echo with a 500 character long argument.

pjonsson, 2022-06-08:
Not sure if it was apparent but I edited the script in my last comment with a new target after a few days. This ticket is tagged for next release and since I don't have windows, could you please try the latest version of the script @damien-v and see if it still works for you?

damien-v, 2022-06-08:
Working on it, I expect to submit a PR today

closed 2022-06-10 by damien-v