contiki-ng/contiki-ng #1510: simple-udp.c: client_process is never set

kluchsinger, 2021-02-24:
In simple-udp.c, the process context is set to client_process right before calling the receive_callback:
[https://github.com/contiki-ng/contiki-ng/blob/48a3799e2d2c52e91cb7153b163bb26dfd9a4b4f/os/net/ipv6/simple-udp.c#L163](url)

However, client_process is not set anywhere.

If I understand it correctly, it should be set in simple_udp_register():
`c->client_process = PROCESS_CURRENT();`

Or am I misunderstanding something here?

damien-v, 2021-03-16:
I think you are correct. Maybe submit a pull request and hopefully someone with sufficient rights will have look at it?

nvt, 2022-05-18:
Thanks for reporting this issue -- it has now been fixed.

closed 2022-05-18 by nvt