contiki-ng/contiki-ng #1513: Tutorial: ContikiNG over-the-air download for offchip CC1310

rajeev1986, 2021-03-01:
Don't know where this should be announced because it's neither an issue nor a pull request. Recently, I have published an extensive tutorial on implementing over-the-air download for CC1310 in ContikiNG. The complete tutorial can be found [here](https://www.rajeevpiyare.com/posts/contiki-ng-ota/). This work is still on-going...

What makes this implementation different from other previous ContikiNG OTA implementations is that:

- it allows one to use the TI's boot image manager (BIM) included in their CC1310 SDK together with the OTA compatible image generated in ContikiNG without changing any inner Contiki-NG source files keeping the OS intact and standardized.
- eliminates implementing a custom bootloader in ContikiNG.
- it allows utilizing the OAD image creation tools provided in the TI's SDK.
- it supports the simplelink/cc1310 platform atm.
- all the source files are made [open-source](https://github.com/rajeev1986/contikNG-offchip-oad-cc1310).
- documentation providing the inner working mechanism of the boot image manager, requirements of an OTA bootable image & its header/metadata information, and how to generate one in ContikiNG.

Hope this will be useful for anyone trying to implement OTA for ContikiNG or for novice starting with CC1310 OTA in general.

Any feedback or comments are most welcome. 

Cheers...

g-oikonomou, 2021-03-12:
@rajeev1986 Rajeev, this looks great. I'll close it since it will remain googlable anyway. Maybe we can start a space in the wiki signposting to external spaces

rajeev1986, 2021-03-12:
@g-oikonomou Yes, please go ahead and close it as an issue here. A Contiki wiki page for such projects and curations sounds like a good plan. Thanks...

closed 2021-03-12 by rajeev1986