contiki-ng/contiki-ng #1526: Problem when trying to use cooja on macOS (Big Sur)

mecking, 2021-03-11:
Hi all,
I'm trying to use(I followed the installation procedure described at https://github.com/contiki-ng/contiki-ng/wiki/Docker section "On macOS") Cooja on my macbook, but I'm getting the error below:
```
henrique@MacBook-Pro-de-Henrique contiki-ng % contiker bash
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
user@4af9c4b422f4:~/contiki-ng$ cd tools/
user@4af9c4b422f4:~/contiki-ng/tools$ cd cooja/
user@4af9c4b422f4:~/contiki-ng/tools/cooja$ ant run
Buildfile: /home/user/contiki-ng/tools/cooja/build.xml

init:

compile:
    [mkdir] Created dir: /home/user/contiki-ng/tools/cooja/build
    [javac] Compiling 172 source files to /home/user/contiki-ng/tools/cooja/build
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00c0c34a, pid=17, tid=0xdb153b40
#
# JRE version: OpenJDK Runtime Environment (8.0_275-b01) (build 1.8.0_275-8u275-b01-0ubuntu1~18.04-b01)
# Java VM: OpenJDK Server VM (25.275-b01 mixed mode linux-x86 )
# Problematic frame:
# J 1517 C2 com.sun.tools.javac.parser.JavaTokenizer.readToken()Lcom/sun/tools/javac/parser/Tokens$Token; (2243 bytes) @ 0x00c0c34a [0x00c0af20+0x142a]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/user/contiki-ng/tools/cooja/hs_err_pid17.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
qemu: uncaught target signal 6 (Aborted) - core
```
MacOS Big Sur 11.2.3

```
henrique@MacBook-Pro-de-Henrique docker % docker version
Client: Docker Engine - Community
 Cloud integration: 1.0.9
 Version:           20.10.5
 API version:       1.41
 Go version:        go1.13.15
 Git commit:        55c4c88
 Built:             Tue Mar  2 20:13:00 2021
 OS/Arch:           darwin/amd64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.5
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       363e9a8
  Built:            Tue Mar  2 20:15:47 2021
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.4.3
  GitCommit:        269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc:
  Version:          1.0.0-rc92
  GitCommit:        ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

Yagoor, 2021-03-14:
Hi,

I was going to make a similar issue.

I was experiencing the same issue while using the following version: https://docs.docker.com/docker-for-mac/release-notes/#docker-desktop-321.
I noticed that while using version 3.2.1 the docker is running everything through Qemu 

```bash
Yagos-MacBook-Pro:contiki-ng rosario$ contiker 
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
user@eb6cc00f0b82:~/contiki-ng$ ps aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
user         1  5.5  0.1 4404716 6176 pts/0    Ssl  08:19   0:00 /usr/bin/qemu-i386 /bin/sh -c bash --login
user         9  6.5  0.3 4405224 12260 pts/0   Sl   08:19   0:00 /usr/bin/qemu-i386 /bin/bash --login
user        22  0.0  0.2 4404880 10700 ?       Rl+  08:17   0:00 /bin/ps aux
user@eb6cc00f0b82:~/contiki-ng$ 

```

When I tried to downgrade the docker to this version, https://docs.docker.com/docker-for-mac/release-notes/#docker-desktop-310, it worked normally and the Qemu is no longer present inside the docker container.

``` bash
Yagos-MacBook-Pro:contiki-ng rosario$ contiker
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
user@7cbe3ffe1711:~/contiki-ng$ ps aux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
user         1  0.2  0.0   2492  1320 pts/0    Ss   08:28   0:00 /bin/sh -c bash --login
user         7  0.0  0.0   4052  3080 pts/0    S    08:28   0:00 bash --login
user        12  0.0  0.0   5464  2532 pts/0    R+   08:28   0:00 ps aux
user@7cbe3ffe1711:~/contiki-ng$ 

```

I'm guessing this is related to some new functionality to enable cross platform dockers, due to the new M1.

alexstanoev, 2021-03-15:
This seems to be an upstream Docker bug, and should be fixed in version 3.2.2 that was released today.
Can you try and see if that fixes the issue?

The following issues from docker/for-win seem relevant:
* 10604 - After Upgrade from Docker 3.2.0 / 3.1.0 to Latest (3.2.1) Java stops functioning inside container
* 10594 - 3.2.0 appears to virtualise 32-bit executables

Yagoor, 2021-03-15:
Hi,

I just tried on the latest docker and I could run Cooja and the Qemu is gone. 
Hopefully @mecking will have the result.

jsolderitsch, 2021-03-16:
Somewhat by coincidence I am seeing the qemu issue with a Docker Desktop for Windows: version 3.2.2 instance. Nothing to do with Java, but trying to run my contiker alias is giving me the same message.
```
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
```
In an earlier comment, a suggestion was made that the Desktop version from today was a remedy. I also see qemu processes in the output of ps aux.
Does the dockerfile for contiki-ng make an architecture choice of linux/386? I am trying to run contikig-ng in an ubuntu 20.04 64 bit wsl2 shell. Should this work out of the box?

mecking, 2021-03-17:
Hi, I tested it and in version 3.2.2 it is working correctly.
I believe that the issue can now be closed

closed 2021-03-17 by mecking