contiki-ng/contiki-ng #1532: Ringbuf has one slot fewer than advertised

fleutot, 2021-03-17:
[ringbuf.c](https://github.com/contiki-ng/contiki-ng/blob/develop/os/lib/ringbufindex.c)
```int
ringbuf_put(struct ringbuf *r, uint8_t c)
{
  /* Check if buffer is full. If it is full, return 0 to indicate that
     the element was not inserted into the buffer.
     XXX: there is a potential risk for a race condition here, because
     the ->get_ptr field may be written concurrently by the
     ringbuf_get() function. To avoid this, access to ->get_ptr must
     be atomic. We use an uint8_t type, which makes access atomic on
     most platforms, but C does not guarantee this.
  */

/*** This condition below is true when "put" is just under "get". 
 * This handles "put" as "the last place where an item was pushed". 
 */

  if(((r->put_ptr - r->get_ptr) & r->mask) == r->mask) {
    return 0;
  }
  /*
   * CC_ACCESS_NOW is used because the compiler is allowed to reorder
   * the access to non-volatile variables.
   * In this case a reader might read from the moved index/ptr before
   * its value (c) is written. Reordering makes little sense, but
   * better safe than sorry.
   */

/*** These lines below handle "put" as "the next place to push a new item".
 * This is not coherent with the earlier check.
 */
  CC_ACCESS_NOW(uint8_t, r->data[r->put_ptr]) = c;
  CC_ACCESS_NOW(uint8_t, r->put_ptr) = (r->put_ptr + 1) & r->mask;
  return 1;
}```

See my comments in the code above, marked with `/***`. My experience is that the last slot in a ringbuf is not accessible, which I encountered writing unit tests.

I'd be happy to be told that I'm wrong. I don't know how to solve this (because "get" == "put" means empty buffer), beside saving the number of pending elements.

alexrayne, 2021-03-17:
yes, exactly that is.
need to prevent buffer come from full-> empty state. Since empty check is `put_ptr == get_ptr`

fleutot, 2021-03-17:
So you confirm that initializing a ringbuffer with size 64 does **not** give 64 slots that are available at a given time, but only 63?

alexrayne, 2021-03-17:
yep, that is. 
there is no way to distinguish empty and full states without this politics.

atiselsts, 2021-03-30:
This does not appear to be a bug. Perhaps the issue is insufficient documentation, which could be improved in the future.

closed 2021-03-30 by atiselsts