contiki-ng/contiki-ng #1540: network is not stable when setting LINKADDR_SIZE to 2

lili1218, 2021-04-01:
Hi,
I am running contiki-V4.5 + rpl-lite + TSCH on cc1310 launch pad. is was quite stable in terms of network. Since I want to increase hop number, I set LINKADDR_SIZE to 2 instead of 8. Just found out, network is broken. most of  time it will miss an MAC-layer ack. May I know do I miss other config setting? or why it will disturb timing when changing link layer address length ? Even I update to latest develop branch or V4.6, it doesn't improve network stability.
thanks for the help~

file change:
```
#ifdef LINKADDR_CONF_SIZE
#define LINKADDR_SIZE LINKADDR_CONF_SIZE
#else /* LINKADDR_SIZE */
#define LINKADDR_SIZE 2
#endif /* LINKADDR_SIZE */
```

log msg:
GW:
```
[INFO: TSCH-LOG  ] {asn 00.00001161 link  1  17   0 12 184 ch 25} uc-1-0 tx LL-631a->LL-67b6, len  33, seq 142, st 2  1
[DBG : TSCH Queue] packet is added put_index 4, packet #0x20002060
[INFO: TSCH      ] TSCH: enqueue EB packet 29 10
[WARN: TSCH      ] ! drop dup ll from 67b6 seqno 20
[INFO: TSCH-LOG  ] {asn 00.0000116f link  1  17   0  9 28 ch 20} uc-1-0 rx LL-67b6->LL-631a, len  75, seq  20, edr  -1
[INFO: TSCH-LOG  ] {asn 00.00001172 link  1  17   0 12 184 ch 26} uc-1-0 tx LL-631a->LL-67b6, len  33, seq 142, st 2  2
[INFO: TSCH-LOG  ] {asn 00.00001183 link  1  17   0 12 184 ch 20} uc-1-0 tx LL-631a->LL-67b6, len  33, seq 142, st 2  3
[INFO: RPL       ] nbr: own state, addr fd00::ff:fe00:631a, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 14, nbr count 1 (Periodic)
[INFO: RPL       ] nbr: fe80::ff:fe00:67b665535,   371 => 65535 --  8    f  (last tx 2 min ago)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] links: 2 routing links in total (Periodic)
[INFO: RPL       ] links: fd00::ff:fe00:631a  (DODAG root) (lifetime: infinite)
[INFO: RPL       ] links: fd00::ff:fe00:67b6  to fd00::ff:fe00:631a (lifetime: 1740 seconds)
[INFO: RPL       ] links: end of list
[WARN: TSCH      ] ! drop dup ll from 67b6 seqno 20
[INFO: TSCH-LOG  ] {asn 00.00001191 link  1  17   0  9 28 ch 25} uc-1-0 rx LL-67b6->LL-631a, len  75, seq  20, edr  -2
[INFO: TSCH-LOG  ] {asn 00.00001194 link  1  17   0 12 184 ch 15} uc-1-0 tx LL-631a->LL-67b6, len  33, seq 142, st 2  4
[INFO: TSCH-LOG  ] {asn 00.000011a5 link  1  17   0 12 184 ch 25} uc-1-0 tx LL-631a->LL-67b6, len  33, seq 142, st 2  5
```

node:
```
[INFO: TSCH-LOG  ] {asn 00.00001161 link  1  17   0 12 184 ch 25} uc-1-0 rx LL-631a->LL-67b6, len  33, seq 142, edr   3, dr  -3
[INFO: TSCH-LOG  ] {asn 00.0000116f link  1  17   0  9 28 ch 20} uc-1-0 tx LL-67b6->LL-631a, len  75, seq  20, st 2  2
[WARN: TSCH      ] ! drop dup ll from 631a seqno 142
[INFO: TSCH-LOG  ] {asn 00.00001172 link  1  17   0 12 184 ch 26} uc-1-0 rx LL-631a->LL-67b6, len  33, seq 142, edr  -2, dr   2
[WARN: TSCH      ] ! drop dup ll from 631a seqno 142
[INFO: TSCH-LOG  ] {asn 00.00001183 link  1  17   0 12 184 ch 20} drift -9 ppm (min/max delta seen: -3/2)
[INFO: TSCH-LOG  ] {asn 00.00001183 link  1  17   0 12 184 ch 20} uc-1-0 rx LL-631a->LL-67b6, len  33, seq 142, edr   0, dr   0
[INFO: TSCH-LOG  ] {asn 00.00001191 link  1  17   0  9 28 ch 25} uc-1-0 tx LL-67b6->LL-631a, len  75, seq  20, st 2  3
[WARN: TSCH      ] ! drop dup ll from 631a seqno 142
[INFO: TSCH-LOG  ] {asn 00.00001194 link  1  17   0 12 184 ch 15} uc-1-0 rx LL-631a->LL-67b6, len  33, seq 142, edr   1, dr  -1
[WARN: TSCH      ] ! drop dup ll from 631a seqno 142
[INFO: TSCH-LOG  ] {asn 00.000011a5 link  1  17   0 12 184 ch 25} uc-1-0 rx LL-631a->LL-67b6, len  33, seq 142, edr   0, dr   0
[DBG : TSCH Queue] packet is added put_index 0, packet #0x20001ad8
[INFO: TSCH      ] TSCH: enqueue EB packet 29 10
[INFO: RPL       ] nbr: own state, addr fd00::ff:fe00:67b6, DAG state: reachable, MOP 1 OCP 1 rank 499 max-rank 1523, dioint 12, nbr count 1 (Periodic)
[INFO: RPL       ] nbr: fe80::ff:fe00:631a  128,   371 =>   499 --  8 rbafp (last tx 0 min ago)
[INFO: RPL       ] nbr: end of list
[INFO: TSCH-LOG  ] {asn 00.000011b3 link  1  17   0  9 28 ch 20} uc-1-0 tx LL-67b6->LL-631a, len  75, seq  20, st 2  4
[INFO: TSCH-LOG  ] {asn 00.000011c4 link  1  17   0  9 28 ch 15} uc-1-0 tx LL-67b6->LL-631a, len  75, seq  20, st 2  5
```

alexrayne, 2021-04-01:
Hallow Lyly. i works with same problem last summer.
there was a broken PANID handling for addrsize=2.
resulting of my work was [PR](https://github.com/contiki-ng/contiki-ng/pull/1402),  maybe this PR helps?
Fixup of this problem was done before this PR. you can inspect my repo, or i look it in few hours.

damien-v, 2021-04-01:
To answer your second question

> Another question is for hop limitation regarding rpl-lite. is there a way to extend hop number more ? currently, it seems it can support max 8 hops around. If I would like to increase hop number what would be the suggested way to do ?

This is a limitation in the code, when the full ip header does not fit in the first fragment the packet is dropped, see [here](https://github.com/contiki-ng/contiki-ng/blob/develop/os/net/ipv6/sicslowpan.c#L1742).
When using RPL-Lite, downwards packets use source routing so the header becomes larger with the number of hops and at around 8-9 hops the packet header cannot fit in the first fragment.

So for this problem, the only solutions I see are:
- use RPL-Classic with storing mode
- If you do not need downward traffic, keep RPL-lite and set RPL_CONF_WITH_DAO_ACK to 0. Then the node can join the network and send upwards. The node won't be able to receive unicast messages, but will still receive multicast messages, so that can be a solution.
- You can also live with the limit and tune the objective function settings to favor weaker links with less hops for the parent selection, so that the hop limit still can let nodes join the network at greater distance from the root. That may also be a bad idea, and I'm not familiar enough with the objective functions to be more specific.

lili1218, 2021-04-07:
thanks for both of your sharing. I will look into the repo and discuss with team regarding hop limitation. Sounds like there is no perfect solution regarding hop limitation in rpl-lite.

atiselsts, 2021-04-19:
As far as I know, Contiki-NG does not claim to support any other linkaddr size than 8. Most of the code should work of different linkaddr sizes, but we don't have any tests for that, and I suspect quite a few areas get broken if the LINKADDR_SIZE constant is changed.

This is something that we should be looking at in the long term.

alexrayne, 2021-04-19:
> This is something that we should be looking at in the long term.

Hm, i have a fixes for 2byte adress with TSCH, but i close relayted to PR #1402, therefore not proposed it yet.

lili1218, 2021-05-11:
hi @alexrayne  @atiselsts 
could I know in theory, if the MAC address is changed last2 bytes only. actually it will also help to grow the hop number right? since IP is based on MAC address. if only last 2 bytes address is changed. then which mean when doing source routing at gw end. it will use 2 byte for each hop instead of 8 bytes. please correct me if I am wrong.