contiki-ng/contiki-ng #1546: Simulate CoAP Server and BR in Cooja

fkerem, 2021-04-15:
I could not find any mote to simulate coap server in Cooja.
Do you have any suggestions?

jeppenodgaard, 2021-04-15:
Whats your environment and what is you compile error?

Building examples/coap/coap-example-server/ with `make -j coap-example-server.cooja TARGET=cooja` in docker image contiker/contiki-ng should work.

atiselsts, 2021-04-19:
I suggest following jeppenodgaard's advice. Let us know if there are futher questions.

closed 2021-04-19 by atiselsts

fkerem, 2021-04-22:
I use the docker image for Mac.
Here is the compilation output I get when I use the command that you provided @jeppenodgaard

`
> make -j coap-example-server.cooja TARGET=cooja MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC 
  CC        coap-example-server.c
  MKDIR     build/cooja/obj
  CC        ../../../arch/platform/cooja/./platform.c
  CC        ../../../arch/platform/cooja/lib/simEnvChange.c
  CC        ../../../arch/platform/cooja/sys/cooja_mt.c
  CC        ../../../arch/platform/cooja/sys/cooja_mtarch.c
../../../arch/platform/cooja/./platform.c:38:10: fatal error: jni.h: No such file or directory
 #include <jni.h>
          ^~~~~~~
compilation terminated.
make: *** [platform.o] Error 1
../../../Makefile.include:412: recipe for target 'platform.o' failed
make: *** Waiting for unfinished jobs....
  CC        ../../../arch/platform/cooja/./rtimer-arch.c
Process returned error code 2
`

Appreciate your help!

fkerem, 2021-04-22:
No, even though I pulled the latest Docker image, I get the same compilation error:

`
> make coap-example-server.cooja TARGET=cooja 
  CC        coap-example-server.c
  MKDIR     build/cooja/obj
  CC        ../../../arch/platform/cooja/./platform.c
../../../arch/platform/cooja/./platform.c:38:10: fatal error: jni.h: No such file or directory
 #include <jni.h>
          ^~~~~~~
compilation terminated.
make: *** [platform.o] Error 1
../../../Makefile.include:412: recipe for target 'platform.o' failed
Process returned error code 2
`

fkerem, 2021-04-22:
Can this be the case?

https://github.com/contiki-ng/contiki-ng/issues/1224#issuecomment-613673076

My JAVA_HOME env variable is pointing to `/usr/lib/jvm/java-8-openjdk-i386/`.