contiki-ng/contiki-ng #1547: SimpleLink CC1352 With Contiki crashes or watchdog reboot with low probability in tsch-slot-operating when receiving a packet on too many nodes

YanikCysca, 2021-04-15:
We have a working setup with 8 nodes or 8 launchpad CC1352P1 working no problem with TSCH and sending sensors packets with Contiki-ng 4.5 and also 4.6 (newest version)
We use TSCH and basic rpl-udp in our application.

Using simplelink platform for CC1352P1 MCU

When we increase number of nodes to 17+, we get crashes both in nodes and coordinator.

It looks like 90% of the time 1 node crashes and then the whole network will crash soon after.

Sometimes it works well for 1 hour plus without any problems... all 17 nodes sending data to coordinator each 5 minutes.
Sometimes it can crash within 15 minutes.

Watchdog reboot triggers, but then it takes too long to reconnect the mesh network and a lot of nodes stay lost.

How can we prevent this crash or reconnect super fast.... we need to have a network of 100+ nodes and up to 8 hops.

Here is the full detail of the crash:
First, in function tsch_rx_slot, in tsch-slot-operation.c, we call the function receive_packet():
[image]

The crash occurs in prop-mode.c in receive_packet() function when reading the radio register:

HWREG(RFC_DBELL_BASE + RFC_DBELL_O_RFHWIFG) & RFC_DBELL_RFHWIFG_MDMSOFT) 

We get a precise BUS FAULT error and then we crash. (We checked the fault registers and we breaked into debug into HARD_FAULT_ISR() to see the status.

After that it seems the radio inside CC1352 is dead or will never recover....

`static int
receiving_packet(void)
{

  if(!prop_radio.rf_is_on) {
    return 0;
  }

  //Really check it
  if(!rf_is_on_truely()){
      count++;
      if(count>10)
      {
          gpio_hal_arch_write_pin(GPIO_PORT, 19, 1);
      }
      prop_radio.rf_is_on = false;
      return 0;
  }

#if MAC_CONF_WITH_TSCH
  /*
   * Under TSCH operation, we rely on "hints" from the MDMSOFT interrupt
   * flag. This flag is set by the radio upon sync word detection, but it is
   * not cleared automatically by hardware. We store state in a variable after
   * first call. The assumption is that the TSCH code will keep calling us
   * until frame reception has completed, at which point we can clear MDMSOFT.
   */
  if(!is_receiving_packet) {
    /* Look for the modem synchronization word detection interrupt flag.
     * This flag is raised when the synchronization word is received.
     */
    if(HWREG(RFC_DBELL_BASE + RFC_DBELL_O_RFHWIFG) & RFC_DBELL_RFHWIFG_MDMSOFT) {
      is_receiving_packet = 1;
    }
  } else {
    /* After the start of the packet: reset the Rx flag once the channel gets clear */
    is_receiving_packet = (cca_request() == CCA_STATE_BUSY);
    if(!is_receiving_packet) {
      /* Clear the modem sync flag */
      RFCHwIntClear(RFC_DBELL_RFHWIFG_MDMSOFT);
    }
  }

  return is_receiving_packet;
#else
  /*
   * Under CSMA operation, there is no immediately straightforward logic as to
   * when it's OK to clear the MDMSOFT interrupt flag:
   *
   *   - We cannot re-use the same logic as above, since CSMA may bail out of
   *     frame TX immediately after a single call this function here. In this
   *     scenario, is_receiving_packet would remain equal to one and we would
   *     therefore erroneously signal ongoing RX in subsequent calls to this
   *     function here, even _after_ reception has completed.
   *   - We can neither clear inside read_frame() nor inside the RX frame
   *     interrupt handler (remember, we are not in poll mode under CSMA),
   *     since we risk clearing MDMSOFT after we have seen a sync word for the
   *     _next_ frame. If this happens, this function here would incorrectly
   *     return 0 during RX of this next frame.
   *
   * So to avoid a very convoluted logic of how to handle MDMSOFT, we simply
   * perform a clear channel assessment here: We interpret channel activity
   * as frame reception.
   */

  if(cca_request() == CCA_STATE_BUSY) {
    return 1;
  }

  return 0;

#endif`

It's like the radio didn't get the time to turn ON?? Or it seems the radio has crashed because of too many receives??

PLEASE HELP as we really need this working for our sensor network and we cannot form a proper network of even 20 nodes....
We are even considering a reward or some consulting for you if you can solve our problem...

Our tsch association takes 1 min plus and even longer when we reboot so we never have a stable network. If you can fix the crash or have a fast association technique. We need all nodes back online within 10 minutes maximum!!!

I can provide our entire code or more information if necessary.

YanikCysca, 2021-04-15:
Here is our full project-conf.h else we use the basic contiki configuration and MAC = TSCH.

`#ifndef __PROJECT_CONF_H__
#define __PROJECT_CONF_H__

#include "ws4-scenario.h"

/*
 * Signal to the rest of the code that we are building with OTA extensions
 * The platform will be responsible to generate an offset firmware and to
 * reserve space for metadata.
 */
#define OTA 1

/*******************************************************/
/******************* 32 KHz crystal configuration ******/
/*******************************************************/
#define SET_CCFG_MODE_CONF_SCLK_LF_OPTION            0x2        // LF XOSC : EXTERNAL CRYSTAL
//#define SET_CCFG_MODE_CONF_SCLK_LF_OPTION            0x3        // LF RCOSC : INTERNAL RC CRYSTAL

/*******************************************************/
/******************* Software configuration ************/
/*******************************************************/
// Use this flag to disable peripherals for node to save current
// such as serial port
#define BUILD_COORD 1

// Only activate peripherals for the coordinator
#if !BUILD_COORD
#define TI_UART_CONF_ENABLE 0
#define TI_NVS_CONF_NVS_EXTERNAL_ENABLE 0
#define TI_SD_CONF_ENABLE 0
#endif

// Activate I2C and External flash (NVS)
#define TI_I2C_CONF_I2C0_ENABLE 1
#define TI_NVS_CONF_ENABLE 1

// Make sure WATCHDOG is disabled else will reset
#define WATCHDOG_CONF_DISABLE 0
#define WATCHDOG_CONF_TIMEOUT_MS  3000

// Disable bootloader
#define CCFG_CONF_ROM_BOOTLOADER_ENABLE 0

//Setup heap mem size (for dynamic memory allocation)
#define HEAPMEM_CONF_ARENA_SIZE 0x2000 //4096 bytes

/*******************************************************/
/******************* Packet configuration **************/
/*******************************************************/
//Enable packet fragmentation
#define SICSLOWPAN_CONF_FRAG 1

//Max packet length.
//Minimum 116 for TSCH.
//Tests showed that messages are dropped if not at least 50+max payload
#define UIP_CONF_BUFFER_SIZE 1280  //default is 1280
//#define RF_CONF_RX_BUF_SIZE               250
//#define PACKETBUF_CONF_SIZE               250

// We use this parameter to turn on retries
#define UIP_CONF_TAG_TC_WITH_VARIABLE_RETRANSMISSIONS 1

// Increase queue lenght from 8 to 32
#define QUEUEBUF_CONF_NUM 32

/******************************************************/
/******************* Radio configuration **************/
/******************************************************/
[... 109 lines cut ...]
//#define RPL_CONF_DIO_INTERVAL_MIN                       12 // 2^(12) = 4.096 s
//#define RPL_CONF_DIO_INTERVAL_DOUBLINGS                 1  // 2^(12+2) = 16.384 s
//#define RPL_CONF_DIS_INTERVAL                           5 * CLOCK_SECOND // 5.0 s
#define RPL_CONF_DIS_INTERVAL                             5 * CLOCK_SECOND // 30.0 s
#define RPL_CONF_PROBING_INTERVAL                       15 * CLOCK_SECOND// 15.0 s
//#define RPL_CONF_DEFAULT_LIFETIME                       RPL_INFINITE_LIFETIME // Infinite duration
#define RPL_CONF_DAO_RETRANSMISSION_TIMEOUT               25 * CLOCK_SECOND //10
#define RPL_CONF_WITH_DAO_ACK                             1

#endif

/*******************************************************/
/******************* Configure Logging *****************/
/*******************************************************/
#define LOG_CONF_LEVEL_MAIN                        LOG_LEVEL_INFO
#define LOG_CONF_LEVEL_RPL                         LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_TCPIP                       LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_IPV6                        LOG_LEVEL_ERR
#define LOG_CONF_LEVEL_6LOWPAN                     LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_MAC                         LOG_LEVEL_WARN
//#define LOG_CONF_LEVEL_FRAMER                      LOG_LEVEL_NONE

//#define TI_UART_CONF_BAUD_RATE 38400

#endif /* __PROJECT_CONF_H__ */`

alexrayne, 2021-04-15:
Hallow.
code on receiving_packet(void) have some updated at cc26x0-cc13x0 target, it have some sufficient differs. Also fixes about rfcore startup. you can some compare you simplelink with it.

you can try cc26x0-cc13x0 target on cc1352 chip with this  [PR#1384](https://github.com/contiki-ng/contiki-ng/pull/1384)

YanikCysca, 2021-04-15:
We will try porting this code., but we are using CC1351P1, which has the +20 dBm output and a RF switch that switches between +20dbM for tx, the 14 dbm channel for RX and also support 2.4 Ghz for BLE. Do we simply need to add the rf switching processes to your pull request to make it work?

alexrayne, 2021-04-15:
> Do we simply need to add the rf switching processes to your pull request to make it work?

i make this port via cutouts from simplelink platform. since i not need +20dbm, i miss it. if you add this feature - it would great.
But, do you strongly demand this feature and BLE, to check out stability of radio in your net? 

> Also, will it work with alot of nodes? and hops?

i've have expirience with net about 8 nodes. for larger nets can't say anything.

YanikCysca, 2021-04-15:
We are not using BLE yet, only RF switch between 14 dbM and tx 20 dbm. It works with simplelink, but crashes after alot of nodes. seems like too much RX to handle?  Tsch and contiki are supposed to be able to handle a large number of nodes no?

alexrayne, 2021-04-16:
Logicaly any net supports large amount of nodes. But phisicaly - you share single air between all multiple nodes. TSCH a bit relax this situation due chanel-hopping, but on dence mesh you got same problem - concurence for slot leads to interference corruption of radio packets. TSCH not solve tis problem, it must solve another instrument - scheduler.

contiki provide simple solution for sheduling - `orchestra`. it works well enough when you have slots large vs nodes. but if you have about 2-3 slots/node or less, it not good. a litle amount of conflicts -that is in nature of `orchestra`, generates a great troubles on TSCH, and it stalls.
To solve conflicts in dense mesh TI provides schedule protocol MSF. there was some work started on it. and have some results demonstrated.

BUT, i guess, your problem not in this problems, but more trivial - some bugs of radio driver. You need trace your nodes, and collect a log of TSCH stack work right before hung, to see why it breaks.

kevin-forest-cysca, 2021-04-16:
To add some information,
The test was done using rpl-classic with Orchestra sender-based period 17. 
The max neighbor table  NBR_TABLE_CONF_MAX_NEIGHBORS was set to 9.

Like we said before, the problems seems to occur fast when there is a lot of nodes (when the scheduler has a lot of timeslot to schedule)

If we add a check before the register read to see if the rfcore is powered on (and skip that function). The mcu does not dies but is not able to be able to reassociate with the coordinator.

We put a led toggle on the interrupt for when the radio executes a DIV_Setup_Cmd. When the coordinator gets this error, no div setup are ever sent even if the code continues to run.

When doing a breakpoint once it happened at the prop mode on function, the RF_Schedulecmd for the rat sync and schedule_rx cmd returned with res = RF_ERROR_ALLOC.

We tried this number of nodes yesterday with the TSCH timeslot at 3 with rpl-lite and no orchestra. The 20 nodes survived, but sometimes the bug (reading the register when the core was powered off ) occurred (but it was way less often (only 3 node for 10 hours), but the node was still able to connect back to the coordinator.

It seems that we cannot sustain the 40ms timeslot or the rat-synchronisation is not going well.

@alexrayne I tried your pr#1384 , but it seems i have some issues with uniflash (the ccfg section is not able to be verified) Also, even if running it, the radio just does not seem to work. I'm testing it with a cc1352p1 launchpad. (It was my first time trying it, there must be some things that i did not do well?)

alexrayne, 2021-04-16:
as for simplelink target - [atiselsts](https://github.com/atiselsts) can help you, he provide it.

as for cc1352p1  - i worked with CC1352P chip at 430MHz band. 

there was makefile for my target chip `Makefile.cc1352`:
```
### Will allow the inclusion of the correct CPU makefile
CPU_FAMILY = cc13x2

# need by silmplelink settings defs
CFLAGS += -DDEVICE_CC1352P

### Add to the source dirs
CONTIKI_TARGET_DIRS += testboard/cc1352

SUPPORTS_PROP_MODE =1

#BOARD_SOURCEFILES += smartrf_settings.c rfprop-tx-power.c

### Include the common launchpad makefile
include testboard/Makefile.testboard 
```

another possible problem on target cc26xx - was corrupted `RADIO_DELAY_BEFORE_TX` [discussed here](https://github.com/contiki-ng/contiki-ng/issues/1330). So, you need make a big `TSCH_CONF_RX_WAIT` (more the 2-3ms) to handle this miss well.

anyway you ought to use deep tsch tracing `TSCH_LOG_CONF_PER_SLOT`  to see what happens.

kevin-forest-cysca, 2021-04-16:
I know that in simplelink platform, the TSCH_LOG_CONF_PER_SLOT  was undefined by the platform. Would there be an issue also on cc26x0?

alexrayne, 2021-04-16:
> I know that in simplelink platform, the TSCH_LOG_CONF_PER_SLOT was undefined by the platform. Would there be an issue also on cc26x0?

i use this loging havily on cc13xx. but you better use faster uart to see this log - 430kbps good.

kevin-forest-cysca, 2021-04-19:
@alexrayne I made the branch with a working cc1352p1 launchpad using the ccxx target instead of simplelink. Now only around 10 nodes are able to connect before everything breaks. Here is my branch https://github.com/kevin-forest-cysca/contiki-ng/tree/pr/cpu-cc13x2. I made the example using the 6tisch/timesync-demo

alexrayne, 2021-04-19:
is it fall for a same cause? looks now it easier to find a bug? need deep log now.

as for rebug, if you inspect branch `feature/cc13-extends` in same repo - there is a bit stuff for rf-core debug. in `rf-core.c` you find 
```
#undef DEBUG
#define DEBUG 0
#if DEBUG
#define PRINTF(...) printf(__VA_ARGS__)
#else
#define PRINTF(...)
#endif

#if 0
#define INFO(...) PRINTF(__VA_ARGS__)
#else
#define INFO(...)
#endif

```

allowing both PRINTF, and INFO you got dumping for commands to rfcore.

atiselsts, 2021-04-19:
@YanikCysca hello and sorry to her about your problems.

This PR should improve turning the radio on: https://github.com/contiki-ng/contiki-ng/pull/1515

Unfortunately I'm busy with other tasks at the moment and don't have free time to work on debugging your problem at the moment, unless you think you can offer payment for work on this issue.

alexrayne, 2021-04-19:
> This PR should improve turning the radio on: #1515

maby you answer us, for 'I know that in simplelink platform, the TSCH_LOG_CONF_PER_SLOT was undefined by the platform.' ?

atiselsts, 2021-04-19:
I don't see any code that undefines `TSCH_LOG_CONF_PER_SLOT` for the Simplelink platform. Where is this coming from?

In the Simplelink platform, it is not possible to printf from interrupt context. On `cc26x0-cc13x0`, it is possible. I find debugging `cc26x0-cc13x0` much easier because of this. But `TSCH_LOG_CONF_PER_SLOT` is logging that is not done from interrupt context.

YanikCysca, 2021-04-19:
Update to all : we seem to be able to make our code work if we turn off orchestra feature, but we have problem at reconnection. When we reset the root with 30+ nodes, some nodes never reconnect and stay stuck.
We will try to work with different architecture but doesn't look easy.

@atiselsts yes we can offer some consulting mandate if you can help us fix our problem! How can I message you privately?

kevin-forest-cysca, 2021-04-19:
> I don't see any code that undefines `TSCH_LOG_CONF_PER_SLOT` for the Simplelink platform. Where is this coming from?
>
> In the Simplelink platform, it is not possible to printf from interrupt context. [...]

Sorry it was in contiki-conf.h of arch/platform/simplelink/cc13xx-cc26xx.  BUt it was the LOG_CONF_LEVEL_FRAMER that was undefined there. 

Finally i added some TSCH_ADD_LOG in the radio module of simplelink to add some logging for it in the interrupt context. (prop-mode.c and sched.c)
It`s not the best way but it works!