contiki-ng/contiki-ng #1548: Problem with tsch_link: the first KA packet after association will never be sent if given true mac address to tsch_schedule_add_link()

wonderfulnx, 2021-04-17:
When I was writing my own TSCH scheduler, I set the `address` parameter in the `tsch_schedule_add_link()` function to the real MAC address of the neighbor instead of using a `tsch_broadcast_address` like in orchestra. Because I wanted the unicast link to be dedicated. So the code is like:
```c
// in tsch-schedule.[hc]
/* Adds a link to a slotframe, return a pointer to it (NULL if failure) */
struct tsch_link *
tsch_schedule_add_link(struct tsch_slotframe *slotframe,
                       uint8_t link_options, enum link_type link_type, const linkaddr_t *address,
                       uint16_t timeslot, uint16_t channel_offset, uint8_t do_remove)

// my schedule plan
static void new_time_source(old, new) // this is called by the call back: TSCH_CALLBACK_NEW_TIME_SOURCE()
{
    ...
    tsch_schedule_add_link(sf_unicast, LINK_OPTION_TX, LINK_TYPE_NORMAL, new_addr,
                                             timeslot_tx, channel_offset, 0);
    tsch_schedule_add_link(sf_unicast, LINK_OPTION_RX, LINK_TYPE_NORMAL, new_addr,
                                             timeslot_rx, channel_offset, 0);
    ...
}
```

The `new_time_source()` function is called immediately after receiving EB from the time source. Therefore, before the parent knows this node, the links are already added. In the `tsch_schedule_add_link()` function, there is a `tx_links_count` attribute increasing:
```c
if(l->link_options & LINK_OPTION_TX) {
  n = tsch_queue_add_nbr(&l->addr);
  /* We have a tx link to this neighbor, update counters */
  if(n != NULL) {
    n->tx_links_count++;
    if(!(l->link_options & LINK_OPTION_SHARED)) {
      n->dedicated_tx_links_count++;
    }
  }
}
```
Since the real MAC address is put to `l->addr`, `n` will be a real neighbor (coordinator for this case). Then the `tx_links_count` increases and is no longer 0. This causes the KA packet from the node never to be sent. And the reason is follows.
1) The default common rule will select the KA packet, of course.
2) When the default common link is active, it gets a packet and a neighbor in `tsch_slot_operation`, it goes
```c
/* Protothread for slot operation, called from rtimer interrupt
 * and scheduled from tsch_schedule_slot_operation */
static
PT_THREAD(tsch_slot_operation(struct rtimer *t, void *ptr))
{
    ....
    current_packet = get_packet_and_neighbor_for_link(current_link, &current_neighbor);
    ....
}
```
3) For the KA packet, the `get_packet_and_neighbor_for_link()` function finally falls into the case to pick any unicast packet in `tsch_queue_get_unicast_packet_for_any()`
4) In `tsch_queue_get_unicast_packet_for_any()`, it goes
```c
/* Returns the head packet of any neighbor queue with zero backoff counter.
 * Writes pointer to the neighbor in *n */
struct tsch_packet *
tsch_queue_get_unicast_packet_for_any(struct tsch_neighbor **n, struct tsch_link *link)
{
  if(!tsch_is_locked()) {
    struct tsch_neighbor *curr_nbr = (struct tsch_neighbor *)nbr_table_head(tsch_neighbors);
    struct tsch_packet *p = NULL;
    while(curr_nbr != NULL) {
      if(!curr_nbr->is_broadcast && curr_nbr->tx_links_count == 0) {
        /* Only look up for non-broadcast neighbors we do not have a tx link to */
        p = tsch_queue_get_packet_for_nbr(curr_nbr, link);
        if(p != NULL) {
          if(n != NULL) {
            *n = curr_nbr;
          }
          return p;
        }
      }
      curr_nbr = (struct tsch_neighbor *)nbr_table_next(tsch_neighbors, curr_nbr);
    }
  }
  return NULL;
}
```
So eventually, the condition `curr_nbr->tx_links_count == 0` stops the KA packet to be selected since `tx_links_count` is not zero for the tsch_neighbor as mentioned above.
In my experiments, the KA packets are enqueued one after another, but can never be actually sent🤕.

I hope that I've explained myself well. And I wonder if it is my wrong usage or a problem in the current implementation?

wonderfulnx, 2021-04-17:
Also, I think for the original orchestra implementation, the coordinator tsch neighbor is created but no link is added. Every time when creating a new link, the `tx_links_count` for a "broadcast neighbor" gets increased. It results in the KA packet being successfully selected since the neighbor "coordinator" has no links and is not a broadcast.
Is this understanding correct?

alexrayne, 2021-04-17:
i not understand step 3)

> For the KA packet, the get_packet_and_neighbor_for_link() function finally falls into the case to pick any unicast packet in tsch_queue_get_unicast_packet_for_any()

why it goes to `tsch_queue_get_unicast_packet_for_any()`, if slot have desired target MAC? i thought `get_packet_and_neighbor_for_link` should look desired neighbor tx-queue and take KA for that nbr. ?

wonderfulnx, 2021-04-18:
> i not understand step 3)

This is `get_packet_and_neighbor_for_link()` function:
```c
/* Get EB, broadcast or unicast packet to be sent, and target neighbor. */
static struct tsch_packet *
get_packet_and_neighbor_for_link(struct tsch_link *link, struct tsch_neighbor **target_neighbor)
{
  struct tsch_packet *p = NULL;
  struct tsch_neighbor *n = NULL;

  /* Is this a Tx link? */
  if(link->link_options & LINK_OPTION_TX) {
    /* is it for advertisement of EB? */
    if(link->link_type == LINK_TYPE_ADVERTISING || link->link_type == LINK_TYPE_ADVERTISING_ONLY) {
      /* fetch EB packets */
      n = n_eb;
      p = tsch_queue_get_packet_for_nbr(n, link);
    }
    if(link->link_type != LINK_TYPE_ADVERTISING_ONLY) {
      /* NORMAL link or no EB to send, pick a data packet */
      if(p == NULL) {
        /* Get neighbor queue associated to the link and get packet from it */
        n = tsch_queue_get_nbr(&link->addr);
        p = tsch_queue_get_packet_for_nbr(n, link);
        /* if it is a broadcast slot and there were no broadcast packets, pick any unicast packet */
        if(p == NULL && n == n_broadcast) {
          p = tsch_queue_get_unicast_packet_for_any(&n, link);
        }
      }
    }
  }
  /* return nbr (by reference) */
  if(target_neighbor != NULL) {
    *target_neighbor = n;
  }

  return p;
}
```
The type of default common rule is `LINK_TYPE_NORMAL`, so the neighbor is obtained by `n = tsch_queue_get_nbr(&link->addr);`. But the link address is `tsch_broadcast_address` in this case. Therefore KA packet can not be selected.

Also, the current orchestra implementation uses `tsch_broadcast_address` as the link address as well. So the correct neighbor **can never be** obtained by `n = tsch_queue_get_nbr(&link->addr);`. It always falls into `tsch_queue_get_unicast_packet_for_any()`.

This really confuses me.

alexrayne, 2021-04-18:
> But the link address is tsch_broadcast_address in this case. 

why there broadcast? do you create slot for KA in `new_time_source` to a broadcast adress, not unicast? 

if it so, your KA packets are must be found in `tsch_queue_get_packet_for_nbr(n, link);` for `n_broadcast`, before fall to `tsch_queue_get_unicast_packet_for_any`

wonderfulnx, 2021-04-18:
> why there broadcast? do you create slot for KA in `new_time_source` to a broadcast adress, not unicast?

KA packet are always enqueued in the time_source's neighbor queue, not the `n_broadcast`. In here:
```c
static void
keepalive_send(void *ptr)
{
  /* If not here from a timer callback, the timer must be stopped */
  ctimer_stop(&keepalive_timer);

  if(tsch_is_associated) {
    struct tsch_neighbor *n = tsch_queue_get_time_source();
    if(n != NULL) {
        linkaddr_t *destination = tsch_queue_get_nbr_address(n);
        /* Simply send an empty packet */
        packetbuf_clear();
        packetbuf_set_addr(PACKETBUF_ADDR_RECEIVER, destination);
        NETSTACK_MAC.send(keepalive_packet_sent, NULL);
        LOG_INFO("sending KA to ");
        LOG_INFO_LLADDR(destination);
        LOG_INFO_("\n");
    } else {
        LOG_ERR("no timesource - KA not sent\n");
    }
  }
}
```
So when the `&link->addr` is a broadcast, the following code in `get_packet_and_neighbor_for_link()` will never select the KA packet.
```c
        n = tsch_queue_get_nbr(&link->addr);
        p = tsch_queue_get_packet_for_nbr(n, link);
```
It gets the `n_broadcast` as `n`, but KA is not in `n_broadcast`'s queue. Therefore, finally it falls into `tsch_queue_get_unicast_packet_for_any()`

alexrayne, 2021-04-18:
OK, looks understand you case - you allocate slot to broadcast MAC, and expects that it got packets from all nbr`s, and some of that - KA. BUT, broadcast-slots only takes packets from nbrs, that have no allocated tx-slots.
somewhere your allocate tx-slot for time-source, therefore you broadcast-slot not see enqueued KA?
is that right?

wonderfulnx, 2021-04-18:
> how this MAC can be broadcast, after you say, you allocate unicast-slot for time-source. why this `link->addr` not uncast MAC of timesource, that is KA for?

Let me try to explain this more clearly. So in the existing orchestra implementation, the address of the neighbor when adding a link is given as `tsch_broadcast_address`, right? When I try to set it to the real MAC address, this problem appears. You can also try this by changing:
```c
// In orchestra-rule-unicast-link-based.c
static void
add_uc_links(const linkaddr_t *linkaddr)
{
  if(linkaddr != NULL) {
    uint16_t timeslot_rx = get_node_pair_timeslot(linkaddr, &linkaddr_node_addr);
    uint16_t timeslot_tx = get_node_pair_timeslot(&linkaddr_node_addr, linkaddr);

    /* Add Tx link */
    tsch_schedule_add_link(sf_unicast, LINK_OPTION_TX | LINK_OPTION_SHARED, LINK_TYPE_NORMAL, &tsch_broadcast_address,
                           timeslot_tx, local_channel_offset, 0);
    /* Add Rx link */
    tsch_schedule_add_link(sf_unicast, LINK_OPTION_RX, LINK_TYPE_NORMAL, &tsch_broadcast_address,
                           timeslot_rx, local_channel_offset, 0);

  }
}
```
to
```c
static void
add_uc_links(const linkaddr_t *linkaddr)
{
  if(linkaddr != NULL) {
    uint16_t timeslot_rx = get_node_pair_timeslot(linkaddr, &linkaddr_node_addr);
    uint16_t timeslot_tx = get_node_pair_timeslot(&linkaddr_node_addr, linkaddr);

    /* Add Tx link */
    tsch_schedule_add_link(sf_unicast, LINK_OPTION_TX | LINK_OPTION_SHARED, LINK_TYPE_NORMAL, linkaddr,
                           timeslot_tx, local_channel_offset, 0);
    /* Add Rx link */
    tsch_schedule_add_link(sf_unicast, LINK_OPTION_RX, LINK_TYPE_NORMAL, linkaddr,
                           timeslot_rx, local_channel_offset, 0);

  }
}
```
After that, I found that my second node (the one that tries to join the first node's network) failed to send the KA packet. It's output of KA is like this:
[image]
but the original one, the one that using `tsch_broadcast_address`, is like this:
[image]

The KA is never sent and the second node can never join the network. So I begin to find out why.

When the second node receives the EB of the first one, it immediately adds two links to the unicast slotframe. In `tsch_schedule_add_link`, `curr_nbr->tx_links_count` gets increased. The current neighbor here for the original orchestra is the `n_broadcast`, but in my case is the true neighbor, i.e. the first node.
Then the second node should send KA to the first node. But currently, the second node hasn't joined the network yet. So in the orchestra rule, this KA packet is selected by the `default common` rule. Now when the `default common` is active, it calls `get_packet_and_neighbor_for_link()` function. So the `link` input to this function is `default common`, and its `link->addr` is of course the `tsch_broadcast_address`.
However, the KA packet is enqueued and targeted only to send to the first node. So it falls to `tsch_queue_get_unicast_packet_for_any`.
Finally, `tsch_queue_get_unicast_packet_for_any` function can only get the KA packet when `curr_nbr->tx_links_count == 0`. In the original orchestra implementation, the neighbor (the first node) has a `tx_links_count` as zero. But in my changed version, it is not zero. And that makes the KA always stay in the queue.

So that's the reason I believe in this problem. All in all, I'm still confused why the orchestra always `uses tsch_broadcast_address` as the input to `tsch_schedule_add_link`.

alexrayne, 2021-04-18:
Thanks. Here my competition ends - i works misirable with orchestra. let me ask, ths moment:
> But currently, the second node hasn't joined the network yet. So in the orchestra rule, this KA packet is selected by the default common rule. 

i not understand what mean "node hasn't joined the network yet". orchestra alredy add unicast-links to first node at this moment via `add_uc_links` ? why them not works? 
i thought, orchestra manages slot-frames. and `default common` - is one, basic availiable. you add your slots to another slot-frame - for unicast rule. how default slot-frame can block slots in unicast frame? the only way, i imagine, - you allocate them with equal frame length, and in exactly one time-slot. So, default time-slot always at position of your allocated tx-slots in uncast frame.

wonderfulnx, 2021-04-18:
> i not understand what mean "node hasn't joined the network yet". orchestra alredy add unicast-links to first node at this moment via `add_uc_links` ? why them not works?

`default common` rule can not **block** the unicast frame, of course. Orchestra matches the packet to the rule in a certain order, you see, which in here should be `eb rule -> unicast link rule -> default common rule`. The KA packet simply isn't selected by the `unicast link rule`. It's like the `unicast link rule` ignores the KA packet currently, so the `default common` rule has to pick it up and match it.

And the reason for `unicast link rule` not selecting the KA, is because it checkes for `nbr_routes` for the first node's link address.
```c
// check whether to select the packet or not
if(nbr_table_get_from_lladdr(nbr_routes, (linkaddr_t *)linkaddr) != NULL) {
      return 1;
}
return 0;
```
And it seems that currently `nbr_routes` does not have the target link address.

So there is another question. Why shouldn't Orchestra match the KA packet to the `unicast rule`?
At this moment, the second node (let's call it node B) hasn't joined the network of the first node (let's call it node A). B only receives the EB packet and adds links to A. But A does not know the existence of B so it can't add links to B. Now if B wants to join the network, he should send a packet (this KA packet) to A. And since A hasn't built the links to B, he can only receive the KA packet from the `default common` slot. Therefore in B, Orchestra matches the KA packet to `default common rule`. This is actually a correct choice, I think.

So, anyways, Is there any suggestions on this problem? Is it a bug of the current implementation?

alexrayne, 2021-04-18:
supopse [atiselsts](https://github.com/atiselsts) can answer you finally, he designer of orchestra.

imho, there is no bug. you try use unicast rule in a way it not designed for. KA supposed to be broadcast adressed, and therefore it can send always - besides established connection or not. 
if you want use unicasts without connection, you should create your own rule, and place it before standart unucast-rule.

atiselsts, 2021-04-19:
I agree, this does not seem to be a bug - just undocumented behavior! If you stick with the Contiki-NG Orchestra, the keepalive packets will be sent as expected. If you design your own scheduler, it's very likely you will step on some undocumented behavior (by the way, for easier experimentation with custom schedules we have the JavaScript simulator at https://github.com/edi-riga/tsch-sim).

wonderfulnx, 2021-04-19:
> imho, there is no bug. you try use unicast rule in a way it not designed for. KA supposed to be broadcast adressed, and therefore it can send always - besides established connection or not.

> I agree, this does not seem to be a bug - just undocumented behavior!

Ok, thanks for the suggestions. So sorry to bother you again but the final question, what should I do to solve this?

- The first choice is to stick to the original implementation, always give the `tsch_broadcast_address` to the `tsch_schedule_add_link()` function as Orchestra does.

- And the second is to simply make a modification to the `tsch_queue_get_unicast_packet_for_any()` function. like to remove the `curr_nbr->tx_links_count == 0` condition.
```c
struct tsch_packet *
tsch_queue_get_unicast_packet_for_any(struct tsch_neighbor **n, struct tsch_link *link)
{
    ...
    while(curr_nbr != NULL) {
      // Original: if(!curr_nbr->is_broadcast && curr_nbr->tx_links_count == 0) {
      if(!curr_nbr->is_broadcast) {
    ...
}
```
It still confuses me. Since a true MAC address input to `tsch_schedule_add_link()` will bring this unexpected behavior, only a broadcast address can be the correct input. So what is the purpose of providing this parameter anyways?

**PS**: The intention of using the true MAC address is that I want to provide a reliable link removing. As Orchestra removes the link only by comparing `timeslots`, `channel offset` and `link options`. It seems not quite "reliable" to me.
```c
// In smart-tisch-rule-eb-per-time-source.c
static void
remove_unicast_link(uint16_t timeslot, uint16_t options)
{
  struct tsch_link *l = list_head(sf_unicast->links_list);
  while(l != NULL) {
    if(l->timeslot == timeslot
        && l->channel_offset == local_channel_offset
        && l->link_options == options) {
      tsch_schedule_remove_link(sf_unicast, l);
      break;
    }
    l = list_item_next(l);
  }
}
```

alexrayne, 2021-04-19:
>  Since a true MAC address input to tsch_schedule_add_link() will bring this unexpected behavior, only a broadcast address can be the correct input. So what is the purpose of providing this parameter anyways?

imho, just nobody tests this ssystem with multiple time-sources. only one coordinator = time-source have. so there is no problem yet.

alexrayne, 2021-04-19:
> I agree, this does not seem to be a bug - just undocumented behavior!

i`d propose you to reclassify it as a bug, since this behaviour is realy unexpected. it`s not intuitive that unicast rule depends on routing table. What if we not use any routing? imho, `nbr_routes` could be ifdef'ed as optional code, and turned on/off by configuration, and by router using?

atiselsts, 2021-04-19:
> it's not intuitive that unicast rule depends on routing table. What if we not use any routing? 

That particular rule is only compiled and linked when `UIP_MAX_ROUTES` is defined to be non-zero. If there is no routing or RPL non-storing mode is used, that particular Orchestra rule will not be present and cannot be used (will give a linker error).

atiselsts, 2021-04-19:
@wonderfulnx 

I suggest fixing the core problem, which is that Orchestra and the core TSCH code have two different notions on what an active neighbor link is. For example, you could change the condition code to 

`!curr_nbr->is_broadcast && !has_active_link(curr_nbr)`

Where `has_active_link()` is defined something like:
```
int has_active_link(struct tsch_neighbor *n)
{
   if (n->tx_links_count == 0) {
     return 0;
   }
   return neighbor_has_uc_link(&n->addr);
}
```

> The intention of using the true MAC address is that I want to provide a reliable link removing. As Orchestra removes the link only by comparing timeslots, channel offset and link options. It seems not quite "reliable" to me.

There is also a function called `tsch_schedule_remove_link` which removes the link by pointer. You can get the pointer initially, when adding the link, as `tsch_schedule_add_link` returns a pointer to the new link.

That said, I don't see a problem with removing link by the three parameters that you mention. For Orchestra, if these three parameters match, then the links are functionally identical. Again, if you design a custom schedule things are different.

wonderfulnx, 2021-04-19:
> For example, you could change the condition code to

Thank you guys very much for helping me to fix this. Now I have changed the condition code as you said and it works pretty well.😄

closed 2021-04-19 by atiselsts