contiki-ng/contiki-ng #1553: Merging RPL Border Router and RPL-UDP Server

fkerem, 2021-04-28:
Hi,

I know that rpl-udp example is a standalone one and it creates its own DODAG tree so that I can't use it with border router. 
However, I want to make it work with RPL border router instead of rpl-udp server.

I actually merge rpl-udp server code with BR's code but this time did not include the line `NETSTACK_ROUTING.root_start();` to assign the public ip address properly. 
The DODAG tree forms correctly but I can't see the neighbors and routes in web server of the BR.
I'm afraid it is not able to assign the prefix or something.

[Here is the merged BR (+ rpl-udp server) code](https://github.com/fkerem/contiki-ng/blob/develop/examples/rpl-br-udp/border-router.c)

[Here is the rpl-udp client that I use](https://github.com/fkerem/contiki-ng/blob/develop/examples/rpl-single-client/udp-client.c)

Can anyone help me to identify what is wrong? I really appreciate your help.

fkerem, 2021-04-29:
In Cooja, I see that the DODAG tree forms and the messages are being sent correctly.
However, when I test this in a real testbed, the messages are not sent.

In serial port, I see such messages:

`1619684335.581500;node-a8-103;[INFO: App       ] Not reachable yet
1619684336.299739;node-a8-101;[INFO: App       ] Not reachable yet`

fkerem, 2021-04-29:
There can be an issue with the testbed, let me check it first.

closed 2021-04-29 by fkerem

alark-mishra-hsc, 2022-03-28:
Hey @fkerem  ! I am on the same page as you were when you first commented over here. Can you please help me in combining the RPL border router and RPL UDP server.

While I have merged the code but when running it in Cooja, I am not getting logs "waiting for prefix", the log only shows the introductory line "Contiki-NG Border Router Running".

Can you please help me in figuring out a way of how I can run RPL Border router with UDP server configuration.