contiki-ng/contiki-ng #1568: Border Router with RPL_CLASSIC config assigns local IP prefixes

fkerem, 2021-05-10:
Hi,

Even though I try to assign a public IP prefix, the border router with the configuration `MAKE_ROUTING=MAKE_ROUTING_RPL_CLASSIC` assigns local prefixes. I don't have the same issue with RPL_LITE, it assigns the public prefix correctly. How can I fix this?

Here is the tunslip6 command:

`./tunslip6 2001:0660:3207:421::33/64 -v2 -L -s /dev/ttyA8_M3 -B 500000`

BR with RPL_CLASSIC:

[image]

BR with RPL_LITE:

[image]

Appreciate your help.

fkerem, 2021-05-10:
Found the same case in other issue which is closed without a proper response: contiki-ng/contiki-ng#779

fkerem, 2021-05-13:
Urgent help needed here, thanks!

tinstructor, 2021-07-12:
First of all, your BR is likely also the RPL root, which is the entity that distributes routable prefixes (such as Unique Local Address prefixes, i.e., fd00::/8, or global address prefixes, i.e., 2000::/3). Now, a node becomes root when `rpl_dag_root_start()` is called, in which the prefix to be distributed in DIOs is set via `rpl_dag_root_set_prefix()` as follows:

https://github.com/contiki-ng/contiki-ng/blob/f608483c5c32ff36807ef562486feb2bfd8ce89a/os/net/routing/rpl-classic/rpl-dag-root.c#L99-L106

Then you just follow the function calls through `rpl_dag_root_set_prefix()`, which is pretty uneventful:

https://github.com/contiki-ng/contiki-ng/blob/f608483c5c32ff36807ef562486feb2bfd8ce89a/os/net/routing/rpl-classic/rpl-dag-root.c#L87-L96

And finally you arrive at `set_global_address()`:

https://github.com/contiki-ng/contiki-ng/blob/f608483c5c32ff36807ef562486feb2bfd8ce89a/os/net/routing/rpl-classic/rpl-dag-root.c#L54-L59

If you were to look further into the default prefix to be distributed you'd see that the ULA prefix is default, which makes sense because that ensures global-like routing behavior within your organization without conflicting with the true global address space (because the routers at the border of your organization MUST NOT route ULAs). Now, I'd wager the problem lies somewhere in the difference in implementation between `rpl_dag_root_start()` in RPL-lite and RPL-classic but you'd have to look into it yourself.

nvt, 2022-04-22:
Thanks for the comments -- we have merged PR #1666.