contiki-ng/contiki-ng #1571: Trying to figure out the error for "gmake: *** No rule to make target"

atulp201, 2021-05-13:
I have done the setup as per the steps mentioned in the tutorial: https://github.com/contiki-ng/contiki-ng/wiki/Platform-simplelink
I'm too trying to run the examples (hello-world & cc26x0-web-demo), for now I'm getting stuck at build step with:
"gmake: *** No rule to make target 'hello-world.elf' "

Plus the only thing I can see is a cross at project name, no cross on either files to check for the error.
Attached the screenshot for reference. 
Kindly please guide me, to get it resolved. Thank you.

[image]

atiselsts, 2021-05-13:
Unfortunately the Contiki-NG build system has changed significantly since the tutorial was written. You should be able to build with the target equal to `hello-world` (without the `.elf`). If that does not work for some reason, try using the target name as the extension: `hello-world.cc26x0-cc13x0`.

atulp201, 2021-05-13:
> hello-world.cc26x0-cc13x0

Thank you, it works without an extension and with "hello-world.cc26x0-cc13x0"
But still it's facing the issue with hex conversion.
[image]

closed 2021-05-13 by g-oikonomou