contiki-ng/contiki-ng #1581: Simulation Script editor plugin is not working : ConstructionException: Construction error for tool of class: org.contikios.cooja.plugins.ScriptRunner

anilprajapati88, 2021-05-28:
org.contikios.cooja.Cooja$PluginConstructionException: Construction error for tool of class: org.contikios.cooja.plugins.ScriptRunner
	at org.contikios.cooja.Cooja.startPlugin(Cooja.java:1882)
	at org.contikios.cooja.Cooja.tryStartPlugin(Cooja.java:1778)
	at org.contikios.cooja.Cooja.tryStartPlugin(Cooja.java:1800)
	at org.contikios.cooja.Cooja$13$1.actionPerformed(Cooja.java:940)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:369)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1020)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1064)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6400)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
	at org.contikios.cooja.Cooja.startPlugin(Cooja.java:1862)
	... 40 more
Caused by: java.lang.ClassCastException: class javax.swing.text.AbstractDocument$DefaultDocumentEventUndoableWrapper cannot be cast to class javax.swing.text.AbstractDocument$DefaultDocumentEvent (javax.swing.text.AbstractDocument$DefaultDocumentEventUndoableWrapper and javax.swing.text.AbstractDocument$DefaultDocumentEvent are in module java.desktop of loader 'bootstrap')
	at jsyntaxpane.CompoundUndoMan.undoableEditHappened(CompoundUndoMan.java:63)
	at java.desktop/javax.swing.text.AbstractDocument.fireUndoableEditUpdate(AbstractDocument.java:293)
	at java.desktop/javax.swing.text.AbstractDocument.handleInsertString(AbstractDocument.java:761)
	at java.desktop/javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:716)
	at java.desktop/javax.swing.text.PlainDocument.insertString(PlainDocument.java:131)
	at java.desktop/javax.swing.text.DefaultEditorKit.read(DefaultEditorKit.java:274)
	at java.desktop/javax.swing.JEditorPane.setText(JEditorPane.java:1428)
	at org.contikios.cooja.plugins.ScriptRunner.updateScript(ScriptRunner.java:619)
	at org.contikios.cooja.plugins.ScriptRunner.<init>(ScriptRunner.java:276)
	... 45 more
[image]

alexrayne, 2021-05-28:
> at org.contikios.cooja.plugins.ScriptRunner.(ScriptRunner.java:276)

that is not ingormable report, if you not say - what version you run, on
wich platform, and steps to reproduce

anilprajapati88, 2021-05-29:
I am running ubantu(64 bit) version 20.04 , running open jdk 11.

alexrayne, 2021-05-31:
> I am running ubantu(64 bit) version 20.04 , running open jdk 11.

cooja not works on 64 bit yet. (year ago it was not able sure). so i have to use docker image for work with cooja.