contiki-ng/contiki-ng #1631: msp430 compiler issue

vSecDev, 2021-09-04:
Hi,

I am new to contiki-ng and looking at the examples from the repo. When I compile and run examples from the terminal, they seem to work OK, I'm getting the expected output. Within cooja, however, I'm getting the errors below. (MSP430 GCC is installed.)
I suspected it is a reference to the compiler's location and tried to fix the issue by updating the compilerpath but with no success.

`
> make nullnet-unicast.sky TARGET=sky 
make: msp430-gcc: No such file or directory
make: msp430-gcc: No such file or directory
/bin/sh: 1: msp430-gcc: not found
/bin/sh: 1: msp430-gcc: not found
/bin/sh: 1: test: -lt: unexpected operator
  MKDIR     build/sky/obj
  CC        ../../os/contiki-main.c
make: msp430-gcc: No such file or directory
make: *** [../../Makefile.include:366: build/sky/obj/contiki-main.o] Error 127
Process returned error code 2
`

Any pointers would be appreciated.

Thanks!

alexrayne, 2021-09-07:
i builds it well in contiki docker container.
can you run your gcc from command line? what pathes it installed?

atiselsts, 2021-09-07:
Check out that you have msp430-gcc in the PATH variable from the shell where you launch Cooja.

vSecDev, 2021-09-07:
> i builds it well in contiki docker container.
> can you run your gcc from command line? what pathes it installed?

It indeed works in docker. I may not have installed MSP430 correctly. The PATH variable includes /root/ti/msp430-gcc/bin. I can run cooja from the terminal OK but continue getting the error above when trying to compile the examlpes. I have found instructions for setting up the toolchain so I will start from scratch.

RockysGit, 2021-11-30:
> > 我在 contiki docker 容器中很好地构建了它。
> > 你能从命令行运行你的 gcc 吗？它安装了什么路径？
> 
> 
> > 从启动 Cooja 的 shell 中检查您在 PATH 变量中是否有 msp430-gcc。
> 
> 它确实适用于 docker。我可能没有正确安装 MSP430。PATH 变量包括 /root/ti/msp430-gcc/bin。我可以从终端运行 cooja OK 但在尝试编译示例时继续出现上述错误。我找到了设置工具链的说明，所以我将从头开始。

Have you solved your problem yet? I hope to conduct udp simulation experiments through Sky Mote, and have the same problem as you. I hope you can help！
Thankes

vSecDev, 2021-12-12:
> > > 我在 contiki docker 容器中很好地构建了它。
> > > 你能从命令行运行你的 gcc 吗？它安装了什么路径？
> > 
> > 
> > > 从启动 Cooja 的 shell 中检查您在 PATH 变量中是否有 msp430-gcc。
> > 
> > 
> > 它确实适用于 docker。我可能没有正确安装 MSP430。PATH 变量包括 /root/ti/msp430-gcc/bin。我可以从终端运行 cooja OK 但在尝试编译示例时继续出现上述错误。我找到了设置工具链的说明，所以我将从头开始。
> 
> Have you solved your problem yet? I hope to conduct udp simulation experiments through Sky Mote, and have the same problem as you. I hope you can help me！ Thanks

Hi, not yet, sorry. This one dropped off the radar but if in the meantime perhaps you can conduct the experiments using the docker container? If you do manage to figure it out, it would be much appreciated if you could post an update here. Good luck with your work!

pjonsson, 2022-05-17:
> It indeed works in docker. I may not have installed MSP430 correctly. The PATH variable includes /root/ti/msp430-gcc/bin. I can run cooja from the terminal OK but continue getting the error above [...]

Your regular user does not have permission to access things in /root.

swarth100, 2024-03-05:
Posting this here as I stumbled into the same issue today and it turned out to be a 64-bit vs. 32-bit binary on a 64-bit system.

If you're on a 64-bit system and the `msp430-gcc` binary is a 32-bit executable, you might encounter this issue if the necessary 32-bit libraries are not installed. You can check the architecture of the binary with the file command:
```
file /usr/local/bin/msp430-gcc
```

If it indicates that the binary is 32-bit and you're on a 64-bit system, you'll need to install the 32-bit libraries. On Ubuntu you can do this with:

```
sudo apt-get install libc6:i386 libncurses5:i386 libstdc++6:i386
```

It worked for me after this