contiki-ng/contiki-ng #1649: Cooja: simulating a border router

rahav, 2021-09-22:
I'm following [this tutorial ](https://github.com/contiki-ng/contiki-ng/wiki/Cooja:-simulating-a-border-router). I **started** the Serial Socket, and **started** the simulation. In the same Terminal from which I started Cooja I opened a new tab (so this is inside the same Contiki container) when I run `make TARGET=cooja connect-router-cooja` I get the following error:

    user@4ef3d8114a88:~/contiki-ng/examples/rpl-border-router_mine$ make TARGET=cooja connect-router-cooja
    ../../os/services/rpl-border-router/embedded/Makefile.embedded:13: warning: overriding recipe for target 'connect-router-cooja'
    ../../arch/platform/cooja/Makefile.cooja:34: warning: ignoring old recipe for target 'connect-router-cooja'
    sudo ../../tools/serial-io/tunslip6 -a 127.0.0.1 fd00::1/64
    client: connect: Connection refused
    tunslip6: can't connect to ``127.0.0.1:60001'': Connection refused
    ../../os/services/rpl-border-router/embedded/Makefile.embedded:13: recipe for target 'connect-router-cooja' failed
    make: *** [connect-router-cooja] Error 1

Can anyone help?

alexrayne, 2021-09-23:
tunslip says? that your cooja have no server/client 60001 port.  you should run simulation, open and START serial server on you board router serial

rahav, 2021-09-23:
I found the problem but now have a new one. Resolution is still needed. To fix the first problem I recommend we change the tutorial (I would do it but I don't have access) from:

From a terminal in the same host that runs Cooja, go to `examples/rpl-border-router`, and run:

    $ make TARGET=cooja connect-router-cooja

to the following:

To communicate with the running Cooja instance, you need to be in the same Contiki container that runs it. In the host from which you originally started the container (where you originally run the `contiker` command), run the following to find out the container id and connect to it:

    > docker ps
    CONTAINER ID   IMAGE                       COMMAND                    CREATED       STATUS       PORTS     NAMES
    7b1d7d525f71    contiker/contiki-ng   "/bin/sh -c 'bash --…"   3 hours ago   Up 3 hours                   affectionate_lalande

    > docker exec -it 7b1d7d525f71 /bin/bash

Of course, in the second command above replace the example id with your CONTAINER ID. After running the second command you will have a prompt (a Terminal) into the container in which the Cooja simulation is already running. Now, `cd` to `examples/rpl-border-router`, and run:

    > make TARGET=cooja connect-router-cooja

This does seem to connect but is starting with a warning and then fail:

../../os/services/rpl-border-router/embedded/Makefile.embedded:13: warning: overriding recipe for target 'connect-router-cooja'
../../arch/platform/cooja/Makefile.cooja:34: warning: ignoring old recipe for target 'connect-router-cooja'
sudo ../../tools/serial-io/tunslip6 -a 127.0.0.1 fd00::1/64
slip connected to ``127.0.0.1:60001''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` mtu 1500 up
ifconfig tun0 add fd00::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 172.17.0.2  netmask 255.255.255.255  destination 172.17.0.2
        inet6 fe80::1  prefixlen 64  scopeid 0x20<link>
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::8600:9296:db05:59c2  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: BR        ] Waiting for prefix
*** Address:fd00::1 => fd00:0000:0000:0000
[INFO: BR        ] Waiting for prefix
...
...
...

after many of above "Waiting for prefix" messages it will abort with the following message:

*** Address:fd00::1 => fd00:0000:0000:0000
tunslip6: slip_send overflow: Success
ifconfig tun0 down
netstat -nr | awk '{ if ($2 == "tun0") print "route delete -net "$1; }' | sh
../../os/services/rpl-border-router/embedded/Makefile.embedded:13: recipe for target 'connect-router-cooja' failed
make: *** [connect-router-cooja] Error 1

What should I do with the two warnings?
../../os/services/rpl-border-router/embedded/Makefile.embedded:13: warning: overriding recipe for target 'connect-router-cooja'
../../arch/platform/cooja/Makefile.cooja:34: warning: ignoring old recipe for target 'connect-router-cooja'

And how do I stop it from aborting?

alexrayne, 2021-09-24:
are your tunslip connected to border-router instance in simulation? your simulation have 2 nodes - router and node.
you have to check that serial socket you started, opened on router node.
if you connects right, then need debug why your router node not establish it`s IPprefix to tunslip.

if you get cooja [there ](https://github.com/alexrayne/cooja/tree/alexrayne), it have a steroids - serial-communication window provides hex-dumps on input and output streams.

rahav, 2021-09-24:
Thank you! Yes, I double checked, the Serial Socket in Cooja is added to the router node. I will check with your version of Cooja.

rahav, 2021-09-24:
@alexrayne Even before this problem, each time I would start Cooja it would display the following WARNING:

run:
     [java] 
     [java] (java:638): dbind-WARNING **: 22:14:48.159: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-BZ7oJsoFPr: Connection refused

Could this be related?

alexrayne, 2021-09-25:
>>Could this be related?
not imagine why