contiki-ng/contiki-ng #1656: No IEEE 802.14.5 and 6LoWPAN frames in pcap

dhanabalschnell, 2021-09-29:
I'm trying to analyze the packets captured via `dumpcap -i tun0` on a Raspberry Pi using wireshark, I'm not sure if I am missing any flags but I can't see any IEEE 802.15.4 or 6LoWPAN frames from the `pcap`, but I can see those frames from the `pcap` generated from `sensniff`. Has anyone ever tried to analyze the packets from `tunslip`? If so, what am I missing?

alexrayne, 2021-09-29:
there is some internal protocol used to comunicate tunslip-contiki, with slip-framing

dhanabalschnell, 2021-09-29:
Okay, in our observation the `pcap` captured through `dumpcap` like I mentioned before isn't recognizable by foren6 at all, Is it possible to capture those and analyze them with tools like foren6? Since the pcap appears to be missing IEEE 802.15.4 and 6LowPAN frames, we can't use foren6 here. Is there a tool of choice for these kinds of analysis (hop to hop timing and performance metrics)?