contiki-ng/contiki-ng #1667: Contiki Docker installation

rahav, 2021-10-15:
I'm following the standard [Contiki Docker](https://github.com/contiki-ng/contiki-ng/wiki/Docker) directions. I tried with multiple Linux versions but Docker refuses to be installed on any i386 OS:

> N: Skipping acquire of configured file 'stable/binary-i386/Packages' as repository 'https://download.docker.com/linux/ubuntu bionic InRelease' doesn't support architecture 'i386'

I've tried 32-bit OS despite [Docker official notice](https://docs.docker.com/engine/install/ubuntu/#prerequisites) that 

> To install Docker Engine, you need the 64-bit version of one of these Ubuntu versions:
Ubuntu Hirsute 21.04
Ubuntu Focal 20.04 (LTS)
Ubuntu Bionic 18.04 (LTS)

Meaning they do not support 32-bit machines.

So I tried with a 64-bit OS and I was successful installing Docker on Ubuntu 20.04. Above linked Contiki container installation completed with zero issues. However, Contiki is not stable running on this environment. In particular I believe that these problems [Cooja intermittently failing while loading #1664](https://github.com/contiki-ng/contiki-ng/discussions/1664) are caused by running Cooja on this 64-bit OS.

So I can't get the recommended Contiki container to work on 32-bit OS because Docker would not install, and I can't use Contiki on 64-bit OS. A predicament.

What am I doing wrong?

rahav, 2021-10-16:
@g-oikonomou @adamdunkels Can you please comment? In 2021, what is the recommended research/development hardware & OS environment to run Contiki?

JMartinezEco, 2022-01-14:
Any updates on this?

mbaddeley, 2022-01-14:
Cooja doesn't work on 64 bit OS versions (although there was some talk about updating this at some point) which is why it is recommended to use docker for that. I'm a bit confused however, it sounds like you were trying to install **docker itself** on a 32 bit OS? This isn't necessary. If your OS is 32-bit, then just install straight on to your machine. 

IF you have a 64-bit host machine, then install Docker with a 32/bit Ubuntu OS in order to run Cooja.

Equally, if you don't need to run Cooja, and are developing on Contiki for embedded devices, there's no need to mess around with any of this.

JMartinezEco, 2022-01-17:
I found an alternate solution. I can perfectly run Cooja on my 64 bit Windows host machine using the 64 bits image of Contiki-ng which at this day is published on Docker Hub on contiker/contiki-ng:master.

arurke, 2024-06-30:
A final note: 64-bit support for Cooja was added in [release v4.8](https://github.com/contiki-ng/contiki-ng/releases/tag/release%2Fv4.8).

closed 2024-06-30 by arurke

IntelliLab2024, 2024-11-15:
> I found an alternate solution. I can perfectly run Cooja on my 64 bit Windows host machine using the 64 bits Docker image of Contiki-ng which at this day is published on Docker Hub on contiker/contiki-ng:master.

@JMartinezEco Can you advise how I got this running? I am having trouble here
https://stackoverflow.com/questions/79187412/how-to-run-contiki-cooja-with-docker