contiki-ng/contiki-ng #1697: "examples/rpl-udp" fails to transmit packets at 30Hz

mycstudy, 2021-11-24:
hello, 
I am running "examples/rpl-udp" on TARGET=cc26x0-cc13x0 BOARD=launchpad/cc2650. cc2650 supports 2.4GHz protocols, which means cc2650 can transmit data at 250bps at the physical layer. In "examples/rpl-udp", what I did was just change the interval of etimer to CLOCK_SECOND/30, so it transmmit information using csma+rpl+udp at 30Hz. However, the server fails to receive every packet. I really don't know why the throughput of the network is so low? Due to my calculation, each packet is 32Byte, and 30Hz*(32*8bit)=7680bits/s<<250kbit/s, which means that this network should be able to transmit packets easily at 30Hz. I have been stuck in it for over one month. 
I would be really appreciate it if someone can give me some suggestion.

alexrayne, 2021-11-24:
look [this ](https://github.com/contiki-ng/contiki-ng/issues/1682)

mycstudy, 2021-11-25:
hi alexrayne, I was not using TSCH.