contiki-ng/contiki-ng #1713: Use coap_send_request() to send coap request, but cooja crash

StarUnderNight, 2021-12-28:
I want to use coap_send_request() to send coap request, and my code as follows:
```
static void test_callback(coap_callback_request_state_t *callback_state) {
  LOG_DBG_H("call back function\n");
}
```
The **root_ep** and **request** is right, don't need care them
```
if (etimer_expired(&check_et)) {
      etimer_reset(&check_et);
      const char msg[] = "sensor data!";  // 普通数据信息，发送给根节点
      coap_message_t request[1];
      coap_init_message(request, COAP_TYPE_CON, COAP_GET, 0);  // sensor data
      coap_set_header_uri_path(request, service_urls[1]);
      coap_set_payload(request, (uint8_t *)msg, strlen(msg));
      uint32_t rt = RTIMER_NOW();
      LOG_DBG_H("info time,[%ld,%ld,%ld],clock time = %ld, coap_time()=%ld,\n",check_et.timer.start, check_et.timer.interval,
       rt, clock_time(), coap_timer_uptime() );

      coap_callback_request_state_t callback_state;
      coap_send_request(&callback_state, &root_ep, request, test_callback);
    }
```
And the crash info as follows:
```
run_bigmem:
     [java]  INFO [AWT-EventQueue-0] (Cooja.java:2880) - External tools default settings: /external_tools_linux_64.config
     [java]  INFO [AWT-EventQueue-0] (Cooja.java:2910) - External tools user settings: /home/aliving/.cooja.user.properties
     [java]  INFO [AWT-EventQueue-0] (Simulation.java:422) - Simulation random seed: 123456
     [java]  INFO [AWT-EventQueue-0] (Visualizer.java:256) - Registered external visualizer: org.contikios.cooja.plugins.skins.DGRMVisualizerSkin
     [java]  INFO [AWT-EventQueue-0] (CompileContiki.java:140) - > make root-node.cooja TARGET=cooja 
     [java]  INFO [AWT-EventQueue-0] (CompileContiki.java:140) - > make sensor-node.cooja TARGET=cooja 
     [java]  INFO [Thread-4] (Simulation.java:253) - Simulation main loop started, system time: 1640614657196
     [java] WARNING: row index is bigger than sorter's row count. Most likely this is a wrong sorter usage.
     [java] #
     [java] # A fatal error has been detected by the Java Runtime Environment:
     [java] #
     [java] #  SIGSEGV (0xb) at pc=0x0000000000000048, pid=99483, tid=99875
     [java] #
     [java] # JRE version: OpenJDK Runtime Environment (11.0.13+8) (build 11.0.13+8-Ubuntu-0ubuntu1.20.04)
     [java] # Java VM: OpenJDK 64-Bit Server VM (11.0.13+8-Ubuntu-0ubuntu1.20.04, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
     [java] # Problematic frame:
     [java] # C  0x0000000000000048
     [java] #
     [java] # Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport %p %s %c %d %P %E" (or dumping to /home/aliving/program/ad-coap/tools/cooja/build/core.99483)
     [java] #
     [java] # An error report file with more information is saved as:
     [java] # /home/aliving/program/ad-coap/tools/cooja/build/hs_err_pid99483.log
     [java] #
     [java] # If you would like to submit a bug report, please visit:
     [java] #   https://bugs.launchpad.net/ubuntu/+source/openjdk-lts
     [java] # The crash happened outside the Java Virtual Machine in native code.
     [java] # See problematic frame for where to report the bug.
     [java] #
     [java] Java Result: 134

BUILD SUCCESSFUL
Total time: 19 minutes 47 seconds
```

StarUnderNight, 2021-12-28:
The log as follows
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x0000000000000048, pid=99483, tid=99875
#
# JRE version: OpenJDK Runtime Environment (11.0.13+8) (build 11.0.13+8-Ubuntu-0ubuntu1.20.04)
# Java VM: OpenJDK 64-Bit Server VM (11.0.13+8-Ubuntu-0ubuntu1.20.04, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  0x0000000000000048
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport %p %s %c %d %P %E" (or dumping to /home/aliving/program/ad-coap/tools/cooja/build/core.99483)
#
# If you would like to submit a bug report, please visit:
#   https://bugs.launchpad.net/ubuntu/+source/openjdk-lts
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: -Xmx1536m org.contikios.cooja.Cooja

Host: AMD Ryzen 5 2600 Six-Core Processor, 12 cores, 12G, Ubuntu 20.04.3 LTS
Time: Mon Dec 27 06:17:52 2021 PST elapsed time: 1183.749069 seconds (0d 0h 19m 43s)

---------------  T H R E A D  ---------------

Current thread (0x00007f66440a2000):  JavaThread "Thread-4" [_thread_in_native, id=99875, stack(0x00007f666fca3000,0x00007f666fda4000)]

Stack: [0x00007f666fca3000,0x00007f666fda4000],  sp=0x00007f66233dd1d8,  free space=18014398508227816k
Native frames: (J=compiled Java code, A=aot compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  0x0000000000000048

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
J 5774  org.contikios.cooja.corecomm.Lib2.tick()V (0 bytes) @ 0x00007f66b86dde88 [0x00007f66b86dde40+0x0000000000000048]
J 5996 c2 org.contikios.cooja.motes.AbstractWakeupMote$1.execute(J)V (9 bytes) @ 0x00007f66b87136bc [0x00007f66b8712f00+0x00000000000007bc]
j  org.contikios.cooja.Simulation.run()V+190
j  java.lang.Thread.run()V+11 java.base@11.0.13
v  ~StubRoutines::call_stub

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x0000000000000048

Register to memory mapping:

RAX=0x00007f66233dd400: <offset 0x0000000000050400> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
RBX=0x00007f66233f2b70: uip_aligned_buf+0x0000000000000030 in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
RCX=0x0 is NULL
RDX=0x0000000000000048 is an unknown value
RSP=0x00007f66233dd1d8: <offset 0x00000000000501d8> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
RBP=0x00007f66233dd210: <offset 0x0000000000050210> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
RSI=0x00007f66233c926d: <offset 0x000000000003c26d> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
RDI=0x00007f66233dd400: <offset 0x0000000000050400> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
R8 =0x0 is NULL
R9 =0x00007f66233dd050: <offset 0x0000000000050050> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
R10=0x00007f66233c926c: <offset 0x000000000003c26c> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
R11=0x00007f66233dceff: <offset 0x000000000004feff> in /home/aliving/program/ad-coap/examples/coap/coap-example-client/build/cooja/mtype375.cooja at 0x00007f662338d000
R12=0x0000000000000011 is an unknown value
R13=0x00000000a89605c0 points into unknown readable memory: 0x0000000000000001 | 01 00 00 00 00 00 00 00
R14=0x00000000a89464b0 is an oop: org.contikios.cooja.contikimote.ContikiMote 
{0x00000000a89464b0} - klass: 'org/contikios/cooja/contikimote/ContikiMote'
R15=0x00007f66440a2000 is a thread

Registers:
RAX=0x00007f66233dd400, RBX=0x00007f66233f2b70, RCX=0x0000000000000000, RDX=0x0000000000000048
RSP=0x00007f66233dd1d8, RBP=0x00007f66233dd210, RSI=0x00007f66233c926d, RDI=0x00007f66233dd400
R8 =0x0000000000000000, R9 =0x00007f66233dd050, R10=0x00007f66233c926c, R11=0x00007f66233dceff
R12=0x0000000000000011, R13=0x00000000a89605c0, R14=0x00000000a89464b0, R15=0x00007f66440a2000
RIP=0x0000000000000048, EFLAGS=0x0000000000010246, CSGSFS=0x002b000000000033, ERR=0x0000000000000014
[... 1313 lines cut ...]
CPU:total 12 (initial active 12) (3 cores per cpu, 2 threads per core) family 23 model 8 stepping 2 microcode 0x8008206, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, sse4.2, popcnt, avx, avx2, aes, clmul, mmxext, 3dnowpref, lzcnt, sse4a, ht, tsc, tscinvbit, tscinv, bmi1, bmi2, adx, fma
CPU Model and flags from /proc/cpuinfo:
model name	: AMD Ryzen 5 2600 Six-Core Processor
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl tsc_reliable nonstop_tsc cpuid extd_apicid pni pclmulqdq ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand hypervisor lahf_lm cmp_legacy extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ssbd ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap xsaveopt arat overflow_recov succor
Online cpus:
0-11
Offline cpus:
12-127
BIOS frequency limitation:
<Not Available>
Frequency switch latency (ns):
<Not Available>
Available cpu frequencies:
<Not Available>
Current governor:
<Not Available>
Core performance/turbo boost:
<Not Available>

Memory: 4k page, physical 13224624k(318608k free), swap 2097148k(2094564k free)

vm_info: OpenJDK 64-Bit Server VM (11.0.13+8-Ubuntu-0ubuntu1.20.04) for linux-amd64 JRE (11.0.13+8-Ubuntu-0ubuntu1.20.04), built on Oct 29 2021 09:11:30 by "unknown" with gcc 9.3.0

END.
```

alexrayne, 2021-12-28:
in cooja sim, crashing of node will crash cooja without any info.
to help with it i provided [coffeecatch](https://github.com/contiki-ng/contiki-ng/pull/1225) exception hook. It intercepts SIGEV and if you lack - it dump stacktrace to log.

try it with [this cooja](https://github.com/alexrayne/cooja/tree/feature/jni-exception-handle), or better with [this one](https://github.com/alexrayne/cooja/tree/alexrayne). They have a bit handling node crashing - automaticaly stops simulation on it, and prints log with dumped stacktrace

StarUnderNight, 2021-12-29:
OK, thank u ,I will try it

closed 2021-12-29 by StarUnderNight