contiki-ng/contiki-ng #1715: Openmote missing in Cooja

LeoLee0097, 2022-01-02:
Dear all
Am I missing something? I saw that Contiki-os/Contiki-ng supports the Openmote/Openmote B, but why I could not see them in the Cooja mote list?
Similar Q for Contiki-OS: https://github.com/contiki-os/contiki/issues/2663

alexrayne, 2022-01-02:
> Similar Q for Contiki-OS: contiki-os/contiki#2663

cooja uses special contiki target for it's simulations. openmote - target
on hardware platform. so, you simulate on cooja-target, and provide
binaries for hardware on according target - openmote

LeoLee0097, 2022-01-05:
@alexrayne, I appreciate your reply, it helped me a lot, and I think it's time to close this issue now.

atiselsts, 2022-01-05:
If you need ARM hardware simulations, look into Renode, it's also supported by Contiki-NG.

Otherwise follow Alex's advice.

Closing the issue now.

closed 2022-01-05 by atiselsts