contiki-ng/contiki-ng #1721: Problem with RPL options on downward rank error

tinstructor, 2022-01-06:
I've noticed that when a node is the endpoint of a downward moving packet in RPL storing mode (RPL-classic) but said node detects a Rank error, it processes the RPL option twice and will hence drop the packet instead of tolerating a single Rank error (because the first time it processes the RPL option it sets the R bit and the second time it obviously detects this bit is already set and thus drops the packet immediately). 
[image]

tinstructor, 2022-01-06:
As far as I can tell this has something to do with the following lines of code:

https://github.com/contiki-ng/contiki-ng/blob/999b0d85f28c8f70b843542e216d6d62084e8563/os/net/ipv6/uip6.c#L1204-L1214
https://github.com/contiki-ng/contiki-ng/blob/999b0d85f28c8f70b843542e216d6d62084e8563/os/net/ipv6/uip6.c#L1289-L1316

arurke, 2024-07-07:
@tinstructor Was this issue resolved by https://github.com/contiki-ng/contiki-ng/pull/2862/files?