contiki-ng/contiki-ng #1722: CC2420 ack CANNOT be parsed

LeoLee0097, 2022-01-07:
Dear all
I got an ERROR here:(. The error is CSMA: ERR - fails to parse 0. As you can see in this figure.
[image]
It seems that these error messages come from the un-parsable ACKs, where the ACK length should be five, but the length is not correct. Some ACKs can be accepted by nodes, but some cannot make them. What happened in the middle?

alexrayne, 2022-01-07:
Maybe this since ack's are reveived by all neghbours?, But only one of them
waits ack, other waits packet.

LeoLee0097, 2022-01-07:
It seems that node 1 sends something -> node 6, and node one should wait for node six's ACK.
I have already tested this kind of experiment with z1 nodes (Cooja motes do not have such a problem) via the null net - unicast.c, three similar nodes, the only differences would be the dest_link_addr, data stream: node A <- node B <- node C and they can only communicate with their neighbour. They showed the same problem ...

LeoLee0097, 2022-01-07:
> Maybe this since ack's are reveived by all neghbours?, But only one of them
> waits ack, other waits packet.

@alexrayne I do appreciate your kind help and useful responses, but I also would like to know how could I do such tests under your advice? E.g., here, I am send packet every second, what should I do to the csma.h configuration file? 

BTW, you can also take my comments under this issue as a reference as well, in the official null net example, the sending interval is 8 clock second by default, but I still experienced same issues with a similar topology when I working with Z1 mote... (@_@)?

alexrayne, 2022-01-07:
i not an expirienced user of csma. a few years ago i used it to test my
hardware with trivial pair connected.
and, as i remember, on my platform ACK was implemented by software. and it
have an statemachine as i described before - ack receives every active
neighbor, and normaly rejects parsing, since it not nomal radio-packet.

so, i,d suggests, that messages about rejecting ACK normaly rises on
receivers, that not expeects it.

LeoLee0097, 2022-01-09:
@atiselsts <-Invitation-> Dear Elsts Could you give me some suggestions?

LeoLee0097, 2022-01-09:
> I am not an experienced user of CSMA. A few years ago, I used it to test my hardware with trivial pair connected, and as I remember, ACK was implemented by software on my platform. And it has a state machine, as I described before - ack receives every active neighbour and normally rejects parsing since it is not a normal radio packet. So, I suggest that messages about rejecting ACK normally rise on receivers, not expect it. пт, 7 янв. 2022 г., 21:18 Leonard ***@***.***>:
-----
@alexrayne 
Dear Alexrayne,
Thank you for your further comments and kind helps. Please let me further understand your viewpoint here. The nodes could only communicate with their neighbour in our current Cooja simulation/emulation cases. But, e.g., Node 1 (the green one) can receive messages from Node 6 and Node 2 nearly simultaneously. Therefore, Node 1 wants an ACK from Node 6, but Node 2's messages just arrived... but if it is a message, why the `csma_security_parse_frame() < 0`...
Here comes the Code implementation segment in csma.c:
```c
// packet comes in now ...
input_packet(void)
{
  ...
  if (packetbuf_datalen() == CSMA_ACK_LEN)
  {
      /* Ignore ack packets */
      LOG_DBG("ignored ack\n");
  }
  else if (csma_security_parse_frame() < 0)
  {
      /* I do not change the security setting and use the default value here. */
      LOG_ERR("failed to parse %u\n", packetbuf_datalen());
  }
  else if (!linkaddr_cmp(packetbuf_addr(PACKETBUF_ADDR_RECEIVER),
                                     &linkaddr_node_addr) &&
              !packetbuf_holds_broadcast())
  {
       LOG_WARN("not for us\n");
  }
  ...
```

alexrayne, 2022-01-09:
i have no csma code under hand now, so i try thik on my far remembers.
Code, you have citayted, i remember, and now see that unexpected ACK
deginitely ignored.  So, your errors have some other reason.

Another problem i've got - size of node adress, not all sizes are works.
not remember why - csma ACK size fixed, or sequrity check restricted for
specified one.

more precize explanation try post on monday night.

alexrayne, 2022-01-10:
Hallow Leo. As i see in csma code: your problem looks caused by mismutch `CSMA_ACK_LEN` with received one.
looks your radio-driver appends 2 bytes more. Why it hapens? 
1) do you use hardware auto-ACK on your radio?
2) does your project builds with `CSMA_SEND_SOFT_ACK` ?

LeoLee0097, 2022-01-11:
> Hallow Leo. As i see in csma code: your problem looks caused by mismutch `CSMA_ACK_LEN` with received one.
> looks your radio-driver appends 2 bytes more. Why it hapens? 
> 1) do you use hardware auto-ACK on your radio?
> 2) does your project builds with `CSMA_SEND_SOFT_ACK` ?
> 
@alexrayne 
Dear Alexrayne
Thank you for your valuable comments and your time:). I also thought about the items you mentioned here. 

Because I am working with Z1 mote, and Z1 mote uses CC2420 to transmit its messages. If my memory is right, CC2420 has Auto-ACK functions (I need a double check here) and here we are using unicast (I mean 1 mote to mote) communication.

As for the message length of the ACK. You are right, in the radio message window, the total length of it is 5, but in the csma.c file, the length of ACK is defined as 3. _I have no idea about who did it_ and I tired to set it to 5, but I got the same issue here... 

Please notice that, I don't change anything related to the radio driver or any files in the MAC layer. I am working only based on the Nullnet example. Moreover, the current example in _[contiki-ng]/example/ nullnet/_ nullnet-unicast.c has the same issue if we use Z1 mote and do some modifications (I will post the modified version of nullnet_unicast.c for you when I arrived my class room).

As for the second item, I didn't use the CSMA_SEND_SOFT_ACK here.

Finally, I do appreciate all your comments and kindness helps! Hoping we could solve this problem here, thank you!

LeoLee0097, 2022-01-11:
@alexrayne 
Dear Alexrayne 
Sorry, I am late. I am back to you now.

## Basic Configuration

I add the following code segment to the project config file to activate the log functions in the MAC layer, which is CSMA here.

```c
/************* system configuration **************/
#ifndef PROJECT_CONF_H_
#define PROJECT_CONF_H_
// #define LOG_CONF_LEVEL_RPL LOG_LEVEL_WARN
// #define LOG_CONF_LEVEL_TCPIP LOG_LEVEL_WARN
// #define LOG_CONF_LEVEL_IPV6 LOG_LEVEL_WARN
// #define LOG_CONF_LEVEL_6LOWPAN LOG_LEVEL_WARN
#define LOG_CONF_LEVEL_MAC LOG_LEVEL_DBG
#define LOG_CONF_LEVEL_FRAMER LOG_LEVEL_WARN
#endif
```
After that, I copy and paste the `nullnet-unicast.c` 3 times and rename then to `nullnet-unicast_1.c`, `nullnet-unicast_2.c` and `nullnet-unicast_3.c`.

I keep everything unchanged except the destination address in line 55 in each file.

+ `nullnet-unicast_1.c` -> Node 1 in Cooja

```c
static linkaddr_t dest_addr = {{0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01}};
```

It is an address that does not exist in my Cooja experiments, but I want Node 1 to send something.

+ `nullnet-unicast_2.c` -> Node 2 in Cooja

```c
static linkaddr_t dest_addr = {{0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}};
```

Node 2 is the children node of Node 1. Therefore, it sends its messages to Node 1, and its destination address is Node 1's address.

+ `nullnet-unicast_3.c` -> Node 3 in Cooja

```c
static linkaddr_t dest_addr = {{0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}};
```

Node 3 is the children node of Node 2. Therefore, it sends its messages to Node 2, and its destination address is Node 2's address.

## Test

### Round 1

Line 75 in `csma.h` -> Using default value `#define CSMA_ACK_LEN 3`

[image]

### Round 2

Line 75 in `csma.h` -> Using radio message's value `#define CSMA_ACK_LEN 5`

[image]

It seems that `ACK_LEN = 5` will cause more `fail to parse 0` errors...

## Reduction (Having a guess here)

### Guess 1

In the `Radio Message Window` in Cooja, it seems that the node that runs the null net example views all the messages from other nodes as data messages. But this reason cannot explain why they do not perform re-transmissions because they do not recognize the ACKs.

[image]

In the highlighted line, `statue = 0` means the packet from Node 3 is successfully received by Node 2. After that, node 3 cleans the queue and immediately returns a `fail to parse 0` Error and an `ignored ACK`.

In `csma.c` file:

```c
/* Incoming messages ... */
...
if (packetbuf_datalen() == CSMA_ACK_LEN)
  {
    /* Ignore ack packets */
    LOG_DBG("ignored ack\n");
  }
  else if (csma_security_parse_frame() < 0)
  {
    LOG_ERR("failed to parse %u\n", packetbuf_datalen());
  }
...
```

Packet free functions `free_packet(...)` are implemented in `csma-output.c` line 296 (by default) and it was called by function `tx_done(...)` in the same file, line 341.

Notice - all the experiments are done with Z1 mote in Cooja.

### Guess 2

Based on the investigation in __Guess 1__, what if it is a scheduling problem?

alexrayne, 2022-01-11:
i not used z1 cc2420 simulations, but works with cc1350 hardware. it have no aut-ACK, i use it with CSMA_SEND_SOFT_ACK.

try it? guessed, CSMA_ACK_LEN==3 provided for soft-ACK.

As for auto-ack mode on cooja-target: i look at cooja-radio.c and see that auto-ack feature not supported - this mode suposed disabled, and not pass to cooja contiki-mote radio.

You simulates Z1-mote. i not imagine how it works with radio in cooja. here my expirience ends.

>It seems that ACK_LEN = 5 will cause more fail to parse 0 errors...

contents of auto-ACK not transparent to me. can you parse them? looks `packetbuf_datalen()` not mutch radio-packet len you see in cooja. 

Guess, that radio-driver should  ommit crc from packet to calculate `packetbuf_datalen()`, so for yoour case it could evaluates to 3, as original code expected.

alexrayne, 2022-01-11:
>Guess 1
in this sample mote3 reports about frame transmited after target mote2 answered. maybe it just not fast enough to start receive ACK after transmit? is it automaticaly starts receive after transmit or go to suspend?

alexrayne, 2022-01-11:
>Guess 1
on second view, log you screenshoted on Guess1 looks well. In `csma-output.c:send_one_packet()` you can see that packet transmits ("tx to ...." ) and waits for ack silent, and reports TX_OK. 

So, looks your ack works well.

LeoLee0097, 2022-01-12:
@alexrayne 
Dear Alexrayne
Sorry for my late reply. Yesterday was hard for me.
Here, I would like to say: "Thank you for your patience and your valuable experiences!"
Following your advice, I would like to try them one by one. Let's do it! 

I updated the `nullnet-unicast.c` code as follow:

```c
/*
 * Copyright (c) 2017, RISE SICS.
 * All rights reserved.
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in the
 *    documentation and/or other materials provided with the distribution.
 * 3. Neither the name of the Institute nor the names of its contributors
 *    may be used to endorse or promote products derived from this software
 *    without specific prior written permission.
 * THIS SOFTWARE IS PROVIDED BY THE INSTITUTE AND CONTRIBUTORS ``AS IS'' AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
 * ARE DISCLAIMED. IN NO EVENT SHALL THE INSTITUTE OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS
 * OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
 * HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY
 * OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF
 * SUCH DAMAGE.
 * This file is part of the Contiki operating system.
 */

/**
 * @file
 *         NullNet unicast example
 * @author
 *         Simon Duquennoy <simon.duquennoy@ri.se>
 */

#include "contiki.h"
#include "net/netstack.h"
#include "net/nullnet/nullnet.h"

#include <string.h>
#include <stdio.h> /* For printf() */

/* Log configuration */
#include "sys/log.h"
#define LOG_MODULE "App"
#define LOG_LEVEL LOG_LEVEL_INFO

/* Configuration */
#define SEND_INTERVAL (10 * CLOCK_SECOND)
static linkaddr_t dest_addr = {{0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00}};

/*---------------------------------------------------------------------------*/
PROCESS(nullnet_example_process, "NullNet unicast example");
AUTOSTART_PROCESSES(&nullnet_example_process);

/*---------------------------------------------------------------------------*/
void input_callback(const void *data, uint16_t len,
                    const linkaddr_t *src, const linkaddr_t *dest)
{
  if (len == sizeof(unsigned))
  {
[... 136 lines cut ...]

[image]

Take the `seqno 252` packet as an example, Node 3 received this packet but Node 4 has no idea about it. Node 4 kept sending the `seqno 252` packet to Node 3 and Node 3 kept dropping it. 

[image]

During this period, Node 4 outputs a `fail to parse 0` error, the last behaviour of Node 4 is received the ACK from Node 2 and the right behaviour should be `ignored ack` but Node 4 returns `not for us`.

```log
02:41.027 | ID:4 | [WARN: CSMA    ] not for us
02:41.029 | ID:4 | [ERR : CSMA      ] failed to parse 0
```

[image]

### Round 4

It seems that `SOFT_ACK` alone cannot fulfil its job, I enable both `ACKs` in `project-config.h` to see what happened.

[image]

Well, no error~ But it seems that no packet was transmitted successfully...

[image]

alexrayne, 2022-01-13:
> Round2 ...
> It is strange, I am using the `UDGM` radio environment and the `Tx & Rx` successful ratio are both `100`. But packets 845 and 846 turns out a `No destination` issue. Except for the initialization of the nodes, every message should be unicast to its destination...

if you look on time-stamps of that packets - thay are out on same time, and guessed jumming each other. If you look them on cooja-timeline, they highlighted by pretty pink color.

alexrayne, 2022-01-13:
> ```
> /* Disable AUTO_ACK in CC2420 */
> #define CC2420_CONF_AUTOACK 1
> ```

As for Round4 - should CC2420_CONF_AUTOACK  be 0 if used SOFT_ACK ?

LeoLee0097, 2022-01-14:
> 
> 
> > ```
> > /* Disable AUTO_ACK in CC2420 */
> > #define CC2420_CONF_AUTOACK 1
> > ```
> 
> As for Round4 - should CC2420_CONF_AUTOACK  be 0 if used SOFT_ACK ?
> 
@alexrayne 
Dear Alexrayne 
Yes, absolutely you are right. I turned the 2420 ACK off and I found that Soft ACK cannot be received. 

This issue could be caused by the two parameters, ACK waiting time and after ACK detected time defined in csma file. Unfortunately, although I can received the soft ACK, I cannot fully understand why we have these two parameters, especially the second one... Moreover, I am still suffering from the fail to parse 0 error...

I managed to print them and here they are.

[image]

[image]

[image]

alexrayne, 2022-01-15:
> I managed to print them and here they are.

that picture is disaster.

As i sayd on round3 - your ack works well with CC2420_CONF_AUTOACK, so you no need exporiments with soft-ack, and progress with your main project.

> This issue could be caused by the two parameters, ACK waiting time and after ACK detected time defined in csma file. Unfortunately, although I can received the soft ACK, I cannot fully understand [...]

i guess z1 platform is slowpoke, tharefore i`d try huge timeouts - about 10x

LeoLee0097, 2022-01-27:
> > I managed to print them, and here they are.
> 
> that picture is a disaster.
> 
> As I said in round3 - your ack works well with CC2420_CONF_AUTOACK, so you do not need experiments with soft-ack and progress with your main project.
> 
> > This issue could be caused by the two parameters, ACK waiting time and after ACK detected time defined in csma file. Unfortunately, although I can receive the soft ACK, I cannot fully understand why we have these two parameters, especially the second one... Moreover, I am still suffering from the failure to parse 0 error.
> 
> I guess the z1 platform is a slowpoke. Therefore I`d try massive timeouts - about 10x

Dear Alexrayne @alexrayne 
Thank you for your comments and suggestions. These suggestions help me greatly, and I think it is time to close and archive this issue now.

closed 2022-01-27 by LeoLee0097