contiki-ng/contiki-ng #1726: RPL lifetime is not refreshed on 868 MHz band using Zolertia Re-Mote rev-A.

fernandomaura97, 2022-01-14:
Hi all, I've tried the 'quick fix' specified in #1227 as I'm using several Re-Motes on the 868 MHz band and I was going crazy trying to change to the 868MHz band. I succesfully changed into it and the RPL seemed to work fine, but I ran into a different type of issue.
However, now I'm finding that the nodes join the RPL network, but drop out of it after their lifetime is over, and don't ever reconnect (lifetime won't update even though they share messages). 

I'm using the last version of Contiki-NG, but I added some custom libraries for implementing an ozone sensor and an i2c multigas sensor, which I placed in the /dev/os folder, if it's of any relevance here. Only two of the nodes make use of these libraries, the other 2 nodes (Temp1, Temp2) use the DHT22 library from the contiki-NG repo and they face the same issue.
The code each device is running consists of an event timer of one minute, followed by the sampling of the sensors and sending the data through the simple_udp function, same as in the rpl-udp examples. 

I will share some of the logs from my root node. Also the output of the sent messages is printed as a JSON, so it can be seen when they stop transmitting. 

```
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co2": 12.044,"no22": 0.822,"PM102": 47,"Noise2": 3908}
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co1": -3.000,"no21": -3.000,"PM101": 70,"Noise1": 192}
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:60d:b6a8, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 20, nbr count 4 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b64d  256,   320 =>   576 --  0   a   (last tx 25 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b653  256,   320 =>   576 --  0   a   (last tx 25 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b664  256,   320 =>   576 --  0   a   (last tx 25 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:616:fe8  256,   320 =>   576 --  0  ba   (last tx 25 min ago)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] links: 5 routing links in total (Periodic)
[INFO: RPL       ] links: fd00::212:4b00:60d:b6a8  (DODAG root) (lifetime: infinite)
[INFO: RPL       ] links: fd00::212:4b00:60d:b64d  to fd00::212:4b00:60d:b6a8 (lifetime: 240 seconds)
[INFO: RPL       ] links: fd00::212:4b00:60d:b664  to fd00::212:4b00:60d:b6a8 (lifetime: 240 seconds)
[INFO: RPL       ] links: fd00::212:4b00:60d:b653  to fd00::212:4b00:60d:b6a8 (lifetime: 240 seconds)
[INFO: RPL       ] links: fd00::212:4b00:616:fe8  to fd00::212:4b00:60d:b6a8 (lifetime: 240 seconds)
[INFO: RPL       ] links: end of list
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co2": 12.044,"no22": 0.840,"PM102": 47,"Noise2": 4084}
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co1": -3.000,"no21": -3.000,"PM101": 70,"Noise1": 128}
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:60d:b6a8, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 20, nbr count 4 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b64d  256,   320 =>   576 --  0   a   (last tx 26 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b653  256,   320 =>   576 --  0   a   (last tx 26 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b664  256,   320 =>   576 --  0   a   (last tx 26 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:616:fe8  256,   320 =>   576 --  0  ba   (last tx 26 min ago)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] links: 5 routing links in total (Periodic)
[INFO: RPL       ] links: fd00::212:4b00:60d:b6a8  (DODAG root) (lifetime: infinite)
[INFO: RPL       ] links: fd00::212:4b00:60d:b64d  to fd00::212:4b00:60d:b6a8 (lifetime: 180 seconds)
[INFO: RPL       ] links: fd00::212:4b00:60d:b664  to fd00::212:4b00:60d:b6a8 (lifetime: 180 seconds)
[INFO: RPL       ] links: fd00::212:4b00:60d:b653  to fd00::212:4b00:60d:b6a8 (lifetime: 180 seconds)
[INFO: RPL       ] links: fd00::212:4b00:616:fe8  to fd00::212:4b00:60d:b6a8 (lifetime: 180 seconds)
[INFO: RPL       ] links: end of list
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co2": 11.983,"no22": 0.847,"PM102": 47,"Noise2": 5648}
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co1": -3.000,"no21": -3.000,"PM101": 70,"Noise1": 380}
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:60d:b6a8, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 20, nbr count 4 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b64d  256,   320 =>   576 --  0   a   (last tx 27 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b653  256,   320 =>   576 --  0   a   (last tx 27 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b664  256,   320 =>   576 --  0   a   (last tx 27 min ago)
[INFO: RPL       ] nbr: fe80::212:4b00:616:fe8  256,   320 =>   576 --  0  ba   (last tx 27 min ago)
[INFO: RPL       ] nbr: end of list
[INFO: RPL       ] links: 5 routing links in total (Periodic)
[INFO: RPL       ] links: fd00::212:4b00:60d:b6a8  (DODAG root) (lifetime: infinite)
[INFO: RPL       ] links: fd00::212:4b00:60d:b64d  to fd00::212:4b00:60d:b6a8 (lifetime: 120 seconds)
[INFO: RPL       ] links: fd00::212:4b00:60d:b664  to fd00::212:4b00:60d:b6a8 (lifetime: 120 seconds)
[INFO: RPL       ] links: fd00::212:4b00:60d:b653  to fd00::212:4b00:60d:b6a8 (lifetime: 120 seconds)
[INFO: RPL       ] links: fd00::212:4b00:616:fe8  to fd00::212:4b00:60d:b6a8 (lifetime: 120 seconds)
[INFO: RPL       ] links: end of list
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co2": 11.983,"no22": 0.854,"PM102": 47,"Noise2": 3928}
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[... 222 lines cut ...]
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b653 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co2": 11.921,"no22": 0.902,"PM102": 47,"Noise2": 3824}
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
[INFO: RPL       ] ext hdr: packet from fd00::212:4b00:60d:b664 to fd00::212:4b00:60d:b6a8 going up, sender closer 0 (256 < 128), rank error 0, loop detected 0
{"co1": -3.000,"no21": -3.000,"PM101": 70,"Noise1": 128}
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:60d:b6a8, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 20, nbr count 4 (Periodic)
[INFO: RPL       ] nbr: fe80::212:4b00:60d:b64d  256,   320 =>   576 --  0   a   (last tx 40 min ago)
[... cut ...] ter coming the next day and leaving them running, only one re-Mote was still part of the RPL network. 
 
After trying different configurations, and verifying that I don't find these problems on the 2.4 GHz band, could I get some insight on what could cause this? 
Because of deadlines on my project, another viable option is to scrap the routing completely, and hard-code the link-layer address of the root node into every device, but I'm also having trouble finding some example of this in the wiki.

tinstructor, 2022-01-14:
> After trying different configurations, and verifying that I don't find these problems on the 2.4 GHz band, could I get some insight on what could cause this? Because of deadlines on my project, [...]

For the time being, you should probably just try RPL-classic in Storing Mode instead of going crazy :)

tinstructor, 2022-01-14:
@fernandomaura97 just add the following line to your project Makefile:
```
MAKE_ROUTING = MAKE_ROUTING_RPL_CLASSIC
```

tinstructor, 2022-01-14:
I have no experience with RPL-lite in non-storing mode but from my deep understanding of RFC 6550 my best guess is that there is something wrong with the Transit Information options that are supposed to be contained in DAO messages, as these govern the lifetime of routing table entries. It may be that DAOs are not sent, incorrectly formatted, incorrectly processed, ...

fernandomaura97, 2022-01-14:
Wow, thank you! I was actually thinking of scrapping RPL directly, but discussing with my colleague about using nullnet or another routing protocol, rpl classic seemed like a viable option! About the storing mode, I'm guessing I'd also have to explicitly define it. I left it all running on 2.3 ghz and it seemed to be working fine. Tomorrow I'll try it on the lab and update :)

tinstructor, 2022-01-14:
Storing Mode is the default mode in RPL-Classic if I'm not mistaken, so no
additional configuration should be needed

atiselsts, 2022-01-15:
> Storing Mode is the default mode in RPL-Classic 

This is correct.

The RPL Classic (or "Contiki-RPL") implementation is used in many real-world applications and as such is much more "battle tested" than the RPL lite implementation.

fernandomaura97, 2022-01-17:
Hi, thanks for the effort! I'm trying the RPL-Classic now on the same network now, but I'm unluckily still getting the same results as before. Some of the nodes stop transmitting after a while, although the root node still counts them as neighbours.  I've tried pinging the devices that stop transmitting, and they get timeouts as results... 

I'll post the terminal logs from the rpl classic approach. At the end of the log, only the "co2" node is transmitting, and any try of pinging the other devices results in timeout, which can be seen.  

```
iotlab@iotlab-HeroBox:~/Desktop/contiki-ng/examples/17-1 dht22-TS$ make login PORT=/dev/ttyUSB0
using saved target 'zoul'
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]
[INFO: RPL       ] Received a DAO from fe80::212:4b00:60d:b64d
[DBG : RPL       ] Received a (unicast) DAO with sequence number 241 from fe80::212:4b00:60d:b64d
[INFO: RPL       ] DAO lifetime: 30, prefix length: 128 prefix: fd00::212:4b00:60d:b64d
[INFO: RPL       ] Adding DAO route
[DBG : RPL       ] Neighbor state changed for fe80::212:4b00:60d:b64d, state=1
[INFO: RPL       ] Neighbor added to neighbor cache fe80::212:4b00:60d:b64d, 0012.4b00.060d.b64d
[INFO: RPL       ] Added a route to fd00::212:4b00:60d:b64d/128 via fe80::212:4b00:60d:b64d
[INFO: RPL       ] Received a DIO from fe80::212:4b00:616:fe8
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,512)
[DBG : RPL       ] Incoming DIO (dag_id, pref) = (fd00::212:4b00:60d:b6a8, 0)
[DBG : RPL       ] Incoming DIO (option, length) = (4, 14)
[INFO: RPL       ] DAG conf:dbl=8, min=12 red=10 maxinc=896 mininc=128 ocp=1 d_l=30 l_u=60
[DBG : RPL       ] Incoming DIO (option, length) = (8, 30)
[INFO: RPL       ] Copying prefix information
[DBG : RPL       ] Prefix announced in DIO
[INFO: RPL       ] Prefix set - will announce this in DIOs
[DBG : RPL       ] Neighbor state changed for fe80::212:4b00:616:fe8, state=1
[INFO: RPL       ] Neighbor added to neighbor cache fe80::212:4b00:616:fe8, 0012.4b00.0616.0fe8
[INFO: RPL       ] Received a DAO from fe80::212:4b00:616:fe8
[DBG : RPL       ] Received a (unicast) DAO with sequence number 241 from fe80::212:4b00:616:fe8
[INFO: RPL       ] DAO lifetime: 30, prefix length: 128 prefix: fd00::212:4b00:616:fe8
[INFO: RPL       ] Adding DAO route
[INFO: RPL       ] Added a route to fd00::212:4b00:616:fe8/128 via fe80::212:4b00:616:fe8
[INFO: RPL       ] Received a DIO from fe80::212:4b00:60d:b664
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,512)
[DBG : RPL       ] Incoming DIO (dag_id, pref) = (fd00::212:4b00:60d:b6a8, 0)
[DBG : RPL       ] Incoming DIO (option, length) = (4, 14)
[INFO: RPL       ] DAG conf:dbl=8, min=12 red=10 maxinc=896 mininc=128 ocp=1 d_l=30 l_u=60
[DBG : RPL       ] Incoming DIO (option, length) = (8, 30)
[INFO: RPL       ] Copying prefix information
[DBG : RPL       ] Prefix announced in DIO
[INFO: RPL       ] Prefix set - will announce this in DIOs
[DBG : RPL       ] Neighbor state changed for fe80::212:4b00:60d:b664, state=1
[INFO: RPL       ] Neighbor added to neighbor cache fe80::212:4b00:60d:b664, 0012.4b00.060d.b664
[INFO: RPL       ] Received a DIO from fe80::212:4b00:60d:b64d
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,448)
[DBG : RPL       ] Incoming DIO (dag_id, pref) = (fd00::212:4b00:60d:b6a8, 0)
[DBG : RPL       ] Incoming DIO (option, length) = (4, 14)
[INFO: RPL       ] DAG conf:dbl=8, min=12 red=10 maxinc=896 mininc=128 ocp=1 d_l=30 l_u=60
[DBG : RPL       ] Incoming DIO (option, length) = (8, 30)
[INFO: RPL       ] Copying prefix information
[DBG : RPL       ] Prefix announced in DIO
[INFO: RPL       ] Prefix set - will announce this in DIOs
[DBG : RPL       ] DIO Timer triggered
[DBG : RPL       ] Sending prefix info in DIO for fd00::
[INFO: RPL       ] Sending a multicast-DIO with rank 128
[DBG : RPL       ] Scheduling DIO timer 158 ticks in future (sent)
[DBG : RPL       ] RPL: MOP 2 OCP 1 rank 128 dioint 12, nbr count 3
[DBG : RPL       ] RPL: end of list
[INFO: RPL       ] Received a DIO from fe80::212:4b00:60d:b664
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,512)
[DBG : RPL       ] Incoming DIO (dag_id, pref) = (fd00::212:4b00:60d:b6a8, 0)
[DBG : RPL       ] Incoming DIO (option, length) = (4, 14)
[INFO: RPL       ] DAG conf:dbl=8, min=12 red=10 maxinc=896 mininc=128 ocp=1 d_l=30 l_u=60
[DBG : RPL       ] Incoming DIO (option, length) = (8, 30)
[INFO: RPL       ] Copying prefix information
[DBG : RPL       ] Prefix announced in DIO
[INFO: RPL       ] Prefix set - will announce this in DIOs
[INFO: RPL       ] Received a DIO from fe80::212:4b00:60d:b653
[DBG : RPL       ] Incoming DIO (id, ver, rank) = (30,240,512)
[DBG : RPL       ] Incoming DIO (dag_id, pref) = (fd00::212:4b00:60d:b6a8, 0)
[... 1955 lines cut ...]
{"co2": 11.738,"no22": 1.340,"PM102": 47,"Noise2": 15328}
[INFO: RPL       ] Received a DAO from fe80::212:4b00:60d:b653
[DBG : RPL       ] Received a (unicast) DAO with sequence number 247 from fe80::212:4b00:60d:b653
[INFO: RPL       ] DAO lifetime: 30, prefix length: 128 prefix: fd00::212:4b00:60d:b653
[INFO: RPL       ] Adding DAO route
[INFO: RPL       ] Added a route to fd00::212:4b00:60d:b653/128 via fe80::212:4b00:60d:b653
Timeout
#0012.4b00.060d.b6a8> fe80::212:4b00:60d:b664
Command not found. Type 'help' for a list of commands
#0012.4b00.060d.b6a8> [DBG : RPL       ] Packet going up, sender closer 0 (256 < 128)
[DBG : RPL       ] Rank OK
[... cut ...] L       ] Packet going up, sender closer 0 (256 < 128)
[DBG : RPL       ] Rank OK
{"co2": 11.738,"no22": 1.340,"PM102": 47,"Noise2": 10704}
ping fe80::212:4b00:60d:b664
Pinging fe80::212:4b00:60d:b664
[DBG : RPL       ] RPL: MOP 2 OCP 1 rank 128 dioint 20, nbr count 4
[DBG : RPL       ] RPL: end of list
Timeout
#0012.4b00.060d.b6a8> ping fe80::212:4b00:60d:b653
Pinging fe80::212:4b00:60d:b653
Received ping reply from fe80::212:4b00:60d:b653, len 4, ttl 64, delay 70 ms
#0012.4b00.060d.b6a8> 

```