contiki-ng/contiki-ng #1730: Installation Problems

grant1888, 2022-01-26:
Hi,

I am new to Linux and was trying installing Coniki-ng 4.7 and have been struggling to get it to work. (I am using Ubuntu 20.04)

When downloading the 4.7 I had to re-install the cooja files as they were empty, revert back to an older version of Java as Java 11 no longer uses XML files. 

BUILD FAILED
/home/user/contiki-ng-master/tools/cooja/build.xml:200: The following error occurred while executing this line:
/home/user/contiki-ng-master/tools/cooja/apps/serial_socket/build.xml:14: Compile failed; see the compiler error output for details.

I was also having issues when trying to download the Toolchain.
user@ubuntu:~$ sudo git clone git@github.com:contiki-ng/contiki-ng.git
[sudo] password for frank: 
Cloning into 'contiki-ng'...
The authenticity of host 'github.com (140.82.121.3)' can't be established.
ECDSA key fingerprint is SHA256:p2QAMXNIC1TJYWeIOttrVc98/R1BUFWu3/LiyKgUfQM.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'github.com,140.82.121.3' (ECDSA) to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Instead, I tried to download the docker image but was having problems with docker itself and getting this error:

user@ubuntu:~$ sudo apt-get install docker-ce
E: Invalid value set for option Signed-By regarding source https://download.docker.com/linux/ubuntu/ focal (not a fingerprint)
E: The list of sources could not be read.
E: Invalid value set for option Signed-By regarding source https://download.docker.com/linux/ubuntu/ focal (not a fingerprint)
E: The list of sources could not be read.

Any help would be appreciated.

arurke, 2022-01-26:
For the toolchain, shortest path in your case is to use https (and no sudo):
```
git clone https://github.com/contiki-ng/contiki-ng.git
cd contiki-ng
git submodule update --init --recursive
```

For docker, make sure you follow the full docker install guide at https://docs.docker.com/engine/install/ubuntu/

grant1888, 2022-01-31:
Hi @arurke thanks for the response it was very helpful.

I have installed docker and followed the install guide and created the alias in ~/.bashrc (I placed this at the bottom of the .bashrc file)

```
export CNG_PATH=</home/user/contiki-ng>
alias contiker="docker run --privileged --sysctl net.ipv6.conf.all.disable_ipv6=0 --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-uni -v /dev/bus/usb:/dev/bus/usb -ti contiker/contiki-ng"
```
 

However, when trying to run contiker I get this error:

```
user@ubuntu:~$ contiker
docker: Error response from daemon: invalid mount config for type "bind": field Source must not be empty.
See 'docker run --help'.
```

arurke, 2022-01-31:
Try without "<>":
`export CNG_PATH=/home/user/contiki-ng` (note that this should be the path to the contiki-ng folder on your computer)

grant1888, 2022-01-31:
I removed the <>

```
export CNG_PATH=/home/user/contiki-ng
alias contiker="docker run --privileged --sysctl net.ipv6.conf.all.disable_ipv6=0 --mount type=bind,source=$CNG_PATH,destination=/home/user/contiki-ng -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-uni -v /dev/bus/usb:/dev/bus/usb -ti contiker/contiki-ng"
```
However, I am still receiving the same error

arurke, 2022-01-31:
Are you certain your changes in .bashrc is in effect? Try `source ~/.bashrc` to apply and then `contiker` again.

If it still does not work, what output do you get from `type contiker`?

grant1888, 2022-01-31:
Sorry, I didn't realise there was a command to apply the changes.

It's working now but I'm not sure if I require a different version?

```
user@ubuntu:~$ contiker
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
user@33b768028e6b:~/contiki-ng$ 
```

I can navigate files, however, when trying to run cooja (/user/contiki/tools/cooja):

```
BUILD FAILED
/home/user/contiki-ng/tools/cooja/build.xml:200: The following error occurred while executing this line:
/home/user/contiki-ng/tools/cooja/apps/serial_socket/build.xml:14: Compile failed; see the compiler error output for details.
```

Thank you for your help and time

arurke, 2022-01-31:
I recently got the same warning but have not bothered investigating since things work. Feel free to update here if you find anything.

Regarding cooja - too little to go by there, and not very versed in its build process. But have you made sure you got all the files  (`git submodule update --init --recursive`), and to run I simply run `cooja` in the container (or `contiker cooja` outside of the container)

arurke, 2022-01-31:
Sorry, can't really help you there. But the idea with docker is to avoid works-on-my-computer, so I would double-check your contiki-ng repository is up to date and a sensible branch is checked out, and that you have not made any changes to the docker container. Perhaps try to clean cooja if you have played with it earlier - in the container: `cd tools/cooja; ant clean; cooja`

Perhaps you can find help in https://gitter.im/contiki-ng/Developers

PS! This issue is more [appropriately](https://github.com/contiki-ng/contiki-ng/wiki/Where-to-report-issues,-ask-questions,-etc.%3F#github-issues) hosted in the [Discussions](https://github.com/contiki-ng/contiki-ng/discussions) section. If you find a way to move it there I think the maintainers and people subscribing to notifications will be pleased.

grant1888, 2022-02-03:
Hi, I've managed to get contiki to work but it works in conitker and only if I use the command `cd tools/cooja; ant clean; cooja`. Also I'm not sure how to move posts into different sections (Issues -> Discussions)

Thank you for your help

pjonsson, 2022-11-24:
@arurke I don't think regular users can convert tickets, could you please move this to the discussions board?