contiki-ng/contiki-ng #1733: Unable to flash Openmote-b

jorritolthuis, 2022-01-31:
I tried to flash the hello-world example onto my Openmote-b using `sudo make TARGET=openmote BOARD=openmote-b PORT=/dev/ttyUSB0 hello-world.upload`.

This resulted in the error
```
../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -b 450000 -p /dev/ttyUSB0 -a 0x00202000 build/openmote/openmote-b/hello-world.bin
Opening port /dev/ttyUSB0, baud 450000
Reading data from build/openmote/openmote-b/hello-world.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
make: *** [../../arch/platform/openmote/Makefile.openmote:46: hello-world.upload] Error 1
```

I had to change the baud rate to 460800 and add `--bootloader-invert-lines` to `BSL_FLAGS` in `arch/platform/openmote/Makefile.openmote` in order to get rid of the error (and successfully flash the board).

I'm not able to determine if this is specific to my setup. Perhaps someone else could check this.

nkaramolegos, 2022-05-16:
@jorritolthuis Do you solve the problem?

jorritolthuis, 2022-05-16:
I had other issues with the Openmote-b, so I switched to the RE-Mote.

Regardless, increasing the baud rate and inverting bootloader lines were the solution to my problem. My request was mostly to verify that the changes work generally, not just on my device.

jorritolthuis, 2022-05-16:
This is the main reason I switched to the RE-Mote. I did not have a Link available.

Somehow the bootloader of the Openmote-b gets corrupted very easily. Even though the error is the same, I believe the issue described initially is different. I was never able to recover from a corrupt bootloader, but I was able to use the Openmote-b after making the changes above.

closed 2024-05-03 by nvt