contiki-ng/contiki-ng #1739: sicslowpan is not tag sent fragments

Hiihtaja77, 2022-02-02:
Hi !

I have faced the next issue : sicslowpan can send to MAC layer multiple packets in the row. If the stack is well loaded, in the buffer of MAC layer we can have fragments from several datagrams. On the MAC layer we send fragments one by one and might not receive ACK for one of fragment. As result, it doesn't have the sense to send other fragments what remains in queue becouse one fragment is failed already and other ones, even they will be received , stack discard. For avoid flooding , all of those remains fragments should be removed from MAC queue. But sislowpan is not provide any extra tag/info if this beginning of new datagram or fragment.
I think in case of error, MAC layer can discard latest fragments until new datagram starts in queue.
It help to discard useless fragments and reduce bloat .

https://github.com/contiki-ng/contiki-ng/blob/master/os/net/ipv6/sicslowpan.c#L1552

alexrayne, 2022-02-02:
imho, situation you describe - not fits in contiki net system. it starts design on micro-systems, that have no buffers for outgoing/incoming packets. With such net-layer,  application must care about buffering and deside when send rest of fragments, or when drop them.
So you have 2 ways here:
1) develop IP stack for "big-systems" way - provide features of transactions in net-buffers, that allow reject all packets of failed transaction.
2) someway update sicslowpan in contiki-way - not just flush that to net-buffers, but handles sending big datagrams during all sending time, responding on net-activity .

atiselsts, 2022-02-02:
As far as I know, the Contiki-NG implementation of 6LoWPAN fragmentation is compatible with the RFCs (I might be wrong here, but if so please point me to specification that requests this feature).

There is no doubt that the performance could be improved, both in standard-compatible and non-standard ways. Doing so looks like a research project. If there is a clear benefit demonstrated we can merge the improved approach in the main branch of the Contiki-NG code.

Or you can just do fragmentation handling at the application layer, as Alex says above.

Hiihtaja77, 2022-02-03:
If at list 2 UDP datagrams are pushed to TX queue, 20 fragments can be there. If  3rd fragment of first datagramm haven't transfered, 7 fragments can be discarded. So it have a sense if high data throughput is expected.
So everything depends how device is used. In case of short datagrams if size less than payload of one frame it no need for sure.

alexrayne, 2022-02-04:
Such a huge datagrams - is something wrong in you application net use. contiki obviously not planed to manage this, you can't just put it in net-buffers - such "memory waste-way" is contrasts to contiki.  

Normal contiki-way - you 2*big datagrams somwhere was prepared, you alredy allocated space for them. Instear of rellocate twice space for net-buffers, other protothread should slowly sends source-buffers packet-by-packet.

But if you make what you want - transaction rejection, it great net-stack improvement. I wanted do this by similar reason, for TSCH. and it looks not very difficult.

Hiihtaja77, 2022-02-04:
In theory MAC layer can analyze header of fragment and identify if this start of datagramm or next one. But this aproach brake layered design. Better if upper layer provide this info naturally.

alexrayne, 2022-02-04:
use packet_buffer attributes, some elephant can place there

Hiihtaja77, 2022-02-04:
Yes we need some guys from Contiki-ng team what able to implement it and not ruin all other components.