contiki-ng/contiki-ng #1742: RPL WITH TSCH AND ORCHESTRA NOT WORKING

jonathanns, 2022-02-04:
I'm still using Contiki-NG version 4.3 beacause i can't make a node-border router network.
The node is a simple hello word  -> In the Makefile i added the:
MAKE_MAC = MAKE_MAC_TSCH
MODULES += os/services/orchestra

The border router is also the example on the contiki and i did the same in the makefile.

Both of them should be working with de default configuration of orchestra and TSCH, but the network is never reachable.

The same network works fine in the version 4.3. Am i doing something wrong? is something missing?

jonathanns, 2022-02-04:
I used that 4.7 tagged version.

alexrayne, 2022-02-05:
cooja simulation fine, but on hardware platform - fails?

jonathanns, 2022-02-07:
did anyone test it?

atiselsts, 2022-02-07:
We will look at it.

One change to suspect if the problem is the speed is the scanning poll frequency. It seems that different chips may require different settings. 

Here is is the change that was included in 4.7:
https://github.com/contiki-ng/contiki-ng/commit/c85be85a262fe8aad00f914a0d793a48eb1cf5cf