contiki-ng/contiki-ng #1745: Effective limit of TSCH nodes in one system

ColossalTurtle, 2022-02-07:
I'm trying to simulate a 125 node in a mix of leaf and mesh, though the issue only compounds when using just mesh nodes.
Basically, in the simplest form I have nodes as per the 6tisch simple node example, except I'm using the radio defines from `contiki-ng/arch/cpu/simplelink-cc13xx-cc26xx/cc13xx-cc26xx-def.h` which changes a few things to match the sub 1ghz radio of the CC13x2 device, which has two overarching effects: a timeslot is 40ms long and the radio does not turn off within each slot, between send and ack.

My issue is, still being quite new to this, I am unsure if the limit I am seeing is by design or lack of turning the right knobs on my part.

Say we have 3 mesh nodes with 122 leafs spread around them in a random way, these nodes do nothing except let RPL do its thing, and I can never get them all to join the network, not within a reasonable length of time but since before they all join I actually start losing nodes, I assume they never will (of course simulating these in cooja). There is way too much RPL traffic which ends up having too many collisions and there just does not seem to be enough bandwidth to resolve them timely.

I did manage to fine tune some simple schedules with specific slots for groups of leafs rx, and that helps but still doesn't resolve. There is still much traffic, but reverting to the default radio timings for the cooja node will handle this traffic... easily might be a stretch but lets say effectively.

Is there a way I can keep the initial EBs so nodes join as fast as possible but, at least for the leafs, then reduce the number of outbound RPL packets? Or is this just dumb and a simple back of napkin calculation shows that I'm trying to hammer a nut here?

I did try both storing and non storing (the latter to use rpl lite), I used orchestra which failed even more spectacularly, and I tried to change so many things I don't understand that my next step is to throw it all away and start fresh, as I did learn a few things along the way, but wanted to ask if this I'm trying to do is even possible at all, and maybe if I'm missing some obvious configurable part of contiki that resolves this.

atiselsts, 2022-02-08:
It's a commonly seen requirement that I see from user, to support large (50+ node) networks with sub-Ghz radios.

In your case when you have just leaf modes, it's definitely doable. Usually the problems comes from the PHY-layer low datarate, which causes networking problems because TSCH timeslot durations are extened from 10ms to ~40ms and:
1) the RPL packet periods by default are quite short, so RPL generates *a lot* of packets
2) the schedule periods remain constant, which means there are much fewer active slots per second
3) there's a backoff mechanism in 802.15.4 that should reduce collisions, but the non-storing mode Orchestra (the default) kind of breaks that backoff mechanism

I'm thinking we should design such a basic simulation (sub-GHz, large network) with tweaked settings and add it to Contiki-NG as an example.

ColossalTurtle, 2022-02-08:
That is very much in sync with what I am seeing. When timeslots increase 4x one would expect everything to scale accordingly, but I am sure that is not the case, as time between RPL packets stays consistent. I did try changing pretty much everything I could from the configurable macros, but nothing seems to touch just that.
I am not using orchestra, and that is on purpose. First, I am learning the ropes of this and every layer I can do  without helps me understand what is underneath, but more importantly, I really don't see the reasoning for it for my use case, it should always be better with a hand tuned schedule to follow.
I do, however, understand the need for this in a more generic implementation, and I will gladly try to implement a set of orchestra rules for my use case, once I understand a bit more of the framework.

Now, point 3 is one that is close to heart; I am using `TSCH_CONF_WITH_LINK_SELECTOR` and `TSCH_CALLBACK_PACKET_READY` to implement a finer control over my schedule, same as orchestra (except I have a static set of rules to follow, of course). And in doing so, when retransmissions are required two things happen:

- The backoff window is counted per passing timeslot, so if I have, which I do, a node link that can only use one specific slot and a non trivial number of slots, there is a good chance the backoff window expires before the next natural timeslot for it, making it useless.
- When retransmissions are scheduled and the backoff window has expired, the flipside of the above is that that tx will ALWAYS happen on the same timeslot, even if there are other links on other timeslots that would satisfy all rules.

I have attempted fixes for both these points, the former by allowing the backoff window to only decrement on matching slots, but this should be carefully done, and by someone that understands the implications a bit better.
The latter is simpler in a way, all we need to do is check if the available slot would match this regardless of having been set statically by the upper layer app. However, a better solution would be to simply inquire the function that selects the channel and timeslot again, passing it an extra "nr of transmissions" arg? I am definitely inclined to try this one.

So, yeah, a lack of proper backoff handling on TSCH using the packet ready cb is definitely a good fix for allowing orchestra to more finely handle scheduling, but circling back, what would you propose be the first thing to try? I mean, getting a shared simulation is easy, but what settings should one tweak to find the sweet spot for this usecase?

I guess what I'm asking, after ranting about something kinda related but not quite the subject of this issue, how can I "sedate" RPL for use in 40ms slots? Is there a way to say "only send DIOs or DISs or whatever else you need to keep the network up every so often"?
I have messed with `TSCH_CONF_EB_PERIOD` and `TSCH_CONF_MAX_EB_PERIOD` but obviously the enhanced beacons are not the full story, as I can use it to control the speed at which network discovery happens, to a point, but stability is always still a problem.

Sorry about the long brain dump, there are still a lot of magic parts to contiki for me.

atiselsts, 2022-02-08:
> I have attempted fixes for both these points, the former by allowing the backoff window to only decrement on matching slots, but this should be carefully done, and by someone that understands the [...]

Both of these would be interesting to see. If the first one is simple to implement, we can add it to Contiki-NG.

About the second one I'm not sure what exactly you propose there.

ColossalTurtle, 2022-02-08:
I will experiment with what you shared this week, thank you.

As for the both my "fix attempts" I will recreate them and propose some code, but the second, if you look at tsch-queue.c:

```
struct tsch_packet *
tsch_queue_get_packet_for_nbr(const struct tsch_neighbor *n, struct tsch_link *link)
{
  if(!tsch_is_locked()) {
    int is_shared_link = link != NULL && link->link_options & LINK_OPTION_SHARED;
    if(n != NULL) {
      int16_t get_index = ringbufindex_peek_get(&n->tx_ringbuf);
      if(get_index != -1 &&
          !(is_shared_link && !tsch_queue_backoff_expired(n))) {    /* If this is a shared link,
                                                                    make sure the backoff has expired */
#if TSCH_WITH_LINK_SELECTOR
        int packet_attr_slotframe = queuebuf_attr(n->tx_array[get_index]->qb, PACKETBUF_ATTR_TSCH_SLOTFRAME);
        int packet_attr_timeslot = queuebuf_attr(n->tx_array[get_index]->qb, PACKETBUF_ATTR_TSCH_TIMESLOT);
        if(packet_attr_slotframe != 0xffff && packet_attr_slotframe != link->slotframe_handle) {
          return NULL;
        }
        if(packet_attr_timeslot != 0xffff && packet_attr_timeslot != link->timeslot) {
          return NULL;
        }
#endif
        return n->tx_array[get_index];
      }
    }
  }
  return NULL;
}
```

within the `TSCH_WITH_LINK_SELECTOR` section the `pack_attr_slotframe` kept the slotframe as returned by `TSCH_CALLBACK_PACKET_READY` and will reuse it on every retransmission, basically precluding use of any shared timeslot that could better match the backoff. If, instead, we would re-query `TSCH_CALLBACK_PACKET_READY` but pass it a number of transmissions attempted argument, we could allow orchestra or my static rules to say "ok, we tried the default timeslot, lets retry on the next available shared one instead" by returning 0xffff or, if we need even finer grained control, return an alternative timeslot to try based on the nr of transmissions.