contiki-ng/contiki-ng #1750: Mobility Plugin not Showing Under Tools

alifa2try, 2022-02-14:
I followed the tutorials here: http://contikiemj.blogspot.com/,  to install the mobility plugin:  https://github.com/anilprajapati88/Mobility-Plugin-for-Contiki-NG,  for Cooja.  There was no any issue at all during the installation but the Mobility Plugin fails to show under the Tools menu.

TitusAwakes, 2022-03-09:
I don't know if you managed to fix it, but I was facing the same issue with the Mobility plugin failing to load. I fixed it by editing this line of build.xml:

`<property name="cooja_jar" value="/opt/contiki-ng/tools/cooja/dist/cooja.jar"/>`
to
`<property name="cooja_jar" value="/home/{myusername}/contiki-ng/tools/cooja/dist/cooja.jar"/>`

and then running `ant jar` in the mobility plugin directory. I also added the directory to the DEFAULT_PROJECTDIRS variable in Settings -> Cooja Extensions. After that, the Mobility option in Tools showed up and the simulation worked as intended.

TitusAwakes, 2022-03-09:
I don't know if you managed to fix this, but did you check if build.xml is set up correctly? I was having the same issue and turned out that I was pointing to the wrong location in the cooja_jar property.

It worked as soon as I changed it to this:
`<property name="cooja_jar" value="/home/{myusername}/contiki-ng/tools/cooja/dist/cooja.jar"/>`

alifa2try, 2022-03-09:
I used another mobility plugin from another repo and it worked fine.

closed 2022-06-07 by atiselsts

abhiverma866, 2022-07-21:
> I used another mobility plugin from another repo and it worked fine.

From where did you download mobility plugin ? What about lo4j.Logger error ?