contiki-ng/contiki-ng #1771: Broken links on Wiki page Organization

zhitaoh, 2022-03-22:
I want to check out who currently make up the maintainers and the board on https://github.com/contiki-ng/contiki-ng/wiki/Organization

But the following two links are broken:
"Contiki-NG has a group of [maintainers](https://github.com/orgs/contiki-ng/teams/maintainers/members), a subset of which constitutes the [board](https://github.com/orgs/contiki-ng/teams/board/members)."

atiselsts, 2022-04-11:
Hi, the links work for me. If I'm logged in from my account, I get taken to the respective pages. If I'm not logged in, I got taken to Github login page.

Perhaps there is some access / visibility issue here?

zhitaoh, 2022-04-11:
I looked at them once again. Apparently the "broken" URLs are under the _teams_ directory, instead of _wiki_. That might explain why Atis you as a merge team member can access them, but I cannot. If this is supposed to be public info, then I suggest copying it to the wiki.

PR #2322 (2022-12-20): Fix broken maintainers link in doc.
Fixes #1771 . The current doc. points to the maintainers team which it seems [github does not intend to be public](https://docs.github.com/en/organizations/organizing-members-into-teams/changing-team-visibility): _"People who are not members of the organization cannot view any teams."_. This PR takes a simple approach and links to the organization members instead - which is correct as long as all members are maintainers, as today. Ping @joakimeriksson to make sure I remember the board-part correctly.
closed 2022-12-20 by atiselsts