contiki-ng/contiki-ng #1795: TSCH is taking too longo to do network association in cc2650

carolinapinho, 2022-04-18:
Hi, i am working with 2 cc2560  launchpads and trying to make them communicate via TSCH by making on mode the coordinator typing "rpl-set-root 1 " in the shell of one device and the other one, the associator, by simple flashing the hello-world example with TSCH. 
I keep waiting several minutes to the suppose association and i keep recieving some info and err from the loggin as presented above and the synchronization is not done.
On the associative node:
[INFO: TSCH      ] scanning on channel 26
[INFO: TSCH      ] scanning on channel 20
[INFO: TSCH      ] scanning on channel 15
[WARN: Frame 15.4] 15.4: for another pan 43981
[ERR : TSCH      ] ! failed to parse 21
.....
[INFO: RPL       ] sending a DIS to ff02::1a
[WARN: TSCH      ] Cannot compute max payload size: not associated
[WARN: 6LoWPAN   ] output: failed to calculate payload size - dropping packet
[INFO: TSCH      ] scanning on channel 15

in the coordinator:
[INFO: RPL       ] nbr: own state, addr fd00::212:4b00:f83:2081, DAG state: reachable, MOP 1 OCP 1 rank 128 max-rank 65535, dioint 19, nbr count 0 (Periodic)
 [INFO: RPL       ] nbr: end of list
 [INFO: RPL       ] No routing links
 [INFO: TSCH      ] TSCH: enqueue EB packet 35 16
 [INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 64, status 0, tx 1
 Hello, world
[DBG : TSCH Queue] packet is added put_index 7, packet #0x20001b30
[INFO: TSCH      ] packet sent to 0000.0000.0000.0000, seqno 72, status 0, tx 1

i read that due to the number of channels the association can take some time but i have waited almost an hour and noting so something must be wrong. I tryied this solution i read to not make the scanning random but didnt work. https://sourceforge.net/p/contiki/mailman/message/36005109/ 
I have to point out that sometimes when i am in the shell trying to ping each device the synchronization happen, so i know that the association is possible but is taking too long and i cant fix that. Can someone help me please?

arurke, 2022-04-18:
Did you by any chance change the pan id (`IEEE802154_CONF_PANID` define) on the associator? Seem it receives a beacon from 0xabcd (which is the default), but does not like it.

atiselsts, 2022-04-18:
I'm aware of a problem with the CC13xx/CC26xx settings in the last release. There was a PR that was supposed to improve the speed, but some users are reporting the opposite and we plan to revert it in the next release:
https://github.com/contiki-ng/contiki-ng/commit/c85be85a262fe8aad00f914a0d793a48eb1cf5cf

I suggest changing TSCH_CONF_ASSOCIATION_POLL_FREQUENCY back to 1000.

For testing the speed, it's best to use a hopping sequence with just one channel. That way you can be sure that the receiver is always on the right channel.

Additionally, adding busywaiting in the TSCH scanning loop has improved the speed on CC2650 nodes in the past.

atiselsts, 2022-04-18:
That said, the problem that I describe will make the scanning longer, but after 1 hour you should be associated even with these settings, I'm pretty sure. Perhaps the PANID is indeed wrong?

carolinapinho, 2022-04-18:
> IEEE802154_CONF_PANID
Thank you so much for your fast response!
I don't remember changing anything like that, how can i confirm? where is this define done?

carolinapinho, 2022-04-18:
> busywaiting in the TSCH scanning loop

Thank you for your response too!
I change the  TSCH_CONF_ASSOCIATION_POLL_FREQUENCY back to 1000 and it worked! now it is associating within seconds. I dont think i need the busy waiting !

Can i ask why did you put this parameter at 10 for this release? what was the theory behind it? And why 1000, can it be even faster?
Thank you so much for your help!

atiselsts, 2022-06-07:
Did the suggestion solve your problem?

1000 is just a number that guarantees very fast polling on all platforms during the scanning phase. The scanning etimer will be added with the max frequency. On CC2650 and most (all?) other platforms any number larger than 128 (CLOCK_SECOND) results in max possible etimer frequency.

carolinapinho, 2022-06-07:
Yes it did. Thank you!

closed 2022-06-08 by atiselsts