contiki-ng/contiki-ng #1867: hello-world on z1: no output on serial terminal

sarmasid, 2022-05-17:
Hello, 

I tried uploading the hello-world example to z1 board several times. To the best of my understanding, uploading is done successfully (blinking of onboard LED etc.). But when I login to the board, there is no output on the serial terminal. Please have a look at the terminal output below. 

I am using Ubuntu 18.04 on a Dell laptop.

```
ubuntu@ubuntu-Inspiron-7560:~/contiki-ng/examples/hello-world$ make TARGET=z1 PORT=/dev/ttyUSB0 hello-world.upload
msp430-objcopy hello-world.z1 -O ihex hello-world.ihex
cp hello-world.ihex tmpimage.ihex
/dev/ttyUSB0
make z1-reset z1-upload
make[1]: Entering directory '/home/ubuntu/contiki-ng/examples/hello-world'
make -k -j 20 z1-reset-sequence
make[2]: Entering directory '/home/ubuntu/contiki-ng/examples/hello-world'
../../tools/zolertia/z1-bsl-nopic --z1 -c /dev/ttyUSB0 -r
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Reset device ...
Done
make[2]: Leaving directory '/home/ubuntu/contiki-ng/examples/hello-world'
make -j 20 z1-upload-sequence
make[2]: Entering directory '/home/ubuntu/contiki-ng/examples/hello-world'
+++++ Erasing /dev/ttyUSB0
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Mass Erase...
Transmit default password ...
+++++ Programming /dev/ttyUSB0
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Changing baudrate to 38400 ...
Program ...
42145 bytes programmed.
+++++ Resetting /dev/ttyUSB0
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Reset device ...
Done
make[2]: Leaving directory '/home/ubuntu/contiki-ng/examples/hello-world'
make[1]: Leaving directory '/home/ubuntu/contiki-ng/examples/hello-world'
rm hello-world.ihex
```
```
ubuntu@ubuntu-Inspiron-7560:~/contiki-ng/examples/hello-world$ make TARGET=z1 PORT=/dev/ttyUSB0 login
rlwrap ../../tools/serial-io/serialdump -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 [OK]
```

atiselsts, 2022-06-01:
Just checked, works for me. Latest Contiki-NG develop, Ubuntu 20.04.4 LTS, msp430-gcc version 4.7.2 20120920 (mspgcc dev 20120911) (GCC) 

Could be a hardware problem on your side?

sarmasid, 2022-06-02:
> Could be a hardware problem on your side?

Thanks for replying. Any suggestions regarding how to ascertain if it is a hardware problem? 

FYI, it worked fine with RIoT.