contiki-ng/contiki-ng #1956: CC1312R7

blueremo, 2022-06-16:
Hi, 
I'm using the cc1312r7 but contiki-ng doesn't support the cc1312r7.

When will the cc1312r7 be supported? 

Also the core sdk is old version. 

Are there any plans to update the SDK? If right, when will it be updated?

alexrayne, 2022-06-16:
you can try this  PR #1384 , to design your chip and plaftorm.

maintaner of contiki-ng developing target for simplelink library, maybe this one can alredy handle yur chip

arurke, 2022-07-01:
hi @blueremo , I have plans to submit a PR with an updated SDK, and that should bring in support for r7 as well. Hoping I can get it done this autumn.

thomasschonbeck, 2023-07-19:
Hi Arurke, did you have time to finalize the work of porting it to r7?

arurke, 2023-07-19:
hi @thomasschonbeck. Unfortunately other things got pushed in front of it, but it is not abandoned in any sense. Part of the work is now being done at my employer so this might impact the timing as to when I can open-source it.

JMartinezEco, 2024-09-26:
Any news on this?

arurke, 2024-09-27:
Unfortunately I found the work needed to get the port suitable for the public contiki-ng repo. is quite significant, and I have simply not found time for it. There is some movement on the SDK (which needs updating), see issues and pull requests in  https://github.com/contiki-ng/coresdk_cc13xx_cc26xx.

micw, 2025-04-30:
Hello,
I recently started from scratch with the simplelink lowpower SDK (https://github.com/TexasInstruments/simplelink-lowpower-f2-sdk) which is the latest SDK from TI and which comes with all the licensing issues removed.

Making contiki-ng work with it was quite straightforward. I just copied the existing files from the existing implementation for simplelink CC1312 and did only minor modifications. Not all stuff ported yet but the radio works.

It absolutely makes sense to integrate syscfg for generation of the ti specific files. Also gpio-hal-arch needs to be implemented since the GPIO stuff from TI has changed from `PIN.h` and `PINCC26XX.h` to `GPIO.h`. But that's straightforward, it's just on/off/toggle.

Unfortunately the implementation I made is quite project specific (no CPU, only the target platform that contains everything). So nothing that can go directly to contiki-ng.

But nevertheless, I'd say it should be absolutely doable to integrate the new SDK and CPU/Platforms.