contiki-ng/contiki-ng #1963: sky led colour is wrong in arch/platform/sky/sky-dev.h

jbrusey, 2022-06-17:
The Telos Mote rev b has 3 leds: red, green and blue, however, in `arch/platform/sky/sky-dev.h`, we have:
```
#define LEDS_CONF_RED    0x10
#define LEDS_CONF_GREEN  0x20
#define LEDS_CONF_YELLOW 0x40
```
This results in `LEDS_BLUE` containing zero, which is counterintuitive and means that trying to turn on the blue LED will have no effect. Before I put in a pull request - just want to check that I haven't gotten the wrong end of the stick.

jbrusey, 2023-03-22:
@pjonsson This is affecting more users than just me. Can you suggest an appropriate way to fix it? Should there be LEDS_CONF_BLUE set to 0x40 in addition to LEDS_CONF_YELLOW?

pjonsson, 2023-05-10:
This hardware predates my involvement with Contiki-NG, but make the pull request, it's always easier to get people to comment about concrete code.

jbrusey, 2023-05-10:
Thanks. Will do when less busy.

closed 2023-05-10 by jbrusey