contiki-ng/contiki-ng #2117: Openmote-B cannot be flashed with latest Docker image

LeoLee0097, 2022-08-04:
Dear all
I am now playing with Openmote-B in the latest Docker image running on my Ubuntu 22.04 laptop and Ubuntu 20.04 workstation; I tried to start with the simplest `Hello-world` example but encountered with the following Errors

```bash
  CC        ../../arch/cpu/cc2538/./ieee-addr.c
  CC        ../../arch/cpu/cc2538/cc2538.lds
  LD        build/openmote/openmote-b/hello-world.elf
  OBJCOPY   build/openmote/openmote-b/hello-world.elf --> build/openmote/openmote-b/hello-world.bin
../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -b 450000 -a 0x00202000 build/openmote/openmote-b/hello-world.bin
Opening port /dev/ttyUSB0, baud 450000
Reading data from build/openmote/openmote-b/hello-world.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
make: *** [../../arch/platform/openmote/Makefile.openmote:44: hello-world.upload] Error 1
```

As can be obtained here, the `hello-world.c` file is successfully compiled. But, when I tried to upload it to my Openmote-B, I went to an error -- `Connecting to target... ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'`. 

Here comes my bash input:
```bash
$ make TARGET=openmote BOARD=openmote-b PORT=/dev/ttyUSB0 hello-world.upload
```

I also tried with this <https://github.com/electrolama/zig-a-zig-ah/issues/22> but it did not work.

Best wishes
Leo

pjonsson, 2022-08-05:
Which Docker image was the last one that worked for you?

LeoLee0097, 2022-08-07:
> Which Docker image was the last one that worked for you?

Hi [pjonsson](https://github.com/pjonsson)

Thank you for your time!

## Solution (temporary?)

I got a temporary solution from issue #1600 . According to issue #1600 , the following files in `Contiki-NG`, note that I am working on ***Contiki-NG Ver 4.8 development branch with the latest Docker image (Ubuntu 20.04 ver)***, need to be refined:

- ***Make file*** `Makefile.include`
```Makefile
ifneq ($(BOARD),)
 @echo >>Makefile.target "BOARD = $(BOARD)"
endif

ifneq ($(BOARD_REVISION),)
 @echo >>Makefile.target "BOARD_REVISION = $(BOARD_REVISION)"
endif
...
VIEWCONF = $(CONTIKI_NG_TOOLS_DIR)/viewconf/viewconf.c
viewconf:
 @echo "----------------- Make variables: --------------"
 @echo "##### \"TARGET\": ________________________________ $(TARGET)"
 @echo "##### \"BOARD\": _________________________________ $(BOARD)"
 @echo "##### \"BOARD_REVISION\": ________________________ $(BOARD_REVISION)"
 @echo "##### \"MAKE_MAC\": ______________________________ $(MAKE_MAC)"
 @echo "##### \"MAKE_NET\": ______________________________ $(MAKE_NET)"
 @echo "##### \"MAKE_ROUTING\": __________________________ $(MAKE_ROUTING)"

...
```

- ***Make file*** `arch/platform/openmote/Makefile.openmote`
```Makefile
# BSL_FLAGS += -e -w -v -b 450000
ifndef PORT
 ifndef MOTE
  MOTE = 1
 endif
 ifeq ($(BOARD),openmote-b)
  INDEX = $(shell echo $(MOTE)*2-1 | bc)
 else
  INDEX = $(shell echo $(MOTE)-1 | bc)
 endif
 PORT=/dev/ttyUSB$(INDEX)
endif

BSL_FLAGS += -e -w -v -b 460800

ifeq ($(BOARD),openmote-b)
 BSL_FLAGS += --bootloader-invert-lines
endif
```

- ***Board configuration*** `arch/platform/openmote/openmote-b/board.c`
```c
#include "dev/gpio.h"
#include "dev/ioc.h"
...
/*---------------------------------------------------------------------------*/
// static void
// configure_unused_pins(void)
// {
//   /* FIXME */
// }
/*---------------------------------------------------------------------------*/
void
board_init()
{
  antenna_init();
  // configure_unused_pins();

  /* configure bootloader pin as input */
  GPIO_SOFTWARE_CONTROL(GPIO_PORT_TO_BASE(GPIO_A_NUM),
      GPIO_PIN_MASK(FLASH_CCA_CONF_BOOTLDR_BACKDOOR_PORT_A_PIN));
  GPIO_SET_INPUT(GPIO_PORT_TO_BASE(GPIO_A_NUM),
      GPIO_PIN_MASK(FLASH_CCA_CONF_BOOTLDR_BACKDOOR_PORT_A_PIN));
  ioc_set_over(GPIO_A_NUM,
      FLASH_CCA_CONF_BOOTLDR_BACKDOOR_PORT_A_PIN,
      IOC_OVERRIDE_ANA);
}
```

- ***Board configuration*** `arch/platform/openmote/openmote-b/board.h`
```h
...
/*---------------------------------------------------------------------------*/
/**
 * \name Serial Boot Loader Backdoor configuration
 * @{
 */
// #ifndef FLASH_CCA_CONF_BOOTLDR_BACKDOOR_PORT_A_PIN
// #define FLASH_CCA_CONF_BOOTLDR_BACKDOOR_PORT_A_PIN 7 /**< Pin PA7 activates the boot loader */
// #endif

// #ifndef FLASH_CCA_CONF_BOOTLDR_BACKDOOR_ACTIVE_HIGH
// #define FLASH_CCA_CONF_BOOTLDR_BACKDOOR_ACTIVE_HIGH 0 /**< A logic low level activates the boot loader */
// #endif
/** @} */
/*---------------------------------------------------------------------------*/
...
// #define ANTENNA_BSP_RADIO_INT    GPIO_PIN_MASK(3)
#define ANTENNA_BSP_RADIO_INT    GPIO_PIN_MASK(4)
#define ANTENNA_BSP_RADIO_EXT    GPIO_PIN_MASK(4)
```

The following files in `Contiki-NG` need to be initialized:

- ***OpenMote RPL-service configuration*** `os/services/rpl-border-router/openmote/module-macros.h`

```h
/* SLIP does not work otherwise */
#define LPM_CONF_MAX_PM 0
#define LPM_CONF_ENABLE 0
```

When uploading the program to Openmote B, we need to:

```bash
make TARGET=openmote BOARD=openmote-b PORT=/dev/ttyUSB1 savetarget
make hello-world.upload PORT=/dev/ttyUSB1
make login PORT=/dev/ttyUSB1
```

An Openmote B has two serial interface, but `PORT=/dev/ttyUSB0` is ***always not accessible***, I have to use `PORT=/dev/ttyUSB1` instead. Moreover, it seems that `savetarget` is important. Otherwise, the Openmote B (sometimes) will not be able to be flashed.

This solution works for Openmote B, whose Chip *Atmel* has code *TH* rather than code *TW*.

## Limitations of Current Solutions in issue #1600

This solution is not feasible for the Openmote B board equipped with Chip *Atmel 86RF 215 -ZU 1906D TW 1906D3T*.

[image]

I can successfully log in to those boards with code *TW* on *Atmel* chip and see hello-world outputs. But I can not flash any new program to them for further tests.

pjonsson, 2022-08-07:
The linked picture doesn't display in Firefox/Chromium.

Even so, it sounds like the issue has more to do with the Openmote platform in Contiki-NG and is unrelated to the Docker image.

LeoLee0097, 2022-08-10:
> The linked picture doesn't display in Firefox/Chromium.
> 
> Even so, it sounds like the issue has more to do with the Openmote platform in Contiki-NG and is unrelated to the Docker image.

Hi @pjonsson 

Thank you for your comments. I have updated the title, and the missing figure is not a big deal. It is a picture of the openmote-b board with the chip _Atmel 86RF 215 -ZU 1906D TW 1906D3T_ I have.

More references are listed here:
https://github.com/RIOT-OS/RIOT/issues/11031
https://gist.github.com/kkrentz/18ce317d0a1db331ccc38be6c7e0ac9e

The problem is that the solutions listed here did not work for me, except for the JTAG method. I will try the JTAG method later.

kkrentz, 2022-09-27:
@LeoLee0097 Does it work with RIOT-OS? Ensure that PA7 is grounded.

tinstructor, 2022-11-30:
@kkrentz @LeoLee0097 I seem to remember having [a similar issue in RIOT](https://github.com/RIOT-OS/RIOT/issues/13044) as well. I don't know if it has been fixed since because I moved on to using Zolertia Remote with Contiki-NG entirely instead of OpenMote-B with RIOT-OS. Anyhow, JTAG should always work