contiki-ng/contiki-ng #2185: Native and Cooja compiler not the same

rahav, 2022-09-22:
I noticed that compilations errors caught when compiling with `TARGET=native` will not be reported when the code is compiler by Cooja. Should they be the same (or run with the same parameters)?

alexrayne, 2022-09-22:
No, their not the same - cooja is a platform on java natve mode environment. That environment armed with specific interfaces - device drivers simulators. Native - loks as a generin aplication on some OS.

pjonsson, 2022-10-28:
The build system was unified in v4.8 (and beyond), so the same compiler should be used. Cooja and native are separate targets though, so the target-specific code is different.

This isn't a bug but the information might be useful for someone else, so I think this should be moved to the discussion forum. I don't think the reporter has permission to move their own tickets to the forum though, so let's hope someone else moves this for us.