contiki-ng/contiki-ng #2206: Could not run cooja

Sarra1990, 2022-10-16:
When I try to run Cooja using ant run, the system gave errors telling that there is no build file. What is the command that I could use to run it?

pjonsson, 2022-10-16:
If you're using the docker image, you can run it with `cooja`. Otherwise, `./gradlew run` in the Cooja directory will build and run it.

Sarra1990, 2022-10-17:
> If you're using the docker image, you can run it with `cooja`. Otherwise, `./gradlew run` in the Cooja directory will build and run it.

Thank you for your answer. I will try it.

Sarra1990, 2022-10-19:
> If you're using the docker image, you can run it with `cooja`. Otherwise, `./gradlew run` in the Cooja directory will build and run it.

When, I do ./gradlew run. I have build failed. Do you have any idea on the origin of the issue?

pjonsson, 2022-10-19:
Do you have a Java 17 compiler installed on the machine?

Gradle gives some information about how to turn on more verbose modes when the build fails, try those and see what it says.

Sarra1990, 2022-10-19:
> Do you have a Java 17 compiler installed on the machine?
> I have java 8 
> Gradle gives some information about how to turn on more verbose modes when the build fails, try those and see what it says.
Thank you for your answer, I will try to see them.

pjonsson, 2022-10-19:
Contiki-NG v4.8 used Java 11 for Cooja, so if you want to run Java 8, you need Contiki-NG v4.7. The Contiki-NG v4.7 Cooja will use ant as its build system, so ant run will work.

Sarra1990, 2022-10-19:
> Contiki-NG v4.8 used Java 11 for Cooja, so if you want to run Java 8, you need Contiki-NG v4.7. The Contiki-NG v4.7 Cooja will use ant as its build system, so ant run will work.

Thank you very much. You helped me a lot.

closed 2022-10-27 by nvt