contiki-ng/contiki-ng #2217: Sending data from cooja mote to python script

Sarra1990, 2022-10-22:
I am a beginner with cooja simulator. I have a sky mote that generates data randomly. This data must be sent to a blockchain node that will run a smart contract to extract some information from this data. Is there a solution to link between cooja simulation and python. In other words, how to send this data from cooja contiki to a python script?

alexrayne, 2022-10-23:
look at serversocket\clientsocket usage in cooja learning manual. that
plugins connects mote uart to host socket. in learning examples sockets
connects to host comm, but i used it to link python script

Sarra1990, 2022-10-23:
> look at serversocket\clientsocket usage in cooja learning manual. that plugins connects mote uart to host socket. in learning examples sockets connects to host comm, but i used it to link python script вс, 23 окт. 2022 г., 00:54 Sarra1990 ***@***.***>:
> […](#)

Thank you for your answer. I will look for it.

closed 2022-10-27 by nvt