contiki-ng/contiki-ng #2229: Lot of errors while flashing to open mote b

divya-1498, 2022-10-25:
Hi, I am new to open mote b and C++. I was trying to flash a sniffer program onto the board when I encountered these errors 

`Compiling          projects\ieee802154-sniffer\src\main.cpp
projects\ieee802154-sniffer\src\main.cpp:61:22: error: 'uart' was not declared in this scope; did you mean 'uart1'?
   61 | static Serial serial(uart);
      |                      ^~~~
      |                      uart1
projects\ieee802154-sniffer\src\main.cpp: In function 'int main()':
projects\ieee802154-sniffer\src\main.cpp:76:5: error: 'spi' was not declared in this scope; did you mean 'spi1'?
   76 |     spi.enable();
      |     ^~~
      |     spi1
projects\ieee802154-sniffer\src\main.cpp:79:5: error: 'uart' was not declared in this scope; did you mean 'uart1'?
   79 |     uart.enable();
      |     ^~~~
      |     uart1
scons: *** [build\openmote-b\projects\ieee802154-sniffer\main.o] Error 1
scons: building terminated because of errors.` 

I have attached the code for your reference. Please let me know how I can rectify this error. 
Thanks & Regards, 
DIvya 
[ieee802154_sniffer.txt](https://github.com/contiki-ng/contiki-ng/files/9858663/ieee802154_sniffer.txt)

kkrentz, 2022-10-27:
This looks like a program for the FreeRTOS operating system and not for Contiki-NG.

nvt, 2022-10-27:
This is not a Contiki-NG issue -- closing.

closed 2022-10-27 by nvt