contiki-ng/contiki-ng #2251: FAILURE: Build failed with an exception.

SolBee, 2022-11-03:
When I try to run Cooja using `./gradlew run`, the system gave errors that 'FAILURE: Build failed with an exception.'. How can I solve this problem?

I got the following results when I used `./gradlew -i run`.

> ubuntu@DESKTOP-U31CE80:~/contiki-ng/tools/cooja$ ./gradlew -i run
Initialized native services in: /home/ubuntu/.gradle/native
Initialized jansi services in: /home/ubuntu/.gradle/native
The client will now receive all logging from the daemon (pid: 2387). The daemon log file: /home/ubuntu/.gradle/daemon/7.5.1/daemon-2387.out.log
Starting 14th build in daemon [uptime: 18 mins 37.264 secs, performance: 98%, non-heap usage: 26% of 256 MiB]
Using 12 worker leases.
Now considering [/home/ubuntu/contiki-ng/tools/cooja] as hierarchies to watch
Watching the file system is configured to be enabled if available
File system watching is active
Starting Build
Settings evaluated using settings file '/home/ubuntu/contiki-ng/tools/cooja/settings.gradle'.
Projects loaded. Root project using build file '/home/ubuntu/contiki-ng/tools/cooja/build.gradle'.
Included projects: [root project 'cooja', project ':cooja']

> Configure project :
Evaluating root project 'cooja' using build file '/home/ubuntu/contiki-ng/tools/cooja/build.gradle'.
Compiling build file '/home/ubuntu/contiki-ng/tools/cooja/build.gradle' using SubsetScriptTransformer.
Compiling build file '/home/ubuntu/contiki-ng/tools/cooja/build.gradle' using BuildScriptTransformer.

> Configure project :cooja
Evaluating project ':cooja' using build file '/home/ubuntu/contiki-ng/tools/cooja/cooja/build.gradle'.
All projects evaluated.
Task name matched 'run'
Selected primary task 'run' from project :
Tasks to be executed: [task ':compileJava', task ':processResources', task ':classes', task ':run']
Tasks that were excluded: []
Resolve mutations for :compileJava (Thread[Execution worker,5,main]) started.
Resolve mutations for :compileJava (Thread[Execution worker,5,main]) completed. Took 0.0 secs.
:compileJava (Thread[Execution worker Thread 8,5,main]) started.

> Task :compileJava UP-TO-DATE
Caching disabled for task ':compileJava' because:
  Build cache is disabled
Skipping task ':compileJava' as it is up-to-date.
:compileJava (Thread[Execution worker Thread 8,5,main]) completed. Took 0.015 secs.
Resolve mutations for :processResources (Thread[included builds,5,main]) started.
Resolve mutations for :processResources (Thread[included builds,5,main]) completed. Took 0.0 secs.
:processResources (Thread[Execution worker Thread 8,5,main]) started.
producer locations for task group 0 (Thread[Execution worker Thread 11,5,main]) started.
producer locations for task group 0 (Thread[Execution worker Thread 11,5,main]) completed. Took 0.0 secs.

> Task :processResources UP-TO-DATE
Caching disabled for task ':processResources' because:
  Build cache is disabled
Skipping task ':processResources' as it is up-to-date.
:processResources (Thread[Execution worker Thread 8,5,main]) completed. Took 0.002 secs.
Resolve mutations for :classes (Thread[included builds,5,main]) started.
Resolve mutations for :classes (Thread[included builds,5,main]) completed. Took 0.0 secs.
:classes (Thread[Execution worker Thread 6,5,main]) started.

> Task :classes UP-TO-DATE
Skipping task ':classes' as it has no actions.
:classes (Thread[Execution worker Thread 6,5,main]) completed. Took 0.0 secs.
Resolve mutations for :run (Thread[Execution worker Thread 2,5,main]) started.
Resolve mutations for :run (Thread[Execution worker Thread 2,5,main]) completed. Took 0.0 secs.
:run (Thread[Execution worker Thread 6,5,main]) started.

> Task :run FAILED
Caching disabled for task ':run' because:
  Build cache is disabled
Task ':run' is not up-to-date because:
  Task has not declared any outputs despite executing actions.
Starting process 'command '/usr/lib/jvm/java-17-openjdk-amd64/bin/java''. Working directory: /home/ubuntu/contiki-ng/tools/cooja Command: /usr/lib/jvm/java-17-openjdk-amd64/bin/java -XX:-UseCompressedOops -XX:-UseCompressedClassPointers --enable-native-access ALL-UNNAMED --add-modules jdk.incubator.foreign -Xms400M -Xmx2048M -Dfile.encoding=UTF-8 -Duser.country -Duser.language=en -Duser.variant -cp /home/ubuntu/contiki-ng/tools/cooja/build/classes/java/main:/home/ubuntu/contiki-ng/tools/cooja/build/resources/main:/home/ubuntu/contiki-ng/tools/cooja/lib/jcommon-1.0.14.jar:/home/ubuntu/contiki-ng/tools/cooja/lib/jdom.jar:/home/ubuntu/contiki-ng/tools/cooja/lib/jfreechart-1.0.11.jar:/home/ubuntu/contiki-ng/tools/cooja/lib/jipv6.jar:/home/ubuntu/contiki-ng/tools/cooja/lib/json-simple-1.1.1.jar:/home/ubuntu/contiki-ng/tools/cooja/lib/picocli-4.6.3.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/de.sciss/syntaxpane/1.2.1/293281498ec1c3152c32c9b2d90c5b440ce91c94/syntaxpane-1.2.1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.apache.logging.log4j/log4j-core/2.19.0/3b6eeb4de4c49c0fe38a4ee27188ff5fee44d0bb/log4j-core-2.19.0.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.openjdk.nashorn/nashorn-core/15.4/f67f5ffaa5f5130cf6fb9b133da00c7df3b532a5/nashorn-core-15.4.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.swinglabs.swingx/swingx-autocomplete/1.6.5-1/6dde6b0041e263c0d9baa94cae3350cdbb5d0e67/swingx-autocomplete-1.6.5-1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.apache.logging.log4j/log4j-api/2.19.0/ea1b37f38c327596b216542bc636cfdc0b8036fa/log4j-api-2.19.0.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm-commons/7.3.1/daaa79ef260eb67404b9a52bc319a024c7f49cfe/asm-commons-7.3.1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm-util/7.3.1/cac1bf54c2fb86671c357d281d1060fe5d50a0de/asm-util-7.3.1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm-analysis/7.3.1/45dfd299ea0c17d534499c4f06417ceccfa2d02/asm-analysis-7.3.1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm-tree/7.3.1/587ce54d243145b2e89598bfcea7823ded73be5d/asm-tree-7.3.1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.ow2.asm/asm/7.3.1/7ec32f922315924e82bf58b36ee1b673b2a9b820/asm-7.3.1.jar:/home/ubuntu/.gradle/caches/modules-2/files-2.1/org.swinglabs.swingx/swingx-common/1.6.5-1/d5ca5d6b5568aaa41a12ca5c9277cf5e37d18199/swingx-common-1.6.5-1.jar org.contikios.cooja.Main -cooja /home/ubuntu/contiki-ng/tools/cooja
Successfully started process 'command '/usr/lib/jvm/java-17-openjdk-amd64/bin/java''
WARNING: Using incubator modules: jdk.incubator.foreign
Trying to start GUI in headless environment, aborting
:run (Thread[Execution worker Thread 6,5,main]) completed. Took 0.229 secs.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':run'.
> Process 'command '/usr/lib/jvm/java-17-openjdk-amd64/bin/java'' finished with non-zero exit value 1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 930ms
3 actionable tasks: 1 executed, 2 up-to-date

pjonsson, 2022-11-03:
The interesting line is this:

`Trying to start GUI in headless environment, aborting`

Your `$DISPLAY` (or similar) is faulty for some reason, and you shouldn't be able to start any other applications that require a graphical user interface either.

SolBee, 2022-11-04:
@pjonsson 
Thank you so much!!!
I got a hint from your answer and was able to solve the problem.

I added export `DISPLAY=$(hostname).local:0` to `~/.bashrc` and cooja worked!!!

closed 2022-11-16 by nfi