contiki-ng/contiki-ng #2285: Bug in IPv6 extension header processing

tinstructor, 2022-11-18:
In `os/net/ipv6/uip6.c` the function `ext_hdr_options_process()` is called twice for every incoming packet (I'm using RPL-Classic in Storing Mode if that matters):
https://github.com/contiki-ng/contiki-ng/blob/0c2a7d18c50e72e49ac0790e84bf7d2aa9350215/os/net/ipv6/uip6.c#L1203-L1214
https://github.com/contiki-ng/contiki-ng/blob/0c2a7d18c50e72e49ac0790e84bf7d2aa9350215/os/net/ipv6/uip6.c#L1295-L1316
This means that when a rank-error is detected, the first call tolerates the single rank error (as per RPL spec) and sets the corresponding bit in the RPL Options. However, the immediate second call to `ext_hdr_options_process()` then causes the packet to be dropped immediately. Effectively, this means that the single-rank-error tolerating mechanism of RPL is currently broken.

arurke, 2023-01-21:
@tinstructor I might be meeting this issue in my work now. Did you get any further on this, e.g. towards a solution? Does it match your understanding that this issue is only happening if the packet is addressed to the current node?

I got to say, debugging that 1000+ lines `uip_process()` function riddled with goto's and defines is not good for mental health.

arurke, 2024-03-07:
Resolved by #2862

closed 2024-03-07 by arurke