contiki-ng/contiki-ng #2315: Cooja build error due to x11 limitation of snap

georgefatsis, 2022-12-03:
Hi Contiki team, 

I am openning this issue just to inform you for a possible issue someone might face.

If the host OS is Ubuntu and the docker was install through snap then you get the followning build error during the cooja execution:

```
> Task :compileJava
warning: using incubating module(s): jdk.incubator.foreign
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
1 warning
> Task :run FAILED
WARNING: Using incubator modules: jdk.incubator.foreign
Exception in thread "main" java.awt.AWTError: Can't connect to X11 window server using ':0' as the value of the DISPLAY variable.
 at java.desktop/sun.awt.X11GraphicsEnvironment.initDisplay(Native Method)
```

The error is a limitation of snap, more info can be found [here](https://askubuntu.com/questions/1249043/run-simple-x11-app-in-docker-container-on-ubuntu-20-04) :
 
```
snap causes several restictions. --hostdisplay does not work because it is not possible to share unix sockets from host, in this case the X unix socket in /tmp/.X11-unix"
```

The solution is to re-install docker using the official docker documentation, [link](https://docs.docker.com/engine/install/ubuntu/).

Please consider including this into your documentation, and also you can close this issue. 

Regards,
George

eduardogomescampos1, 2023-06-24:
I followed the docker installation guide and I'm facing the exact same issue. Do you have any suggestions to overcome this?

joakimeriksson, 2023-07-10:
Added an additional warning on other dockers than the one we link to.

closed 2023-07-12 by joakimeriksson