contiki-ng/contiki-ng #2323: No such file or directory

tlzfighting, 2022-12-08:
I Include the rpl.h file in the hello-world.c.

[image]

>  When I enter the compile command "make TARGET=cc2538dk build/cc2538dk/hello-world.hex", I get the following error:

hello-world.c:26:50: fatal error: os/net/ipv6/routing/rpl-lite/rpl.h: No such file or directory
compilation terminated.
make: *** [../../Makefile.include:498：build/cc2538dk/obj/hello-world.o] error .

How can Isolve this ?  Looking forward to your valuable suggestions!

arurke, 2022-12-08:
Try skipping `ipv6/`

`#include "net/routing/rpl-lite/rpl.h"` should work fine.

tlzfighting, 2022-12-09:
> It works . Thank you ,sir！

closed 2022-12-09 by arurke