contiki-ng/contiki-ng #2371: Getting an error on Contiki, trying to build nullnet broad cast with button-sensor.h

Damiloju123, 2023-01-05:
[image]

Any idea on how to resolve this ?

alexrayne, 2023-01-05:
you have to write button sensor module on your native-platform.
this null-net example well designed on cooja somulation, or bedded
platforms - they always have buttons on dedign-kit.

Damiloju123, 2023-01-06:
> you have to write button sensor module on your native-platform. this null-net example well designed on cooja somulation, or bedded platforms - they always have buttons on dedign-kit. чт, 5 янв. 2023 г., 03:05 Damiloju Olagbade ***@***.***>:
> […](#)

I’m trying to do this for Zolertia mote and also cooja

pjonsson, 2023-06-19:
Thank you for reporting the problem, I think it is fixed in develop:
```
user@c64d0ddcf8fd:~/contiki-ng/examples/nullnet$ make TARGET=native
  MKDIR     build/native/obj/.deps
  CC        nullnet-unicast.c
  CC        ../../arch/platform/native/./platform.c
  CC        ../../arch/platform/native/./clock.c
  CC        ../../arch/platform/native/dev/xmem.c
  CC        ../../arch/platform/native/dev/buttons.c
  CC        ../../arch/platform/native/./cfs-posix.c
  CC        ../../arch/platform/native/./cfs-posix-dir.c
  CC        ../../arch/cpu/native/./rtimer-arch.c
  CC        ../../arch/cpu/native/./watchdog.c
  CC        ../../arch/cpu/native/dev/eeprom.c
  CC        ../../arch/cpu/native/./int-master.c
  CC        ../../arch/cpu/native/dev/gpio-hal-arch.c
  CC        ../../os/contiki-main.c
  CC        ../../os/dev/button-hal.c
  CC        ../../os/dev/gpio-hal.c
  CC        ../../os/dev/leds.c
  CC        ../../os/dev/nullradio.c
  CC        ../../os/dev/serial-line.c
  CC        ../../os/lib/aes-128.c
  CC        ../../os/lib/assert.c
  CC        ../../os/lib/ccm-star.c
  CC        ../../os/lib/circular-list.c
  CC        ../../os/lib/crc16.c
  CC        ../../os/lib/dbl-circ-list.c
  CC        ../../os/lib/dbl-list.c
  CC        ../../os/lib/heapmem.c
  CC        ../../os/lib/hexconv.c
  CC        ../../os/lib/ifft.c
  CC        ../../os/lib/list.c
  CC        ../../os/lib/memb.c
  CC        ../../os/lib/random.c
  CC        ../../os/lib/ringbuf.c
  CC        ../../os/lib/ringbufindex.c
  CC        ../../os/lib/trickle-timer.c
  CC        ../../os/net/link-stats.c
  CC        ../../os/net/linkaddr.c
  CC        ../../os/net/nbr-table.c
  CC        ../../os/net/net-debug.c
  CC        ../../os/net/netstack.c
  CC        ../../os/net/packetbuf.c
  CC        ../../os/net/queuebuf.c
  CC        ../../os/net/mac/mac-sequence.c
  CC        ../../os/net/mac/mac.c
  CC        ../../os/net/mac/csma/anti-replay.c
  CC        ../../os/net/mac/csma/ccm-star-packetbuf.c
  CC        ../../os/net/mac/csma/csma-output.c
  CC        ../../os/net/mac/csma/csma-security.c
  CC        ../../os/net/mac/csma/csma.c
  CC        ../../os/net/mac/framer/frame802154.c
  CC        ../../os/net/mac/framer/frame802154e-ie.c
  CC        ../../os/net/mac/framer/framer-802154.c
  CC        ../../os/net/mac/framer/nullframer.c
  CC        ../../os/net/nullnet/nullnet.c
  CC        ../../os/net/routing/nullrouting/nullrouting.c
  CC        ../../os/sys/atomic.c
  CC        ../../os/sys/autostart.c
  CC        ../../os/sys/compower.c
  CC        ../../os/sys/ctimer.c
  CC        ../../os/sys/energest.c
  CC        ../../os/sys/etimer.c
  CC        ../../os/sys/log.c
  CC        ../../os/sys/mutex.c
  CC        ../../os/sys/node-id.c
  CC        ../../os/sys/process.c
  CC        ../../os/sys/rtimer.c
  CC        ../../os/sys/stimer.c
  CC        ../../os/sys/timer.c
  LD        build/native/nullnet-unicast.native
  CP        build/native/nullnet-unicast.native --> nullnet-unicast.native
  CC        nullnet-broadcast.c
  LD        build/native/nullnet-broadcast.native
  CP        build/native/nullnet-broadcast.native --> nullnet-broadcast.native
user@c64d0ddcf8fd:~/contiki-ng/examples/nullnet$ 
```