contiki-ng/contiki-ng #2414: Simplelink Launchpad CC1312R PWM Driver issue

rajeev1986, 2023-03-21:
I am trying to run PWM on CC1312R Launchpad, however, it seems the driver is not binding and or working properly and I keep getting NULL return for `PWM_open()`;  The PWM pin is routed to CC1312R1_LAUNCHXL_PIN_RLED.

Below is the PWM config code taken from the TI's PWM example and modified for Contiki App. Any insights on it or if I am missing anything will be super helpful.
```
PROCESS_THREAD(pwm_process, ev, data)
{
  static struct etimer timer;

  /* Period and duty in microseconds */
  uint16_t   pwmPeriod = 3000;
  uint16_t   duty = 0;
  uint16_t   dutyInc = 100;

  PWM_Handle pwm1 = NULL;
  PWM_Params params;

  /* Call driver init functions. */
  PWM_init();

  PWM_Params_init(&params);
  params.dutyUnits = PWM_DUTY_US;
  params.dutyValue = 0;
  params.periodUnits = PWM_PERIOD_US;
  params.periodValue = pwmPeriod;

  pwm1 = PWM_open(Board_PWM0, &params);
  if (pwm1 == NULL) 
  {
    /* CC1312R1_LAUNCHXL_PWM0 did not open */
    LOG_ERR("Could not open PWM0\n");
    return 0;
  }
  else
  {
    PWM_start(pwm1);
  }

  PROCESS_BEGIN();

  LOG_INFO("Starting PWM loop process...\n");

  while (1) 
  {
    PWM_setDuty(pwm1, duty);

    duty = (duty + dutyInc);

    if (duty == pwmPeriod || (!duty)) 
    {
        dutyInc = - dutyInc;
    }

    etimer_set(&timer, 7);
    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&timer));  
  }

  PROCESS_END();
}
```

rajeev1986, 2023-03-22:
Aha maybe @g-oikonomou, can shed some light onto this issue!

arurke, 2023-03-22:
I just glanced, but an immediate comment: Be careful with non-static variables inside a process. I suggest you make them static or move it outside of the process.

arurke, 2023-03-23:
Sorry I only have time to glance at this, but one thing to check would be the pin - are you sure it has not already been claimed by some other drivers.

rajeev1986, 2023-03-23:
> Sorry I only have time to glance at this, but one thing to check would be the pin - are you sure it has not already been claimed by some other drivers.

Going over the board pin assignments, all looks good to me.  I also checked it with the pin assignment from TI-RTOS core sdk, where the TI-RTOS pwm example works but the contiki one is failing to open the pwm instance. I had no issues so far with other peripheral drivers such as I2C, SPI, & UART which work as expected. I will continue digging and see if anything surfaces up.

@arurke appreciate though for your time. Cheers

rajeev1986, 2023-03-23:
@arurke, uff finally managed to get it going... So this is what I had to do to make the PWM driver bind and open the pwm instance properly in Contiki-NG. Although, a long route (not sure still why this does not function directly from the app layer) but works for me for now :)

I had to create a pwm driver layer `pwm0-arch.c` at `arch/cpu/simplelink-cc13xx-cc26xx/dev`, `init()` it in `platform.c` and provide the relevant apis in the `pwm0-arch.h` that can be called by the app layer to start the specified PWM handler with the specified settings, set the duty, period, etc,.

No pin configuration issues were there that you mentioned earlier.

Now the dev can move on.....

closed 2023-03-23 by rajeev1986

arurke, 2023-03-23:
Great that you figured it out! Fell free to open a pull request if you believe others might benefit from it.