contiki-ng/contiki-ng #2415: Applying Reinforcement Learning technique to RPL protocol

NajimDH, 2023-03-28:
Thank you for your video,
Based on many papers, for **routing functionality** in IoT-WSN networks, we can utilize artificial intelligence algorithms to provide the routing protocol with self-learning and self-adaptive capabilities. Can anyone please give guidance about how to apply Reinforcement Learning to **RPL protocol** using the **COOJA** simulator?
thanks in advance.

khanhongithub, 2023-04-30:
I am also looking for an example

NajimDH, 2023-05-01:
Hi, Because there is a little feedback, if you like we can work together on this problem.

khanhongithub, 2023-05-01:
Hey for sure. Im looking forward to discussing it

NajimDH, 2023-05-01:
Hi Khanh,

Thank you for your reply, I will send you via WhatsApp.

Regards,