contiki-ng/contiki-ng #2460: COOJA4.9: Fail to reload the existing DGRM

LeoLee0097, 2023-06-11:
The newest version of COOJA raises the following error when I reload my Simulation with Z1. Please notice that there is no error message during my first run, the problem happens during the reload.

```java
WARNING: Using incubator modules: jdk.incubator.foreign
 INFO [SwingWorker-pool-2-thread-1] (Simulation.java:174) - Simulation /home/user/contiki-ng/examples/E_Exp/delay_linear_dgrm.csc random seed: 123456
 INFO [AWT-EventQueue-0] (MessageListUI.java:268) - 
COMPILATION OUTPUT:

 INFO [AWT-EventQueue-0] (MessageListUI.java:273) - org.contikios.cooja.Cooja$SimulationCreationException: Unknown error
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:1383)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.GUI$30.doInBackground(GUI.java:1308)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.GUI$30.doInBackground(GUI.java:1296)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.base/java.lang.Thread.run(Thread.java:833)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) - Caused by: java.lang.RuntimeException: Unknown class: org.contikios.cooja.radiomediums.DGRMDestinationRadio
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.radiomediums.DirectedGraphMedium.simulationFinishedLoading(DirectedGraphMedium.java:416)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.Simulation.<init>(Simulation.java:332)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.Cooja.createSimulation(Cooja.java:1425)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.Cooja.loadSimulationConfig(Cooja.java:1377)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     ... 8 more
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) - Caused by: java.lang.NoSuchMethodException: org.contikios.cooja.radiomediums.DGRMDestinationRadio.<init>()
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.base/java.lang.Class.getConstructor0(Class.java:3585)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at java.base/java.lang.Class.getDeclaredConstructor(Class.java:2754)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     at org.contikios.cooja.radiomediums.DirectedGraphMedium.simulationFinishedLoading(DirectedGraphMedium.java:412)
 INFO [AWT-EventQueue-0] (MessageListUI.java:273) -     ... 11 more
 INFO [AWT-EventQueue-0] (MessageListUI.java:275) -
```

pjonsson, 2023-06-12:
What is the newest version of Cooja, the version you get in the develop branch for this repository after running `git submodule update`, or HEAD in the master branch in the cooja repository?

Cooja has been sped up, and that has exposed some old race conditions. Do you have the option to reload/load the simulation, and if so, does it work on the second attempt?

Please file future Cooja bugs at https://github.com/contiki-ng/cooja

LeoLee0097, 2023-06-15:
@pjonsson Hi pjonsson
Great thanks for your reply. I am working on the latest Contiki-NG (Dev). Here are my operations:
```bash
git clone [Contiki-ng] --depth==1
git submodule update --init --recursive
```
After that, I downloaded the Docker image and started my container. I directly ran the Cooja:
```bash
cng # attach my image shell
# In my image shell, Contiki-NG folder
cooja # Run the Cooja
```
In the Cooja, I used `ctrl+R` to reload my DGRM experiment, then I failed again and again.

The error is shown in the first comment above.

pjonsson, 2023-06-15:
I'm not sure if it's the issue you are seeing, but a workaround for one race condition on reload was just merged to the Cooja repository (https://github.com/contiki-ng/cooja/pull/1042).

Unfortunately, the Cooja that is in the develop branch right now is quite a few commits behind master in the cooja repository, but the submodule will be updated "soon".

In the meantime (I'm not sure how `--depth 1` affects submodules), I think:
```
cd tools/cooja
git checkout master
cd ../..
```
should be sufficient for you to get the latest fixes available in the Cooja master branch. That also contains a fix to an issue where Cooja correctly detected errors on simulation load, but didn't display them. I'm not sure if the display error-fix applies for your situation, but if it does, you get a reload-button instead of Cooja just saying "didn't work" without any choice of action.

nfi, 2023-06-15:
This is a bug that was fixed in https://github.com/contiki-ng/cooja/pull/1210. It should work better if you test with latest upstream Cooja like Peter suggested.

nfi, 2023-06-19:
Contiki-NG has been updated with the latest version of Cooja via https://github.com/contiki-ng/contiki-ng/pull/2463, including a fix for this issue.

closed 2023-06-19 by nfi