contiki-ng/contiki-ng #2468: /home/user/contiki-ng has wrong permissions inside container

isinyaaa, 2023-06-22:
I'm running `contiker/contiki-ng` in rootless mode in Linux as described in the [Docker/Setup docs](https://docs.contiki-ng.org/en/develop/doc/getting-started/Docker.html#setup).

This is what I get when trying to `make` the `hello_world` example:

```raw
user@635ff2dfa9a8:~/contiki-ng$ cd examples/hello-world/
user@635ff2dfa9a8:~/contiki-ng/examples/hello-world$ make
TARGET not defined, using target 'native'
Error cleaning LFS object: open /home/user/contiki-ng/.git/modules/arch/cpu/gecko/lib/gecko_sdk/lfs/tmp/3360473382: no such file or directory
error: external filter 'git-lfs filter-process' failed
fatal: app/bluetooth/common/cbap_lib/lib/cbap_CM33_gcc.a: clean filter 'lfs' failed
fatal: 'git status --porcelain=2' failed in submodule arch/cpu/gecko/lib/gecko_sdk
  MKDIR     build/native/obj/.deps
mkdir: cannot create directory ‘build’: Permission denied
make: *** [../../Makefile.include:239: build/native/obj/.deps] Error 1
user@635ff2dfa9a8:~/contiki-ng/examples/hello-world$ ls -la
total 12
drwxrwxr-x 1 root root   60 Jun 21 17:02 .
drwxrwxr-x 1 root root  378 Jun 21 17:02 ..
-rw-rw-r-- 1 root root  107 Jun 21 17:02 Makefile
-rw-rw-r-- 1 root root  570 Jun 21 17:02 README.md
-rw-rw-r-- 1 root root 2616 Jun 21 17:02 hello-world.c
```

I have also checked that the permissions on the source code directory on the host are set normally.

pjonsson, 2023-06-23:
I doubt the issue is caused by Arch Linux, but here's what it looks like on my Ubuntu 22.04:
```
user@b92fdf9180a3:~/contiki-ng$ ls -la
total 10564
drwxrwxr-x 11 user user    4096 Jun 15 09:59 .
drwxr-x---  1 user user    4096 Jun 17 10:17 ..
drwxrwxr-x 10 user user    4096 Jun 23 08:51 .git
...
```
The linked page does not mention rootless mode, and I'm not aware of anyone using it with Contiki-NG. I suggest you try with a fresh clone of the Contki-NG repository and see if that works. If you have a uid/gid that is different from 1000 on the host, you could start the image with `-e LOCAL_UID=$(id -u $USER) -e LOCAL_GID=$(id -g $USER)` so `/home/user` will be owned by your uid/gid instead of the default 1000/1000.

isinyaaa, 2023-06-23:
> The linked page does not mention rootless mode, and I'm not aware of anyone using it with Contiki-NG.

I'm sorry about the confusion. I've tried executing docker in non-rootless mode and it works fine :)

closed 2023-06-23 by isinyaaa